home-assistant/operating-system #2496: 10.0 Failed to start QEMU Guest Agent when rebooted from Proxmox HyperVisor

runwaygoats, 2023-04-21:
### Describe the issue you are experiencing

QEMU Guest Agent fails to start only when I reboot Home Assistant Operating System 10.0 (HAOS) from Proxmox QEMU by the command "qm reboot <vmid>".

No problem when HAOS 10.0 is rebooted from website "Home Assistant > System" or Home Assistant CLI by command "host reboot"

This problem exist only on HAOS 10.0. Tested on HAOS 9.5 and HAOS 9.5 seems immune to this problem. 

When I reboot again from within Home Assistant Website or CLI, it fixes itself.

Home Assistant Operating System is run on Proxmox PVE 7.4.3.

Virtual Machines has been created from following disk images:
  haos_ova-10.0.qcow2
  haos_ova-9.5.qcow2

Fresh installation.
  Only username, password, country and language is set during onboarding.
  Nothing else is changed.
  No device, service, add-on, etc. added.

Proxmox QEMU Virtual Machine Configuration
  agent: 1
  bios: ovmf
  cpu: kvm64,flags=+md-clear;+pcid;+spec-ctrl;+ssbd;+pdpe1gb;+aes
  machine: q35
  memory: 2048
  meta: creation-qemu=7.2.0,ctime=1682083929

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Reboot HAOS 10.0 within Proxmox by command "qm reboot <vmid>.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Fresh installation.
No logs.
```

### Anything in the Host logs that might be useful for us?

```txt
Fresh installation.
No logs.
```

### System information

## System Information

version | core-2023.4.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.24
arch | x86_64
timezone | UTC
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 30.8 GB
disk_used | 3.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 21, 2023 at 1:36 PM
-- | --
current_recorder_run | April 21, 2023 at 10:07 AM
estimated_db_size | 0.46 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

On HAOS 9.5, during boot I see only one "QEMU Guest Agent started" message.

On HAOS 10.0, I see "Failed to start QEMU Guest Agent", after rebooting within Proxmox.
If I reboot after within Home Assistant, I see many "Stopped QEMU Guest Agent" and "Started QEMU Guest Agent" messages. But QEMU Guest Agent works fine, even if I see these messages.

runwaygoats, 2023-04-22:
Home Assistant CLI

login
systemctl status qemu-guest.service

...
restart counter is at 5.
...
qemu-guest.service: Start request repeated too quickly.
...

systemctl start qemu-guest.service

Service started normally.

runwaygoats, 2023-04-23:
# journalctl -b | grep 'qemu-guest'

Apr 22 23:49:25 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 23:49:25 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 1.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 2.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 3.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 4.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 5.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Start request repeated too quickly.
Apr 22 23:49:26 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.

# systemctl start qemu-guest
# systemctl status qemu-guest

● qemu-guest.service - QEMU Guest Agent
     Loaded: loaded (/usr/lib/systemd/system/qemu-guest.service; enabled; preset: enabled)
     Active: active (running) since Sun 2023-04-23 01:25:14 UTC; 1min 51s ago
   Main PID: 10450 (qemu-ga)
      Tasks: 2 (limit: 2325)
     Memory: 588.0K
        CPU: 68ms
     CGroup: /system.slice/qemu-guest.service
             └─10450 /usr/libexec/qemu-ga -m virtio-serial -p /dev/virtio-ports/org.qemu.guest_agent.0

Apr 23 01:25:31 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:25:34 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:25:44 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:25:55 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:05 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:16 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:26 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:37 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:47 homeassistant qemu-ga[10450]: info: guest-ping called
Apr 23 01:26:58 homeassistant qemu-ga[10450]: info: guest-ping called

runwaygoats, 2023-04-23:
journalctl -b

Apr 23 12:34:13 homeassistant systemd[1]: Started Dropbear SSH daemon.
Apr 23 12:34:13 homeassistant qemu-ga[320]: 1682253253.180264: critical: error opening channel '/dev/virtio-ports/org.qemu.guest_agent.0': No such file or directory
Apr 23 12:34:13 homeassistant qemu-ga[320]: 1682253253.180612: critical: failed to create guest agent channel
Apr 23 12:34:13 homeassistant qemu-ga[320]: 1682253253.180621: critical: failed to initialize guest agent channel
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 23 12:34:13 homeassistant systemd[1]: Finished Permit User Sessions.

Apr 23 12:34:13 homeassistant systemd[1]: Finished HassOS Configuration Manager.
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 1.
Apr 23 12:34:13 homeassistant systemd[1]: Stopped QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Started QEMU Guest Agent.
Apr 23 12:34:13 homeassistant qemu-ga[335]: 1682253253.306796: critical: error opening channel '/dev/virtio-ports/org.qemu.guest_agent.0': No such file or directory
Apr 23 12:34:13 homeassistant qemu-ga[335]: 1682253253.307041: critical: failed to create guest agent channel
Apr 23 12:34:13 homeassistant qemu-ga[335]: 1682253253.307081: critical: failed to initialize guest agent channel
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 23 12:34:13 homeassistant systemd-logind[294]: New seat seat0.

Apr 23 12:34:13 homeassistant systemd[1]: Started User Login Management.
Apr 23 12:34:13 homeassistant containerd[322]: time="2023-04-23T12:34:13.438948029Z" level=info msg="starting containerd" revision= version=1.6.20
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 2.
Apr 23 12:34:13 homeassistant systemd[1]: Stopped QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Started QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Started Hostname Service.
Apr 23 12:34:13 homeassistant qemu-ga[338]: 1682253253.480887: critical: error opening channel '/dev/virtio-ports/org.qemu.guest_agent.0': No such file or directory
Apr 23 12:34:13 homeassistant qemu-ga[338]: 1682253253.481102: critical: failed to create guest agent channel
Apr 23 12:34:13 homeassistant qemu-ga[338]: 1682253253.481109: critical: failed to initialize guest agent channel
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 23 12:34:13 homeassistant NetworkManager[289]: <info>  [1682253253.4859] hostname: hostname: using hostnamed

Apr 23 12:34:13 homeassistant NetworkManager[289]: <info>  [1682253253.6246] manager: startup complete
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 3.
Apr 23 12:34:13 homeassistant systemd[1]: Stopped QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Started QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Finished Network Manager Wait Online.
Apr 23 12:34:13 homeassistant systemd[1]: Reached target Network is Online.
Apr 23 12:34:13 homeassistant qemu-ga[352]: 1682253253.640096: critical: error opening channel '/dev/virtio-ports/org.qemu.guest_agent.0': No such file or directory
Apr 23 12:34:13 homeassistant qemu-ga[352]: 1682253253.640244: critical: failed to create guest agent channel
Apr 23 12:34:13 homeassistant qemu-ga[352]: 1682253253.640250: critical: failed to initialize guest agent channel
Apr 23 12:34:13 homeassistant audit: BPF prog-id=15 op=LOAD
Apr 23 12:34:13 homeassistant systemd[1]: Starting Notify NFS peers of a restart...
Apr 23 12:34:13 homeassistant systemd[1]: Starting Network Time Synchronization...
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 23 12:34:13 homeassistant sm-notify[354]: Version 2.6.2 starting
Apr 23 12:34:13 homeassistant systemd[1]: Started Notify NFS peers of a restart.
Apr 23 12:34:13 homeassistant NetworkManager[289]: <info>  [1682253253.6565] manager: NetworkManager state is now CONNECTED_GLOBAL
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Scheduled restart job, restart counter is at 4.
Apr 23 12:34:13 homeassistant systemd[1]: Stopped QEMU Guest Agent.
Apr 23 12:34:13 homeassistant systemd[1]: Started QEMU Guest Agent.
Apr 23 12:34:13 homeassistant qemu-ga[359]: 1682253253.786621: critical: error opening channel '/dev/virtio-ports/org.qemu.guest_agent.0': No such file or directory
Apr 23 12:34:13 homeassistant qemu-ga[359]: 1682253253.786906: critical: failed to create guest agent channel
Apr 23 12:34:13 homeassistant qemu-ga[359]: 1682253253.786950: critical: failed to initialize guest agent channel
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Main process exited, code=exited, status=1/FAILURE
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
[... cut ...]
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Start request repeated too quickly.
Apr 23 12:34:13 homeassistant systemd[1]: qemu-guest.service: Failed with result 'exit-code'.
Apr 23 12:34:13 homeassistant systemd[1]: Failed to start QEMU Guest Agent.
Apr 23 12:34:14 homeassistant dockerd[368]: time="2023-04-23T12:34:14.168652165Z" level=info msg="Starting up"
Apr 23 12:34:14 homeassistant dockerd[368]: time="2023-04-23T12:34:14.169011697Z" level=warning msg="Running experimental build"
Apr 23 12:34:14 homeassistant systemd[1]: Reached target Sound Card.

runwaygoats, 2023-04-23:
# ls -ahl /dev/virtio-ports/

total 0      
drwxr-xr-x    2 root     root          80 Apr 23 12:34 .
drwxr-xr-x   18 root     root        3.4K Apr 23 12:34 ..
lrwxrwxrwx    1 root     root          11 Apr 23 12:34 com.redhat.spice.0 -> ../vport2p1
lrwxrwxrwx    1 root     root          11 Apr 23 12:34 org.qemu.guest_agent.0 -> ../vport1p1

agners, 2023-04-24:
I guess your last message is after a problematic startup? it seems that the files are present after bootup, but not at bootup then? My best guess is that this is related to the startup ordering.

runwaygoats, 2023-04-24:
> I guess your last message is after a problematic startup? it seems that the files are present after bootup, but not at bootup then? My best guess is that this is related to the startup ordering.

Home Assistant is running on a Virtual Machine. Virtual Machine has only one disk. If there was a problem with that, it would not boot up.

This is the list of Home Assistant disks:

# lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
sda      8:0    0   32G  0 disk 
|-sda1   8:1    0   32M  0 part /mnt/boot
|-sda2   8:2    0   24M  0 part 
|-sda3   8:3    0  256M  0 part /
|-sda4   8:4    0   24M  0 part 
|-sda5   8:5    0  256M  0 part 
|-sda6   8:6    0    8M  0 part 
|-sda7   8:7    0   96M  0 part /var/lib/systemd
|                               /var/lib/NetworkManager
|                               /var/lib/bluetooth
|                               /etc/systemd/timesyncd.conf
|                               /etc/hostname
|                               /etc/hosts
|                               /etc/NetworkManager/system-connections
|                               /root/.docker
|                               /root/.ssh
|                               /etc/modules-load.d
|                               /etc/udev/rules.d
|                               /etc/dropbear
|                               /etc/modprobe.d
|                               /mnt/overlay
`-sda8   8:8    0 31.3G  0 part /var/log/journal
                                /var/lib/docker
                                /mnt/data
zram0  252:0    0    0B  0 disk 
zram1  252:1    0   32M  0 disk 
zram2  252:2    0   16M  0 disk /tmp

agners, 2023-04-25:
This likely will be resolved with 10.1 (on the beta channel soon). If you still experience the problem with HAOS 10.1, please reopen this issue again.