home-assistant/operating-system #2498: Docker - more then 500 unused volumes, increasing number

hklages, 2023-04-21:
### Describe the issue you are experiencing

There is an increasing number of unused volumes - right now 527 - only one is used. On 2023-03-09 every ten minutes a volume has been created. During that time the Fritzbox Router was in trouble. All Addons are from the Home Assistant Add On store.
Unfortunately I can not look inside the volumes to get further information but I am worrying about the amount.

[image]

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Portainer dashboard shows the number of volumes - all except one from Raspberrymatic - are unsed.

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
Apr 21 15:43:32 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.49N8KX.mount: Deactivated successfully.
Apr 21 15:44:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.W3NHoa.mount: Deactivated successfully.
Apr 21 15:44:32 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.dZ7FXQ.mount: Deactivated successfully.
Apr 21 15:45:02 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.YxTfqu.mount: Deactivated successfully.
Apr 21 15:45:15 homeassistant systemd-journald[134]: Data hash table of /var/log/journal/b2bc3481b6364419a4028b5141f7d10f/system.journal has a fill level at 75.0 (85333 of 113777 items, 41943040 file size, 491 bytes per hash table item), suggesting rotation.
Apr 21 15:45:15 homeassistant systemd-journald[134]: /var/log/journal/b2bc3481b6364419a4028b5141f7d10f/system.journal: Journal header limits reached or header out-of-date, rotating.
Apr 21 15:47:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.XMYaSw.mount: Deactivated successfully.
Apr 21 15:49:05 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.q4TMZX.mount: Deactivated successfully.
Apr 21 15:49:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.6Y8LCU.mount: Deactivated successfully.
Apr 21 15:50:06 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.yIkpIt.mount: Deactivated successfully.
Apr 21 15:50:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.dvLgFR.mount: Deactivated successfully.
Apr 21 15:52:39 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.QeORn1.mount: Deactivated successfully.
Apr 21 15:55:05 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.tFReuM.mount: Deactivated successfully.
Apr 21 15:56:12 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.bBpcDv.mount: Deactivated successfully.
Apr 21 15:56:36 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.ZFtDIX.mount: Deactivated successfully.
Apr 21 15:57:43 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.fuZtr9.mount: Deactivated successfully.
Apr 21 16:00:46 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.lATnyW.mount: Deactivated successfully.
Apr 21 16:01:38 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.z0Vk6n.mount: Deactivated successfully.
Apr 21 16:02:08 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.vmTR9z.mount: Deactivated successfully.
Apr 21 16:02:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.fWmc1E.mount: Deactivated successfully.
Apr 21 16:03:08 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.99Deza.mount: Deactivated successfully.
Apr 21 16:04:08 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.1uDXpE.mount: Deactivated successfully.
Apr 21 16:06:39 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.r1rPWR.mount: Deactivated successfully.
Apr 21 16:07:39 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.XUvnra.mount: Deactivated successfully.
Apr 21 16:08:23 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.M9TwrV.mount: Deactivated successfully.
Apr 21 16:08:40 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.wtsvNQ.mount: Deactivated successfully.
Apr 21 16:09:10 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.v8iSBn.mount: Deactivated successfully.
Apr 21 16:09:40 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.putzhx.mount: Deactivated successfully.
Apr 21 16:10:25 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.4pXLwQ.mount: Deactivated successfully.
Apr 21 16:11:11 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.NTmjZT.mount: Deactivated successfully.
Apr 21 16:14:42 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.TMpXba.mount: Deactivated successfully.
Apr 21 16:15:29 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.MRw6oV.mount: Deactivated successfully.
Apr 21 16:15:42 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.MRhXqe.mount: Deactivated successfully.
Apr 21 16:16:12 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.wxQbdm.mount: Deactivated successfully.
Apr 21 16:17:31 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.wIIZsM.mount: Deactivated successfully.
Apr 21 16:19:03 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.VGa5Uv.mount: Deactivated successfully.
Apr 21 16:19:43 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.nw1Rtb.mount: Deactivated successfully.
Apr 21 16:20:13 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.RsCSn6.mount: Deactivated successfully.
[... 43 lines cut ...]
Apr 21 16:55:35 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.z7tITy.mount: Deactivated successfully.
Apr 21 16:56:06 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.dDxxls.mount: Deactivated successfully.
[... cut ...]
Apr 21 17:03:07 homeassistant systemd-journal-gatewayd[892079]: microhttpd: MHD_OPTION_EXTERNAL_LOGGER is not the first option specified for the daemon. Some messages may be printed by the standard MHD logger.
Apr 21 17:03:12 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-6529a745d587fda76598e87f56b7c85669729b746b92422cdc4f5d254d8a4203-runc.vFX532.mount: Deactivated successfully.
Apr 21 17:03:26 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-50f094f0f4db2ac4a77b1afe681640cb30ccbf5ab438302b170bce8c2dcda599-runc.vNHvF0.mount: Deactivated successfully.
```

### System information

No repairs

agners, 2023-04-24:
@hklages do you happen to know which add-on is creating those? Can you try delete the one without "unused" label, it should generate an error saying which container is using it.

Also, can you provide System information? System information can be found in [Settings -> System -> Repairs -> System Information (top right menu)](https://my.home-assistant.io/redirect/system_health/), Click the copy button at the bottom of the pop-up and paste it here.

hklages, 2023-04-24:
Hi.

System repair says:"There are currently no repairs available"

The used volume is from Raspberrymatic add in. I can see that as I belongs
to a running container ( with Portainer)

For the unused volumes I don't have any information.

agners, 2023-04-25:
Any useful information in the RaspberryMatic add-on logs?

hklages, 2023-04-26:
No.  It is strange. How can I access the Docker in Home Assistant? With terminal? Putty? Are the Docker volumes at "/var/lib/docker/volumes/"? 

```
Mounting /data as /usr/local (Home Assistant Add-On): OK
Starting watchdog...
Identifying host system: oci, OK
Initializing RTC Clock: no hardware found
Running sysctl: OK
Checking for Factory Reset: not required
Checking for Backup Restore: not required
Initializing System: OK
Setup ca-certificates: OK
Starting logging: OK
Init onboard LEDs: init, OK
Starting irqbalance: OK
Saving 256 bits of creditable seed for next boot
Starting iptables: OK
Starting network: eth0: link up, fixed, firewall, inet up, 172.30.33.2, OK
Identifying Homematic RF-Hardware: ....HmRF: HMIP-RFUSB/eQ-3 HmIP-RFUSB@usb-0000:01:00.0-1.3, HmIP: HMIP-RFUSB/eQ-3 HmIP-RFUSB@usb-0000:01:00.0-1.3, OK
Updating Homematic RF-Hardware: HMIP-RFUSB: 4.4.18, not necessary, OK
Starting hs485dLoader: disabled
Starting xinetd: OK
Starting eq3configd: OK
Starting lighttpd: OK
Starting ser2net: disabled
Starting ssdpd: OK
Starting sshd: OK
Starting ha-proxy: OK
Starting NUT services: disabled
Initializing Third-Party Addons: OK
Starting LGWFirmwareUpdate: ...OK
Setting LAN Gateway keys: OK
Starting hs485d: disabled
Starting multimacd: .......OK
Starting rfd: .OK
Starting HMIPServer: ..............OK
Starting ReGaHss: .OK
Starting CloudMatic: OK
Starting Third-Party Addons: OK
Starting crond: OK
Setup onboard LEDs: booted, OK
Finished Boot: 3.69.6.20230407 (raspmatic_oci_arm64)

```

agners, 2023-04-28:
> No. It is strange. How can I access the Docker in Home Assistant? With terminal? Putty? Are the Docker volumes at "/var/lib/docker/volumes/"?

You can use [this guide](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host) to access the root shell of Home Assistant OS via SSH on port 22222. The volumes are at `/mnt/data/docker/volumes/`.

hklages, 2023-04-28:
Thanks. I am currently on vacation, will try that next week and let you
know.

Baxxy13, 2023-05-02:
I can confirm that starting the RaspberryMatic AddOn creates a new volume at /mnt/data/docker/volumes.
Stopping (and starting) or restarting from HA or within the AddOn leaves the volume as unused behind and creates a new one.
cli:
[image]
portainer:
[image]

hklages, 2023-05-02:
Thanks @Baxxy13  - I can confirm that with every HA restart or every Raspberrymatic Add On restart a new volume is being created and the previous one shows up in the list of unused. 

Is is ok to delete the unused one? Usually that is ok. 

I just opened an issue on the [add on homepage.](https://github.com/jens-maus/RaspberryMatic/issues/2303)

jens-maus, 2023-05-02:
Hmn. This sounds strange since the add-on does of course not explicitly create these volumes. Thus, there might be some problems/issues within the volume/docker management in HAos!?! @agners Is there something strange going on here or do you have any explanation why @hklages and @Baxxy13 are seeing new container volumes being created with each new add-on start? This was definitely not the case in previous versions of HAos+RaspberryMatic add-on. And as you know, the add-on usually does not have control on volume generation. Thus, could other add-on potentially suffer from the same issue? See here for the RaspberryMatic add-on config: https://github.com/jens-maus/RaspberryMatic/blob/master/home-assistant-addon/config.yaml

hklages, 2023-05-03:
The creation date of the oldest volume (2021-10-22 09:55:57) is the same as the date I removed the physical CCU and installed the USB stick. There are 28 in the time period 2021-10-22 to 2021-12-31 and similar amount of data after that. 

A lot of volumes were created in the time from 2023-03-05 to 2023-03-09 - about 6 per hour. I was on vacation and neither Home Assistant nor the whole network (including the very stable Fritzbox-Router) was reachable.  Maybe the Fritzbox crashed and Home Assistant rebooted permanently. On 2023-03-09  - I was still on vacation - everything recovered and I was able to access the whole network and no volumes were created anymore.

agners, 2023-05-03:
@hklages thanks for the detailed information. It seems that this is system and/or add-on restart related.

@jens-maus the Supervisor does not make use of volumes really. We use bind mounts from what I can tell. But I see that the RaspberryMatic add-on indeed uses a volume:

```json
            {
                "Type": "volume",
                "Name": "d250fa15b40ae30a6b9821097dbfd7138670d9d06d561f3e85f08313b97442c2",
                "Source": "/mnt/data/docker/volumes/d250fa15b40ae30a6b9821097dbfd7138670d9d06d561f3e85f08313b97442c2/_data",
                "Destination": "/usr/local",
                "Driver": "local",
                "Mode": "",
                "RW": true,
                "Propagation": ""
            },
```

Where is the Dockerfile to that add-on?

agners, 2023-05-03:
I guess this is the problem: https://github.com/jens-maus/RaspberryMatic/blob/master/buildroot-external/board/oci/Dockerfile#L25

Supervisor doesn't really handle volumes requested by the container image. It seems that by default Docker creates a volume whenever the container gets created, and doesn't cleanup automatically.

Ideally you just remove this `VOLUME` specification in the Dockerfile, then no more volumes should appear.

Not sure if there are good reasons to use volumes in your case, but if you'd like volume support in Supervisor we should create an issue in the [Supervisor repo](https://github.com/home-assistant/supervisor) and continue discussion there.

@harshjk from what I can see the contents of the volume is just a static configuration file, it should be safe to get rid of those. Use `docker volume prune -f` to remove all volumes.

I am closing the issue here as this is not really an OS problem.

closed 2023-05-03 by agners

jens-maus, 2023-05-03:
> I guess this is the problem: https://github.com/jens-maus/RaspberryMatic/blob/master/buildroot-external/board/oci/Dockerfile#L25
> 
> Supervisor doesn't really handle volumes requested by the container image. It seems that by default Docker creates a volume whenever the container gets created, and doesn't cleanup automatically.
> 
> @jens-maus ideally you just remove this `VOLUME` specification in the Dockerfile, then no more volumes should appear.
> 
> Not sure if there are good reasons to use volumes in your case, but if you'd like volume support in Supervisor we should create an issue in the [Supervisor repo](https://github.com/home-assistant/supervisor) and continue discussion there.

That's actually a good finding. The reason for this VOLUME statement is, that RaspberryMatic can also be used completely unrelated to HomeAssistant by running it as a dedicated docker container (e.g. under Ubuntu) and that's the reason for this statement since all its config data is usually stored under `/usr/local`. And that Dockerfile is simply used for both (RaspberryMatic addon under HA and dedicated RaspberryMatic docker container) use cases and then in the HA case a startup script is simply bind mounting `/data` to `/usr/local` (see https://github.com/jens-maus/RaspberryMatic/blob/master/buildroot-external/overlay/base-raspmatic_oci/etc/init.d/S00DataMount). Now of course the question is how I could prevent the docker volume to be created each time (which I didn't know) for the HA case and keep it going for the custom docker container use since I of course don't want to generate two different docker contains just to manage this difference.... Any suggestion how to handle this @agners?

RubenKelevra, 2024-11-01:
I've been noticing storage issues as well, and I suspect this might be the reason. I’m running on a 64 GB SD card and currently have only 1.2 GB free, even though my full backups are around 3.4 GB each, and I typically keep two backups locally. I’m not sure what’s using up the remaining space.

According to Portainer, I have 342 volumes, but only two are actively in use. Unfortunately, there’s no straightforward way to identify which containers created the unused volumes.

The volumes currently mounted by add-ons are as follows:

- `addon_db21ed7f_postgres` at `/var/lib/postgresql/data`
- `addon_db21ed7f_epicgamesfree` at `/usr/app/config`

@agners suggested opening a ticket in [Home Assistant Supervisor](https://github.com/home-assistant/supervisor), but since that hasn’t happened yet, I figured I’d chime in here.

After clearing out the unused volumes, I freed up 1.3 GB, and Home Assistant is *way* more responsive now. It had gotten really sluggish over the last few months, so this was a big improvement—though I’m still a bit puzzled about what’s using the rest of my storage.

Thanks for any insights, and I hope this helps others experiencing similar issues!

Impact123, 2024-11-02:
> I’m still a bit puzzled about what’s using the rest of my storage.

If you have discord you can follow this where I explained how to do that: https://discord.com/channels/330944238910963714/332167321311510530/1098322242812784793