home-assistant/operating-system #2499: After the OS 10.0 update, no further updates can be performed

gya352, 2023-04-21:
### Describe the issue you are experiencing

I updated the OS to version10. 
No further updates can be made after that.

When I want to update the Home Assistant Core to 4.5 comes an error:
`Error updating Home Assistant Core 'HomeAssistantCore.update' blocked from execution, no host internet connection`
 
When I want to update the ESPHome addon to 4.0 comes an error:
`Error updating ESPHome: 'AddonManager.update' blocked from execution, no host internet connection`

Internet is working on other device.
I switched off/on the Home Assistant PC, and the ADSL router, but the problem remained.

After that I downgraded the OS to 9.5 and everything works fine.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.updated the OS to 10.0
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-21 19:37:46 WARNING (MainThread) [supervisor.jobs] 'Updater.fetch_data' blocked from execution, no supervisor internet connection
```

### Anything in the Host logs that might be useful for us?

```txt
2023-04-21 20:23:13.575 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139643510990464] Error updating ESPHome: 'AddonManager.update' blocked from execution, no host internet connection
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 167, in async_install
await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 51, in _wrapper
raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: 'AddonManager.update' blocked from execution, no host internet connection
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1820, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1857, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 216, in handle_service
await service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 977, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 156, in async_install
await entity.async_install_with_progress(version, backup)
File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 417, in async_install_with_progress
await self.async_install(version, backup)
File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 169, in async_install
raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating ESPHome: 'AddonManager.update' blocked from execution, no host internet connection
2023-04-21 20:23:18.703 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [139643510990464] Error updating Home Assistant Core 'HomeAssistantCore.update' blocked from execution, no host internet connection
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 303, in async_install
await async_update_core(self.hass, version=version, backup=backup)
File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 51, in _wrapper
raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: 'HomeAssistantCore.update' blocked from execution, no host internet connection
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 199, in handle_call_service
await hass.services.async_call(
File "/usr/src/homeassistant/homeassistant/core.py", line 1820, in async_call
task.result()
File "/usr/src/homeassistant/homeassistant/core.py", line 1857, in _execute_service
await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 216, in handle_service
await service.entity_service_call(
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
future.result() # pop exception if have
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 977, in async_request_call
await coro
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
await result
File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 156, in async_install
await entity.async_install_with_progress(version, backup)
File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 417, in async_install_with_progress
await self.async_install(version, backup)
File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 305, in async_install
raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Core 'HomeAssistantCore.update' blocked from execution, no host internet connection
```

### System information

## System Information

version | core-2023.4.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.24
arch | x86_64
timezone | Europe/Budapest
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1333
Downloaded Repositories | 2
HACS Data | ok

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 116.7 GB
disk_used | 9.2 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.5.0), Terminal & SSH (9.7.0), Mosquitto broker (6.2.0), ESPHome (2023.3.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 0
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 2023. április 11. 09:57
-- | --
current_recorder_run | 2023. április 21. 19:38
estimated_db_size | 373.15 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

[image]

agners, 2023-04-22:
Can you share the Host logs?

Is that reproducible after a reboot still?

gya352, 2023-04-22:
Now I turned the HA computer off and on, then started the installation of the Core 4.6 update. The result is the same as before.

Host logs:

```
Apr 22 18:01:25 homeassistant kernel: audit: type=1327 audit(1682186485.245:224): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F37356532356161393061666261353031633565373036653166
Apr 22 18:01:25 homeassistant kernel: audit: type=1334 audit(1682186485.245:225): prog-id=54 op=UNLOAD
Apr 22 18:01:25 homeassistant kernel: audit: type=1334 audit(1682186485.245:226): prog-id=53 op=UNLOAD
Apr 22 18:01:25 homeassistant kernel: audit: type=1334 audit(1682186485.245:227): prog-id=55 op=LOAD
Apr 22 18:01:25 homeassistant kernel: audit: type=1300 audit(1682186485.245:227): arch=c000003e syscall=321 success=yes exit=14 a0=5 a1=c000135a50 a2=78 a3=0 items=0 ppid=3142 pid=3153 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Apr 22 18:01:29 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
Apr 22 18:01:29 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
Apr 22 18:01:30 homeassistant bluetoothd[389]: Path /org/bleak/66/140524823577984 reserved for Adv Monitor app :1.46
Apr 22 18:01:32 homeassistant systemd-resolved[360]: Using degraded feature set TCP instead of UDP for DNS server 192.168.1.254.
Apr 22 18:01:32 homeassistant systemd-resolved[360]: Using degraded feature set TCP instead of UDP for DNS server 192.168.1.254.
Apr 22 18:01:34 homeassistant dockerd[490]: time="2023-04-22T18:01:34.063607434Z" level=warning msg="[resolver] failed to read from DNS server: 192.168.1.254:53, query: ;version.home-assistant.io.\tIN\t A" error="read udp 172.30.232.2:43485->192.168.1.254:53: i/o timeout"
Apr 22 18:01:34 homeassistant dockerd[490]: time="2023-04-22T18:01:34.087428779Z" level=warning msg="[resolver] failed to read from DNS server: 192.168.1.254:53, query: ;github.com.\tIN\t AAAA" error="read udp 172.30.232.2:41584->192.168.1.254:53: i/o timeout"
Apr 22 18:01:34 homeassistant dockerd[490]: time="2023-04-22T18:01:34.087484073Z" level=warning msg="[resolver] failed to read from DNS server: 192.168.1.254:53, query: ;github.com.\tIN\t A" error="read udp 172.30.232.2:56083->192.168.1.254:53: i/o timeout"
Apr 22 18:01:36 homeassistant dockerd[490]: time="2023-04-22T18:01:36.566322700Z" level=warning msg="[resolver] failed to read from DNS server: 192.168.1.254:53, query: ;version.home-assistant.io.\tIN\t A" error="read udp 172.30.232.2:35853->192.168.1.254:53: i/o timeout"
Apr 22 18:01:42 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
Apr 22 18:01:42 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
Apr 22 18:01:46 homeassistant systemd[1]: var-lib-docker-overlay2-7dd7a734c70990194c96d957c7a67d53103e423998d965861c4c08633dfe83f1\x2dinit-merged.mount: Deactivated successfully.
Apr 22 18:01:46 homeassistant systemd[1]: mnt-data-docker-overlay2-7dd7a734c70990194c96d957c7a67d53103e423998d965861c4c08633dfe83f1\x2dinit-merged.mount: Deactivated successfully.
Apr 22 18:01:46 homeassistant kernel: hassio: port 8(vethd1588fd) entered blocking state
Apr 22 18:01:46 homeassistant kernel: hassio: port 8(vethd1588fd) entered disabled state
Apr 22 18:01:46 homeassistant kernel: device vethd1588fd entered promiscuous mode
Apr 22 18:01:46 homeassistant kernel: kauditd_printk_skb: 1 callbacks suppressed
Apr 22 18:01:47 homeassistant kernel: audit: type=1700 audit(1682186506.995:228): dev=vethd1588fd prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
Apr 22 18:01:47 homeassistant kernel: audit: type=1300 audit(1682186506.995:228): arch=c000003e syscall=44 success=yes exit=40 a0=c a1=c0000fd530 a2=28 a3=0 items=0 ppid=1 pid=490 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="dockerd" exe="/usr/bin/dockerd" subj=unconfined key=(null)
Apr 22 18:01:47 homeassistant kernel: audit: type=1327 audit(1682186506.995:228): proctitle=2F7573722F62696E2F646F636B657264002D480066643A2F2F002D2D636F6E7461696E6572643D2F72756E2F636F6E7461696E6572642F636F6E7461696E6572642E736F636B
Apr 22 18:01:46 homeassistant NetworkManager[388]: <info>  [1682186506.9982] manager: (veth56b6f32): new Veth device (/org/freedesktop/NetworkManager/Devices/21)
Apr 22 18:01:47 homeassistant NetworkManager[388]: <info>  [1682186507.0023] manager: (vethd1588fd): new Veth device (/org/freedesktop/NetworkManager/Devices/22)
Apr 22 18:01:47 homeassistant systemd[1]: Started libcontainer container 68ca711d1b92173059c6f147d3d06c4127ffdc4ebae57324ea973e379e5a10ee.
Apr 22 18:01:47 homeassistant kernel: audit: type=1334 audit(1682186507.147:229): prog-id=56 op=LOAD
Apr 22 18:01:47 homeassistant kernel: audit: type=1334 audit(1682186507.148:230): prog-id=57 op=LOAD
Apr 22 18:01:47 homeassistant kernel: audit: type=1300 audit(1682186507.148:230): arch=c000003e syscall=321 success=yes exit=15 a0=5 a1=c0001ad7f8 a2=78 a3=0 items=0 ppid=3388 pid=3398 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe [... cut ...]
Apr 22 18:03:23 homeassistant systemd-resolved[360]: Using degraded feature set TCP instead of UDP for DNS server 192.168.1.254.
Apr 22 18:03:23 homeassistant systemd-resolved[360]: Using degraded feature set TCP instead of UDP for DNS server 192.168.1.254.
Apr 22 18:03:32 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
Apr 22 18:03:32 homeassistant systemd-resolved[360]: Using degraded feature set UDP instead of TCP for DNS server 192.168.1.254.
```

agners, 2023-04-23:
What type of DNS server is 192.168.1.254?

gya352, 2023-04-24:
ADSL router from internet provider. ZTE Speedport Entry 2i. It hasn't changed, it's always been like that.

agners, 2023-04-24:
This is most likely some unfortunate interaction with your local DNS server and the systemd-resolved resolver HAOS is using.

In PR #2506 I've bumped to the latest version, this might address the problem. I'll backport the change to HAOS 10.1, let's see.

agners, 2023-04-24:
Some providers block Google DNS. But generally it should work with HAOS.

I assume with 192.168.1.1 things are working fine then, and this can be closed?

closed 2023-04-25 by agners