home-assistant/operating-system #2500: Raspberry PI 4 cannot boot from SD after upgrade to 10

BastiOfBerlin, 2023-04-21:
### Describe the issue you are experiencing

After upgrading to HAOS 10, the system won't boot up anymore. Unlike several other issues already opened, I do not use any NVMe or USB-attached SSDs, but instead the good old micro SD card to boot from.

I hava already tried to update the bootloader as suggested in other issues, without any effect. After attaching a monitor, I only get the follwing 5 lines of output:

```
Net:    eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
PCI: Failed autoconfig bar 10
starting USB...
Bus xhci_pci:
```

Then it stops and nothing more. Consequently, I am not able to downgrade manually. The automatic restoral doesn't work either as no OS is being loaded.

For now, I'll flash 9.5 on a spare SD and restore a backup.

Would be nice if you had any idea on how to fix this.

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 10.0 via UI
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System information

_Note that this information was collected **after** the backup restore on the same board with a fresh SD card!_

## System Information

version | core-2023.4.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.84-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 57.8 GB
disk_used | 5.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Check Home Assistant configuration (3.11.0), Cloudflared (4.1.2), ESPHome (2023.3.2), File editor (5.5.0), Let's Encrypt (4.12.8), Nextcloud Backup (0.17.7), NGINX Home Assistant SSL proxy (3.4.0), Samba share (10.0.0), Terminal & SSH (9.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 7
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 7. April 2023 um 12:49
-- | --
current_recorder_run | 21. April 2023 um 21:46
estimated_db_size | 685.29 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

ubkr, 2023-04-22:
I've a similar situation. Running from a USB attached SSD and after performing the upgrade via the UI my RPi4 4Gb ends up in the same state as above. I've updated the RPi4 boot loader but no change.
  
U-boot starts but can't initialize the USB chip. It then continues to try network booting but will eventually stop and present a U-boot prompt that I can't interact with due to the fact that USB is not present. 

Log from boot (OCR by google photos and then edited by me, some errors from OCR might still be there...) : 

Net: eth0: ethernet@7d580000
PCIe BRCM: Link up, 5.0 Gbps x1 (SSC)
PCI: Failed autoconfig bar 10
starting USB...
Bus xhcl pci: pci hose_bus_to_phys: invalid physical address 
xhci-pci init cannot map PCI men bar
probe failed, error -5 
No working controllers found 
Card did not respond to voltage select?: -110 
Card did not respond to voltage select!: -110
MMC Device 2 not found
no mmc device at slot 2

Device 0: unknown device
starting USB... 
Bus xhci-pci: pci hose_bus_to_phys: invalid physical address 
xhci-pci init cannot map PCI Men bar 
probe failed, error -5 
No working controllers found
USB is stopped. Please issue 'usb start' first. 
starting USB...
Bus xhci_pci: pci hose_bus_to_phys: invalid physical address 
xhci-pci init cannot map PCI men bar 
probe failed, error -5
No working controllers found
ethernet@74580000 Waiting for PHY auto negotiation to complete...... TIMEOUT !
bcmgenet: PHY startup failed: -110
missing environment variable: pxeuuid
Retrieving file: pxelinux.cfg/01-dc-a6-32-03-41-c0
ethernet@74580000 Waiting for PHY auto negotiation to complete......... TIMEOUT bengenet: PHY startup failed: -110
Retrieving file: pxelinux.cfg/00000000
ethernet@7d580000 Waiting for PHY auto negotiation to complete....

agners, 2023-04-24:
@ubkr your problem is more likely related to #2482. The boot log is interesting though, and could help to investigate that further. Can you re-post it in #2482? Also, do you use the 32-bit or 64-bit image?

BastiOfBerlin, 2023-04-24:
> @BastiOfBerlin do you have anything connected on any of the USB ports? If so, does the system boot if you plug everything out? And if so, what do you have connected exactly?

TL;DR: no, nothing attached at all, except for eth network.

Long version:
No, originally, the RP4 was running with only power and network attached. While testing with a monitor attached, I also connected USB keyboard and mouse via a hub, which didn't change a thing. I've also tried with a monitor, but without anything connected to USB.

The Raspi is enclosed in a [Coolermaster case](https://www.coolermaster.com/de/de-de/catalog/gehause/raspberry-pi/pi-case-40-v2/#overview) (original Kickstarter version). AFAIK this only adds better accessible GPIOs and this button should be connected via GPIO as well.

agners, 2023-04-24:
I've been able to reproduce the problem on my Raspberry Pi 4 8GB with 32-bit U-Boot. PR #2505 addresses the problem, I'll backpart the PR and make it part of the upcomming 10.1 release.

closed 2023-04-24 by agners

ubkr, 2023-04-24:
> @ubkr your problem is more likely related to #2482. The boot log is interesting though, and could help to investigate that further. Can you re-post it in #2482? Also, do you use the 32-bit or 64-bit image?

@agners quit sure it was due to the problem you found and fixed in PR #2505 since I for some reason run with a 32-bit U-Boot. Did some testing tonight and was just about to report when I saw your fix. I'll be happy to try it out when there is a build. Thank you!!

agners, 2023-04-25:
You can find `rpi4` builds of HAOS 10.1 with the fix applied here:
https://github.com/home-assistant/operating-system/releases/tag/10.1

BastiOfBerlin, 2023-04-29:
I can confirm it works now: Flashed 10.1 on the old SD card and restored a newly created backup. Everything's running again. Thanks a lot!