home-assistant/operating-system #2501: Mosquitto add-on v2 can't run on v10 of HA OS

26tajeen, 2023-04-22:
### Describe the issue you are experiencing

It says "error: out of memory." and "mosquitto vesion 1.4.15 terminating". I know I said version 2, but I think that's version 2 of the add-on as different from the actual version of mosquitto the add-on is running 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

5.15.84-v7

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install v10
2. Try and use Mosquitto broker
3. Find it won't work due to above issue
...

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### System information

.

### Additional information

.

agners, 2023-04-22:
How much memory does your system have?

26tajeen, 2023-04-22:
Looking at the [hardware page](http://ha:8123/config/hardware) I have 1Gb of memory and only 50% is being used. But this is on the working core OS. It's an rpi3

agners, 2023-04-24:
I can't reproduce this on a Raspberry Pi 3 64-bit installation.

Can you provide more details please, mainly also the System information as per original issue form:

Optional Copy the full System Health in this text area.
System information can be found in [Settings -> System -> Repairs -> System Information (top right menu)](https://my.home-assistant.io/redirect/system_health/), Click the copy button at the bottom of the pop-up and paste it here.

26tajeen, 2023-04-27:
## System Information

version | core-2023.3.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.84-v7
arch | armv7l
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.31.0
Stage | running
Available Repositories | 1278
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 27.7 GB
disk_used | 7.4 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.1), File editor (5.6.0), Mosquitto broker (2.0), RPC Shutdown (2.2), Terminal & SSH (9.7.0), ZeroTier One (0.16.2), Zigbee2mqtt (1.18.1-1), SMTP to MQTT (1.0.3), Duck DNS (1.15.0), NGINX Home Assistant SSL proxy (3.4.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 1
views | 3
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24 April 2023 at 13:19
-- | --
current_recorder_run | 24 April 2023 at 14:19
estimated_db_size | 64.65 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

Grantsauto, 2023-04-28:
I am having the same issue with MQTT after updating to 10, I deleted and reinstalled MQTT add on with no luck. Rolled back to 9.5. Logs state that listening websockets on port 1884 can't be opened.

Version - core-2023.5.0b1
Host Operating System | Home Assistant OS 9.5
Update Channel | beta
Supervisor Version | supervisor-2023.04.1
Rpi4-64

Grantsauto, 2023-04-28:
Mosquito Broker is 5.1

Grantsauto, 2023-04-28:
I updated the broker to 6.2.1 and all is good after reconfiguring my setup and continuing with OS 10

closed 2023-10-27 by github-actions[bot]