home-assistant/operating-system #2502: Cannot load HASSOS 10 (upgrade from 9.5) - System hangs on boot

ggian311, 2023-04-22:
### Describe the issue you are experiencing

When trying to update to v.10 the system hangs on boot.
After twinkling with it a bit, I found out that the USB-connected devices (a Skyconnect and a Z-wave stick, were causing the boot process to get stuck in random places.

After removing them, the boot process gets stuck always in the same spot, right after the sound card identification, hanging on _"Wait until kernel time synchronized"_

I've tried:
- updating time via `date -s`
- restarting router
- change eth cable
- change eth port
- disable IPv6

Running on a laptop, intel CPU, 16gb Ram, sound card integrated on motherboard, static IP address on IPv4

The following logs are resulting from `ha host logs -b -1 -n 10000`

[](https://pastebin.com/1ByFLnbV)

There is something weird: the logs show 11:43 but everything was rebooted at 14:21

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

I simply launched the update from the UI. 
I am now back on 9.5 which is working without issue

### Anything in the Supervisor logs that might be useful for us?

```txt
2023-04-22 14:10:30.676 ERROR (bellows.thread_0) [bellows.uart] CRC error in frame b'099b7b7e' (b'9b7b' != b'70d9')
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: bellows.uart
Source: /usr/local/lib/python3.10/site-packages/bellows/uart.py:92
First occurred: 14:10:30 (1 occurrences)
Last logged: 14:10:30

CRC error in frame b'099b7b7e' (b'9b7b' != b'70d9')
```

### System information

## System Information

version | core-2023.4.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4799
Installed Version | 1.32.1
Stage | running
Available Repositories | 1339
Downloaded Repositories | 31

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.0
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 468.7 GB
disk_used | 16.3 GB
healthy | true
supported | failed to load: Unsupported
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.110.3), File editor (5.5.0), Samba share (10.0.0), SSH & Web Terminal (13.1.0), Cloudflared (4.1.2), Node-RED (14.1.4), Mosquitto broker (6.2.0), A better presence (1.2.0), Plex Media Server (3.2.4), Z-Wave JS (0.1.77), Studio Code Server (5.5.7), Portainer (2023.1.0), Grocy (0.19.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 8
-- | --
resources | 21
views | 30
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 14, 2023 at 10:28
-- | --
current_recorder_run | April 22, 2023 at 14:10
estimated_db_size | 411.00 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

### Additional information

The one thing that jumps up from the System Info is that Supervisor is unsupported but I have no idea why.

agners, 2023-04-22:
Can you attach the complete log from `ha host logs -b -1 -n 10000`?

ggian311, 2023-04-24:
@agners I've been trying to copy the full output but I'm getting stuck. Using the [SSH & Web Terminal addon](https://github.com/hassio-addons/addon-ssh). Do you have any suggestions, please?

agners, 2023-04-24:
If you have access to the config directory you can redirect the output to a file:

```
ha host logs -b -1 -n 10000 > /config/host.log
```

ggian311, 2023-04-24:
Thank you @agners I've updated the [pastebin](https://pastebin.com/1ByFLnbV) with the full output

Deinagkistrodon, 2023-04-24:
I believe to have the same problem on my RPI3. Network is not starting during boot, hence the Realtek error in your log.

agners, 2023-04-24:
This seems to be a boot log of Home Assitant OS 9.5 (the first lines says "homeassistant kernel: Linux version 5.15.90", HAOS 10 uses Linux 6.1).

Maybe you did a reboot meanwhile? You can use earlier boots by lowering the -b argument to -2, -3 ..., you can check the first lines by using head, e.g.

```
ha host logs -b -2 -n 10000 | head
```

ggian311, 2023-04-25:
That's weird, it goes back a maximum of 6 boots logs but none of them have the 6.1 kernel. I will try again today to run the upgrade an see immediately after which results I get in the log

ggian311, 2023-04-25:
well, something is off or perhaps a setting is missing that I am not aware of. I just tried to upgrade again to 10, got the usual error (also waited for some 30 minutes on it), rebooted into 9.5, and got the log info. Except, the log shows nothing from today... <scratch_head></scratch_head>

[image]

agners, 2023-04-25:
It does automatically reboot after the update correct?

My best bet is that the system crashes early at reboot and then falls back to the previous installation.

Can you connect a monitor and see what is happening after the update?

bboudewijns, 2023-04-25:
Following this. My system hangs as well since the OS 10 update. Spent many hours today to get it back up and running. I am running HASSOS on a Intel NUC. 9.5 no issues, 10.0 many boot loops and instable behavior.

ggian311, 2023-04-25:
@agners correct: it updates and then reboots. And then crashes/hangs on network load. 
I need to physically long press the power button for it to restart. At which point it tries 2 more times loading v.10, hangs, I force restart, and eventually defaults back to 9.5

agners, 2023-04-25:
Hm, the stack trace just above of this would be interesting.

If we are lucky, this is resolved with the newer Linux 6.1.25 kernel which will be part of HAOS 10.1. HAOS 10.1 will soon be available on the beta channel.

piechade, 2023-05-04:
> Hm, the stack trace just above of this would be interesting.
> 
> If we are lucky, this is resolved with the newer Linux 6.1.25 kernel which will be part of HAOS 10.1. HAOS 10.1 will soon be available on the beta channel.

I'm on Operating System 10.1 without beta channel.

closed 2023-05-31 by agners