home-assistant/operating-system #2513: CPU Utilization is increased with Operating System 10.1

pimw1, 2023-04-25:
### Describe the issue you are experiencing

The CPU utlization is increased after updating from Operating System 10.0 to 10.1
Observations:
-Average increase is 2 percent point. 
-Spikes are visible to 18% which did not happen with 10.0

[image]

I'm running on a RPI4, 64 bit.
Other info:
Supervisor 2023.04.1
Home Assistant 2023.4.6

I can share some profiler callgrind.out file in private , if that would help the analysis.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to settings -> system -> hardware and track the cpu utliztion.
2. The observed cpu utilization is higher in 10.1 than in 10.0

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2023-04-26:
Can you check with Glances or on the terminal using top which process is causing high CPU load?

This is most likely a the same issue as reported in #2476.

agners, 2023-04-27:
Unfortunately #2476 is a mixed bag of issues, some of the reports spike the CPU to 100% due to add-on issues. But this is not the report of the original poster: In that case it was also a CPU increase of a few procent points, caused by containerd.

That said, indeed that issue was about HAOS 10.0.

It is surprising that 10.1 changes this for you, there are not many changes in 10.1 (see https://github.com/home-assistant/operating-system/releases/tag/10.1). Are you sure it is not some other component which changed? Can you try downgrading again to see if this indeed helps? Use `ha os update --version 10.0` to downgrade.

pimw1, 2023-04-27:
I've downgraded to 10.0, did some measurements and upgraged back to 10.1. See below the results, from left to right:
10.1 (first time) -> 10.0 (downgrade) -> 10.1 (upgraded again)
[image]

WouterJN, 2023-05-03:
I'm running on a Proxmox server and after upgrading from 9.x to 10.1 i have the same problem;

[image]

yildiraymeric, 2023-05-16:
Yes I was having issues with bluetooth and was downgraded toı 9.5.. Then new update happended and switched back to 10.1 and then I realized that my device is using more power now (I've installed a power metered smart plug).. Here are the screen shots of the power usage difference;

[image]

There is approximately 2watts of power usage difference..

pimw1, 2023-06-16:
Hi, the upgrade to OS 10.3 resolved the problem for me. CPU utilization went down by 2 percent point on my RPI4. All looking good now.

pimw1, 2023-06-16:
Great to hear! Do others still have issues with 10.3? If not, i'll close the issue.

WouterJN, 2023-06-16:
@pimw1 For me the issue is not resolved, I stil have double the CPU utilisation after upgrading from 9.5 to 10.3 Beta (from 5% to 10%+).

closed 2023-09-22 by github-actions[bot]