home-assistant/operating-system #2519: ZHA broke after 10.1 upgrade

sebiweise, 2023-04-27:
### Describe the issue you are experiencing

Zigbee devices not working after upgrade.
Downgrading from 10.1 to 9.5 via ha os update --version 9.5 doesnßt fix the issue.
Also flashing the zigbee dongle doesnßt help to bring the ZHA integration back up.

Restarting HA or the rpi is working, however the ZHA integration is loading forever and then just crashing.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade from OS 9.5 to 10.1

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.config_entries
Source: components/zha/core/gateway.py:205
First occurred: 7:56:09 PM (1 occurrences)
Last logged: 7:56:09 PM

Error setting up entry Sonoff Zigbee 3.0 USB Dongle Plus, s/n: f0631140341aec11a3ce77e5f01c6278 - ITead for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 383, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 122, in async_setup_entry
    await zha_gateway.async_initialize()
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 205, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.10/site-packages/zigpy/application.py", line 211, in new
    await app.ota.initialize()
  File "/usr/local/lib/python3.10/site-packages/zigpy/ota/__init__.py", line 126, in initialize
    await self.async_event("initialize_provider", self._app.config[CONF_OTA])
  File "/usr/local/lib/python3.10/site-packages/zigpy/util.py", line 78, in async_event
    for result in await asyncio.gather(*tasks, return_exceptions=True):
asyncio.exceptions.CancelledError
```

### Anything in the Host logs that might be useful for us?

```txt
No error or warning found
```

### System information

## System Information

version | core-2023.4.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 5.15.84-v8
arch | aarch64
timezone | Europe/Berlin
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | failed to load: timeout
GitHub Web | ok
GitHub API Calls Remaining | 4645
Installed Version | 1.32.1
Stage | running
Available Repositories | 1278
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.1
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 109.3 GB
disk_used | 8.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | failed to load: timeout
installed_addons | File editor (5.6.0), MariaDB (2.5.2), Samba share (10.0.1), Cloudflared (4.1.4), ESPHome (2023.4.2), Terminal & SSH (9.7.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 10
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | April 18, 2023 at 5:29 AM
-- | --
current_recorder_run | April 27, 2023 at 7:51 PM
estimated_db_size | 602.98 MiB
database_engine | mysql
database_version | 10.6.10

</details>

lancer73, 2023-04-27:
I had this too, but with a SkyConnect dongle (doing only Zigbee, not thread).

10.0 worked 10.1 didn't. As I was in a hurry I rolled back to 10.0. Only tried restarting HA after re plugging the device.

Error generated looked similar.

Will try again in the weekend to get more debugging logs.

Krazy998, 2023-04-27:
Not sure if same issue. However with 10.1 I noticed I had to unplug and replug the zigbee USB for Zigbee2MQTT to be able to detect and use it. 

I attempted restarting the VM from proxmox, I am unable to reproduce the same behaviour. So far it seems to be work OK after a 1 off unplug and replug...

agners, 2023-04-28:
> Downgrading from 10.1 to 9.5 via ha os update --version 9.5 doesnßt fix the issue.

Hm, so it is unlikely that it is actually HAOS's fault then.

What could be is that your Zigbee dongle crashed hard. A reboot might not be enough. Did you try to remove your dongle or completely power off your system and start again.

sebiweise, 2023-04-28:
Already tried that. Also flashing the zigbee dongle doesn´t help to bring the ZHA integration back up.

agners, 2023-04-28:
How do you flash the dongle? Can you try flashing with the Silicon Labs Flasher add-on? Can you also make sure that you disabled all add-ons which might access your Zigbee device? (E.g. Multiprotocol/Z2M test installations or similar).

What version of Home Assistant Core are you using? 

Still, seems very unlikely that this is directly related to the HAOS update, this is most likely some conincidence. Maybe the Zigbee dongle just died right there, or some other update broke communication with that device. We have 10k's of updates which don't seem to have a problem with ZHA/Zigbee dongles.

sebiweise, 2023-04-28:
So after some restarts Zigbee is working now. But just after reloading the zha integrations after it fails. I will try flashing the zigbee dongle again with the Silicon Labs Flasher AddOn

sebiweise, 2023-04-28:
```txt
[12:24:44] INFO: Checking /dev/ttyUSB0 identifying Sonoff Zigbee 3.0 USB Dongle Plus from ITead.
[12:24:44] WARNING: No firmware found for the selected device, assuming firmware is installed.
[12:24:44] INFO: universal-silabs-flasher-up script exited with code 0
```

agners, 2023-04-28:
> So after some restarts Zigbee is working now. But just after reloading the zha integrations after it fails. I will try flashing the zigbee dongle again with the Silicon Labs Flasher AddOn

That very much sounds like two things accessing the same device (e.g. some add-on). Make always sure that only one thing is accessing the serial port. E.g. when flashing using the add-on make sure to disable/remove ZHA.

> [12:24:44] WARNING: No firmware found for the selected device, assuming firmware is installed.

Yeah you need to pass the URL to a firmware in the configuration.

Since Zigbee is working again I assume this is not OS related and closing this issue.

closed 2023-04-28 by agners

lancer73, 2023-04-28:
Tried it again today anticipating a full debug run and now it just works.... :-)