home-assistant/operating-system #2520: Raspberry PI 3 not booting since I upgraded to HAOS 10.1

bossun82, 2023-04-28:
### Describe the issue you are experiencing

Hi, 
Yesterday, I upgraded the HA OS to version 10.1. 

This morning I noticed that my Raspberry PI 3 does not boot any more. I tried to connect my hdmi screen and usb keyboard but absolutely nothing happens. Only the power lights are working.

I removed the SD card to analyse it. I can see the boot partition but there is a non allocated partition !! I Hope that there is no issue with that and I can recover my last setup

[image]

My SD Card is quite new < 6months

Can someone help please?

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Can not provide any log
```

### Anything in the Host logs that might be useful for us?

```txt
Can not provide any log
```

agners, 2023-04-28:
I've upgraded two Raspberry Pi 3 instances to HAOS 10.1 successfully here. According to analytics we have about 20k updates to HAOS 10.1, and since we have ~10% Raspberry Pi 3 installations there are many Raspberry Pi 3 updates which worked.

So this looks like an isolated incident. My best guess is that the SD-card simply died :sob: . It is not uncommon that this happens when flash cells get written which hasn't been written for a longer period of time (e.g. the read-only OS partitions).

Ideally you use your last backup and setup a new installation. If you don't have one, you might be able to restore the partition table using `fdisk` or similar tools, but it is a hassle, and this is not really the place to discuss data recovery.

nikakoy131, 2023-05-10:
I have the same issue on my Raspberry Pi 3. Fresh installation and restoring from backup helped me.
I'm not sure, but it seems my SD card is ok, new installation working fine for now. The data partition was ok, so I mounted it on my PC and was able to get the last backup

nikakoy131, 2023-06-02:
same situation while upgrading from 10.1 to 10.2

closed 2023-09-08 by github-actions[bot]