home-assistant/operating-system #2521: Upgrade to 10.1 - Home Assistant Operating System update failed

djnas123, 2023-04-28:
### Describe the issue you are experiencing

When attempting an upgrade from HA OS 10.0 to 10.1 the installation starts and fails, when I look in the supervisor logs I see this message:

 Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Navigate to Settings
2. Click on Upgrade box, click Install
3. AFter around 30 seconds, the screen refreshes and nothing has happened. 

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-28 17:35:44 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/10.1/haos_rpi4-64-10.1.raucb
23-04-28 17:35:52 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-10.1.raucb
23-04-28 17:35:57 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.core
Source: components/hassio/update.py:214
First occurred: 09:03:52 (3 occurrences)
Last logged: 17:35:57

Error executing service: <ServiceCall update.install (c:01GZ3E77D33D0M05AZPKQWS8SV): entity_id=['update.home_assistant_operating_system_update'], version=10.1, backup=False>
Error executing service: <ServiceCall update.install (c:01GZ3J0GJGT9WM03KSTXDE4GVP): entity_id=['update.home_assistant_operating_system_update'], version=10.1, backup=False>
Error executing service: <ServiceCall update.install (c:01GZ4BGWS2KV3T449FS10R0FE4): entity_id=['update.home_assistant_operating_system_update'], version=10.1, backup=False>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 212, in async_install
    await async_update_os(self.hass, version)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 51, in _wrapper
    raise HassioAPIError(data["message"])
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1838, in catch_exceptions
    await coro_or_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1857, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 216, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 798, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 977, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 838, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 156, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 417, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 214, in async_install
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Error updating Home Assistant Operating System: Unknown error, see supervisor
```

### System information

## System Information

version | core-2023.4.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.10
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/London
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1273
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.1
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 219.4 GB
disk_used | 19.8 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | AdGuard Home (4.8.6), Cloudflared (4.1.4), File editor (5.6.0), Home Assistant Google Drive Backup (0.110.3), Matter Server (4.3.1), Mosquitto broker (6.2.1), Plex Media Server (3.2.4), Samba share (10.0.1), Terminal & SSH (9.7.0), Z-Wave JS (0.1.79), GivTCP (2.1.7)

</details>

<details><summary>Dashboards</summary>

dashboards | 6
-- | --
resources | 5
views | 8
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 18 April 2023 at 06:25
-- | --
current_recorder_run | 28 April 2023 at 08:45
estimated_db_size | 217.94 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
remaining_requests | Exceeded API Allowance
rooftop_site_count | 1
polling_hours | 6 -> 19 and once at midnight

</details>

kmmccafferty96, 2023-05-04:
Seeing this as well

khacduy1978, 2023-05-06:
Same to me on Odroid XU4. I can not update from Hass OS 9.5 to 10.0 and 10.1. Thank you.

Here are Supervisor logs:
23-05-06 21:36:34 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-10.1.raucb
23-05-06 21:36:49 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-kernel0' not found

patvdleer, 2023-05-12:
@khacduy1978 please be aware of https://github.com/home-assistant/operating-system/issues/2538

miguelebf, 2023-05-16:
i have same issue with raspberry pi 3

agners, 2023-05-16:
```
error: Failed updating slot kernel.0: Opening output device /dev/disk/by-partlabel/hassos-kernel0 failed: Device or resource busy
```

Usually this indicates the rauc wasn't able to detect which system it booted from. It is a bit unclear to me why only few people see this issue, I am assuming it is related to some boot files corruption or similar.

Since the system is working at that point, probably the best way is to take a full backup, and do a reinstall. If the installation is on a SD card, maybe replacing it is a good idea since this can be caused by corruption.

djnas123, 2023-05-16:
Hey, following on from the suggestion, I boot from SSD, but had the same issue upgrading from 9.5 to 10. The workaround I performed then was create a back up, purchase and set up a new SSD. I used to boot from SD and that seemed to upgrade fine.

ruudk, 2023-06-08:
Same issue here. I run Raspberry Pi CM 4 and boot from an USB SSD. Worked fine, until I tried to upgrade. It doesn't work whatever I try.

ruudk, 2023-06-08:
> The workaround I performed then was create a back up, purchase and set up a new SSD.

@djnas123 Doing a full install will probably install the latest version? Did you try an upgrade since then? Did that work?

miguelebf, 2023-06-08:
> i have same issue with raspberry pi 3

I solved my problem by updating HACS. I had a problem with HACS because I tried to update automatically and it didn't work I did it manually and it worked after that I updated home assistant and there was no problem.

ruudk, 2023-06-09:
FWIW, I don't have HACS.

closed 2023-09-15 by github-actions[bot]

matti125, 2023-10-08:
I’ve got the same issue, and I am booting an RPi4 from SSD