home-assistant/operating-system #2522: AppArmour error after attempting upgrade to 10.1

ABredhauer, 2023-04-29:
### Describe the issue you are experiencing

After attempting to upgrade to 10.1 I'm now getting an error about running in an unsupported mode

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Attempt to upgrade from 9.5 to 10.1
2. System restarts
3. upgrade not succesful and still running on 9.5
4. Unsupported System - AppArmour issues notification appears
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-04-29 10:11:33 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
23-04-29 10:11:35 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1 - 172.30.32.3
23-04-29 10:11:35 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
23-04-29 10:11:35 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.04.1
23-04-29 10:11:35 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
23-04-29 10:11:36 INFO (SyncWorker_1) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.04.1 - 172.30.32.4
23-04-29 10:11:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
23-04-29 10:11:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
23-04-29 10:11:36 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
23-04-29 10:11:37 INFO (SyncWorker_0) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0 - Host
23-04-29 10:11:37 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-04-29 10:11:37 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 11 Home Assistant secrets
23-04-29 10:11:37 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi4-64-homeassistant with version 2023.4.6
23-04-29 10:11:37 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 9.5 / BootSlot A
23-04-29 10:11:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/df2164f9 repository
23-04-29 10:11:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
23-04-29 10:11:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
23-04-29 10:11:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository
23-04-29 10:11:38 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/15d21743 repository
23-04-29 10:11:39 INFO (MainThread) [supervisor.store] Loading add-ons from store: 74 all - 74 new - 0 remove
23-04-29 10:11:39 INFO (MainThread) [supervisor.addons] Found 8 installed add-ons
23-04-29 10:11:39 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.7.0
23-04-29 10:11:39 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.2.1
23-04-29 10:11:39 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to ghcr.io/funkeyflo/ps5-mqtt/aarch64 with version 1.3.1
23-04-29 10:11:39 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.6.0
23-04-29 10:11:39 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/hassio-addons/sqlite-web/aarch64 with version 3.7.1
23-04-29 10:11:39 INFO (SyncWorker_7) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-samba with version 10.0.1
23-04-29 10:11:39 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to thomasmauerer/samba-backup-aarch64 with version 5.2.0
23-04-29 10:11:39 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to ghcr.io/esphome/esphome-hassio with version 2023.4.2
23-04-29 10:11:39 INFO (MainThread) [supervisor.backups.manager] Found 5 backup files
23-04-29 10:11:40 INFO (MainThread) [supervisor.discovery] Loaded 2 messages
23-04-29 10:11:40 INFO (MainThread) [supervisor.ingress] Loaded 5 ingress sessions
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.SETUP
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-04-29 10:11:40 INFO (MainThread) [supervisor.jobs] 'ResolutionFixup.run_autofix' blocked from execution, system is not running - CoreState.SETUP
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.SETUP
23-04-29 10:11:40 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-04-29 10:11:40 INFO (MainThread) [__main__] Running Supervisor
23-04-29 10:11:40 WARNING (MainThread) [supervisor.core] System running in a unsupported environment!
23-04-29 10:11:40 INFO (MainThread) [supervisor.os.manager] Rauc: A - marked slot kernel.0 as good
23-04-29 10:11:40 INFO (MainThread) [supervisor.addons] Phase 'AddonStartup.INITIALIZE' starting 0 add-ons
[... 236 lines cut ...]

oldest_recorder_run | April 21, 2023 at 12:53 AM
-- | --
current_recorder_run | April 29, 2023 at 10:12 AM
estimated_db_size | 1390.65 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

<details><summary>Solcast PV Forecast</summary>

can_reach_server | ok
-- | --
remaining_requests | 2
rooftop_site_count | 2
polling_hours | 6 -> 18 and once at midnight

</details>

### Additional information

Unsupported system - AppArmor issues

System is unsupported because AppArmor is working incorrectly and add-ons are running in an unprotected and insecure way. Use the link to learn more and how to fix this.

Saxophtrion, 2023-05-11:
Hi,

Did you resolve the issue ? I have the same setup and exactly the same problem ... I don't have any solution yet ...

agners, 2023-05-11:
The question is why did the upgrade fail in first place. Can you check the host logs after an upgrade attempt? You should be able to see them in full using `ha host logs -b -1`,

Saxophtrion, 2023-05-11:
I have news, i don't know if it's related or not but it eventually worked :

When i was trying to update, i was refreshing the page after the lost connection (5 or 10 sec after the update click) --> 20 sec later, the system was rebooting and back to 9.5.

This morning i tried to just click on the update button and close the GUI. I came back 10 after to check, it was good, booted in 10.1.
I don't know if the GUI refresh can affect an update, but it seems to be the only change between the fail and the success ...

agners, 2023-05-11:
I think it is unlikely that the Frontend contributed here, but maybe just the second attempt made it go through?

Saxophtrion, 2023-05-11:
Don't think so, i was trying all the week ... Should be my 10th attempt i think ... Absolutely no reason to work now and not 3 days ago ...

ABredhauer, 2023-05-11:
I've definitely had many more than 2 attempts and haven't been successful yet. Sometimes when HA restarts it has kicked me back to 9.4 which I don't understand at all since I started on 9.5 and tried to upgrade to 10.1. Will try again shortly and get the logs using the command above

ABredhauer, 2023-05-11:
Update failed again still stuck on 9.5 output of ha host logs -b -1
[core-ssh ~]$ ha host logs -b -1
May 10 23:02:46 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.thWxeh.mount: Deactivated successfully.
May 10 23:03:14 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.nkoy0y.mount: Deactivated successfully.
May 10 23:04:16 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.sICUWk.mount: Deactivated successfully.
May 10 23:04:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.CJJSgb.mount: Deactivated successfully.
May 10 23:04:46 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.lRRszr.mount: Deactivated successfully.
May 10 23:04:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.nzzljj.mount: Deactivated successfully.
May 10 23:05:44 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.czopYk.mount: Deactivated successfully.
May 10 23:06:18 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.pxutcV.mount: Deactivated successfully.
May 10 23:06:44 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.nOFNkh.mount: Deactivated successfully.
May 10 23:06:49 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.aiJBJk.mount: Deactivated successfully.
May 10 23:07:14 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.AgaZon.mount: Deactivated successfully.
May 10 23:08:17 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.dba4o9.mount: Deactivated successfully.
May 10 23:08:50 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.tnQSFN.mount: Deactivated successfully.
May 10 23:09:45 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.d0Cu1D.mount: Deactivated successfully.
May 10 23:09:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.Ij5auj.mount: Deactivated successfully.
May 10 23:09:50 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.lLoSB4.mount: Deactivated successfully.
May 10 23:10:21 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.chcWAa.mount: Deactivated successfully.
May 10 23:10:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.RUkokL.mount: Deactivated successfully.
May 10 23:11:15 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.ksvr3u.mount: Deactivated successfully.
May 10 23:11:45 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.v8OCMe.mount: Deactivated successfully.
May 10 23:11:47 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.BONJrl.mount: Deactivated successfully.
May 10 23:12:15 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.fXdJ11.mount: Deactivated successfully.
May 10 23:12:22 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.NmARZz.mount: Deactivated successfully.
May 10 23:12:48 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.3faaz6.mount: Deactivated successfully.
May 10 23:13:18 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.qgReB8.mount: Deactivated successfully.
May 10 23:14:23 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-0a7a22b5f66eafafb2ad9e9b352747b85871f64b8f26f11fbe16039686262055-runc.q37pPF.mount: Deactivated successfully.
May 10 23:15:46 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.zUip56.mount: Deactivated successfully.
May 10 23:15:48 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.vl8wx0.mount: Deactivated successfully.
May 10 23:16:16 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.zy7MbL.mount: Deactivated successfully.
[... cut ...]
May 10 23:41:54 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-da292bc0663f95eedac6bd3bda9d18be2862f5df6317b0bf2b2f8ee23b9c3d63-runc.ZMit9F.mount: Deactivated successfully.
May 10 23:42:22 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.FILx0B.mount: Deactivated successfully.
May 10 23:43:22 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-ba1019494fa1cf8cef8cfdb8306ff94fd1fd45e450fcfd4d8857015c6e340eb1-runc.NhGLHh.mount: Deactivated successfully.

ABredhauer, 2023-05-18:
I tried taking a backup and flashing my SSD with 10.1. As soon as I restored the backup I'm back to 9.5? I wasnt expecting that

ABredhauer, 2023-06-11:
Is anyone able to help debug this or give me some commands to run to produce some logs I can share to help diagnose? The issue is still occurring so I'm not able to upgrade beyond 9.5. I've t2023.6ried updating to 10.2 and it fails in the same way. I've also attempted again after updating to HA 2023.6 but same behaviour. Occasionally I seem to get sent back to HAos 9.4 but can ssh in and run the ha os update --version 9.5 command to get back to 9.5

doming50, 2023-07-06:
I have had this same issue for months trying to update from HAOS 8.5.

Any light on what the cause and possible solutions would be greatly appreciated.

ABredhauer, 2023-07-06:
I have not made any progress on this either. I've turned on SSH access and have looked at logs before and after attempting the upgrade but have not noticed anything that looks like an issue but also don't have a lot of knowledge of what I'm looking at either. The current loop i am stuck in is HA OS 9.5 -> attempt upgrade to 10.3 -> reboot and now i'm on 9.4 -> upgrade to 9.5 -> reboot success I'm on 9.5 -> attempt upgrade to 10.3 -> reboot and i'm back to 9.4

Happy to provide logs or run commands and provide output but dont have the skills to debug this myself

closed 2023-10-12 by github-actions[bot]