home-assistant/operating-system #2523: OS Upgrade from 10.0 to 10.1 bricked HA

endelight, 2023-04-29:
### Describe the issue you are experiencing

Since two months HA was running fine. Today I did an upgrade using "ha os upgrade". During next boot it failed to start with the message:
[image]

Tried to reboot multiple times, always same. Also no success even waiting 20 min.

Docker shows this:
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 10.1 using "ha os upgrade"

### Anything in the Supervisor logs that might be useful for us?

```txt
No access to web frontend. I have no idea how to access those logs on the emergency command line.a
```

### Anything in the Host logs that might be useful for us?

```txt
No access to web frontend. I have no idea how to access those logs on the emergency command line.a
```

### System information

Running on VirtualBox 6 on Mac mini.

### Additional information

VM has 4 GB RAM, 32 GB disk size, 2 CPU.

I would love to provide more info, but have no knowledge regarding docker or about the HA architecture. I also could not find any useful logs on the emergency command line except tons of docker overlay2 logs. It is scaring that such an update can damage the whole system.

agners, 2023-04-29:
> except tons of docker overlay2 logs. 

What kind of logs are those?

> It is scaring that such an update can damage the whole system.

That is definitely not normally the case. We have 20k+ reported successful upgrades (see https://analytics.home-assistant.io/). This most likely is some problem with that VirtualBox version on your platform.

Can you use `systemctl list-units --failed` to see what failed? Then get the log of the failed units using `journalctl -u <unit>`.

endelight, 2023-04-29:
> What kind of logs are those?

I just made find / -iname log to see if there are any logs I can look into. I do not have any screenshots of them but they were like this: 
[image]

> That is definitely not normally the case. We have 20k+ reported successful upgrades (see https://analytics.home-assistant.io/). This most likely is some problem with that VirtualBox version on your platform.

Absolutely believe you :) I use the latest VirtualBox 6 version, as HA is not compatible with VirtualBox 7

**Meanwhile after tons of restarts and changing between Slot 0 and Slot 1 I could finish the upgrade to OS 10.1.**

The only issue I am facing after the above failure are all these **Could not fetch changelog** errors:

2023-04-29 11:17:10.501 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_vscode/changelog return code 500
2023-04-29 11:17:10.504 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_vscode: 
2023-04-29 11:17:10.526 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_nodered/changelog return code 500
2023-04-29 11:17:10.529 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_nodered: 
2023-04-29 11:17:10.551 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_influxdb/changelog return code 500
2023-04-29 11:17:10.553 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_influxdb: 
2023-04-29 11:17:10.578 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_grafana/changelog return code 500
2023-04-29 11:17:10.580 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_grafana: 
2023-04-29 11:17:10.618 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_phpmyadmin/changelog return code 500
2023-04-29 11:17:10.620 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_phpmyadmin: 
2023-04-29 11:17:10.645 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/a0d7b954_unifi/changelog return code 500
2023-04-29 11:17:10.648 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for a0d7b954_unifi: 
2023-04-29 11:17:10.678 ERROR (MainThread) [homeassistant.components.hassio.handler] /addons/cebe7a76_hassio_google_drive_backup/changelog return code 500
2023-04-29 11:17:10.680 WARNING (MainThread) [homeassistant.components.hassio] Could not fetch changelog for cebe7a76_hassio_google_drive_backup: 
2023-04-29 11:19:50.851 ERROR (MainThread) [homeassistant.components.hassio.handler] /store return code 500
2023-04-29 11:19:50.859 WARNING (MainThread) [homeassistant.components.hassio] Can't read Supervisor data:

closed 2023-04-30 by endelight