home-assistant/operating-system #2525: WiFi card not detected - Broadcom BCM943228HMB

wesleylancel, 2023-04-30:
### Describe the issue you are experiencing

The built-in WiFi card of my HP T620 thin client - a Broadcom BCM943228HMB - is not being recognized.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install Home Assistant generic image
2. Check if WiFi works (verified through `nmcli device wifi`)

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
Apr 30 19:44:33 homeassistant kernel: audit: type=1300 audit(1682883873.076:168): arch=c000003e syscall=46 success=yes exit=508 a0=3 a1=7ffc4b7868e0 a2=0 a3=7ffc4b7868cc items=0 ppid=2029 pid=2047 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="iptables" exe="/usr/sbin/xtables-nft-multi" subj=unconfined key=(null)
Apr 30 19:44:33 homeassistant kernel: audit: type=1327 audit(1682883873.076:168): proctitle=2F7573722F7362696E2F69707461626C6573002D2D77616974002D74006E6174002D4900444F434B45525F504F5354524F5554494E47002D73003132372E302E302E3131002D7000746370002D2D73706F7274003431393639002D6A00534E4154002D2D746F2D736F75726365003A3533
Apr 30 19:44:33 homeassistant kernel: eth0: renamed from vetha75d35d
Apr 30 19:44:33 homeassistant kernel: IPv6: ADDRCONF(NETDEV_CHANGE): veth3ed27ec: link becomes ready
Apr 30 19:44:33 homeassistant kernel: hassio: port 5(veth3ed27ec) entered blocking state
Apr 30 19:44:33 homeassistant kernel: hassio: port 5(veth3ed27ec) entered forwarding state
Apr 30 19:44:33 homeassistant NetworkManager[377]: <info>  [1682883873.1036] device (veth3ed27ec): carrier: link connected
Apr 30 19:44:33 homeassistant systemd-resolved[372]: System hostname changed to 'homeassistant'.
Apr 30 19:44:33 homeassistant systemd-resolved[372]: Detected conflict on homeassistant.local IN A 192.168.123.98
Apr 30 19:44:33 homeassistant systemd-resolved[372]: Hostname conflict, changing published hostname from 'homeassistant' to 'homeassistant10'.
Apr 30 19:44:33 homeassistant systemd[1]: Started libcontainer container d05fcd6fc20d7fffb5af332cc65e0e098e4b61d9fbba2cc2d1d04ec62f8b7f3e.
Apr 30 19:44:33 homeassistant kernel: audit: type=1334 audit(1682883873.968:169): prog-id=38 op=LOAD
Apr 30 19:44:33 homeassistant systemd-resolved[372]: System hostname changed to 'homeassistant'.
Apr 30 19:44:34 homeassistant systemd-resolved[372]: Detected conflict on homeassistant.local IN A 192.168.123.98
Apr 30 19:44:34 homeassistant systemd-resolved[372]: Hostname conflict, changing published hostname from 'homeassistant' to 'homeassistant2'.
Apr 30 19:44:35 homeassistant bluetoothd[378]: Endpoint registered: sender=:1.36 path=/MediaEndpoint/A2DPSink/sbc
Apr 30 19:44:35 homeassistant bluetoothd[378]: Endpoint registered: sender=:1.36 path=/MediaEndpoint/A2DPSource/sbc
Apr 30 19:44:35 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
Apr 30 19:44:35 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
Apr 30 19:44:35 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
Apr 30 19:44:36 homeassistant systemd[1]: var-lib-docker-overlay2-b9659d7c2a2bfe02a63c06f8f132fd7dc4e63e81fdbb7f2b4b45f30886c57b0b\x2dinit-merged.mount: Deactivated successfully.
Apr 30 19:44:36 homeassistant systemd[1]: mnt-data-docker-overlay2-b9659d7c2a2bfe02a63c06f8f132fd7dc4e63e81fdbb7f2b4b45f30886c57b0b\x2dinit-merged.mount: Deactivated successfully.
Apr 30 19:44:37 homeassistant systemd[1]: Started libcontainer container ed84dede8ed1ec37b6953ceea6a3548e122a63d5180ad79fea00342f203c523b.
Apr 30 19:44:37 homeassistant systemd-resolved[372]: System hostname changed to 'homeassistant'.
Apr 30 19:44:37 homeassistant systemd-resolved[372]: Detected conflict on homeassistant IN A 192.168.123.98
Apr 30 19:44:37 homeassistant systemd-resolved[372]: Hostname conflict, changing published hostname from 'homeassistant' to 'homeassistant4'.
Apr 30 19:44:39 homeassistant systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Apr 30 19:44:42 homeassistant systemd[1]: Started libcontainer container cc42030f66648ca14c57f5556b0774d9119ee79ba57b052f3606e07b7d9edb01.
Apr 30 19:44:42 homeassistant kernel: kauditd_printk_skb: 23 callbacks suppressed
Apr 30 19:44:42 homeassistant kernel: audit: type=1334 audit(1682883882.592:181): prog-id=46 op=LOAD
Apr 30 19:44:42 homeassistant kernel: audit: type=1300 audit(1682883882.592:181): arch=c000003e syscall=321 success=yes exit=14 a0=5 a1=c0001377f8 a2=78 a3=0 items=0 ppid=3032 pid=3044 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Apr 30 19:44:42 homeassistant kernel: audit: type=1327 audit(1682883882.592:181): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F63633432303330663636363438636131346335376635353536
Apr 30 19:44:42 homeassistant kernel: audit: type=1334 audit(1682883882.593:182): prog-id=47 op=LOAD
Apr 30 19:44:42 homeassistant kernel: audit: type=1300 audit(1682883882.593:182): arch=c000003e syscall=321 success=yes exit=16 a0=5 a1=c000137590 a2=78 a3=0 items=0 ppid=3032 pid=3044 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Apr 30 19:44:42 homeassistant kernel: audit: type=1327 audit(1682883882.593:182): proctitle=72756E63002D2D726F6F74002F7661722F72756E2F646F636B65722F72756E74696D652D72756E632F6D6F6279002D2D6C6F67002F72756E2F636F6E7461696E6572642F696F2E636F6E7461696E6572642E72756E74696D652E76322E7461736B2F6D6F62792F63633432303330663636363438636131346335376635353536
Apr 30 19:44:42 homeassistant kernel: audit: type=1334 audit(1682883882.593:183): prog-id=47 op=UNLOAD
Apr 30 19:44:42 homeassistant kernel: audit: type=1334 audit(1682883882.593:184): prog-id=46 op=UNLOAD
Apr 30 19:44:42 homeassistant kernel: audit: type=1334 audit(1682883882.593:185): prog-id=48 op=LOAD
Apr 30 19:44:42 homeassistant kernel: audit: type=1300 audit(1682883882.593:185): arch=c000003e syscall=321 success=yes exit=14 a0=5 a1=c000137a50 a2=78 a3=0 items=0 ppid=3032 pid=3044 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(none) ses=4294967295 comm="runc" exe="/usr/bin/runc" subj=unconfined key=(null)
Apr 30 19:44:42 homeassistant systemd-resolved[372]: System hostname changed to 'homeassistant'.
[... 119 lines cut ...]
supervisor_api | ok
version_api | ok
installed_addons | Cloudflared (4.1.4), File editor (5.6.0), SSH & Web Terminal (13.1.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 30 april 2023 om 17:58
-- | --
current_recorder_run | 30 april 2023 om 21:44
estimated_db_size | 0.76 MiB
database_engine | sqlite
database_version | 3.38.5

</details>

agners, 2023-05-02:
The latest nightly build have the driver included, can you give this a try?
https://os-builds.home-assistant.io/11.0.dev20230501/

wesleylancel, 2023-05-02:
@agners wow, thx for the quick fix! Is there an easy way for me to update my running system to a dev build?

agners, 2023-05-03:
Yes, you can temporarily switch to the dev channel:

```
ha supervisor options --channel=dev
ha supervisor reload
ha supervisor update
```

Check if there is indeed an OS update an install it
```
ha os info
ha os update
```

After the update, don't forget to go back to the stable channel. The OS won't automatically downgrade, so you can continue using the dev build:
```
ha supervisor options --channel=stable
ha supervisor reload
```

To downgrade to the latest OR release, you have to downgrade manually then:
```
ha os update --version=10.1
```

wesleylancel, 2023-05-04:
Hi @agners, thanks for the help! I've managed to update but am not getting WiFi to work yet.

`nmcli device wifi` still doesn't show anything, and rescan doesn't work.

`nmcli radio` shows `WIFI` as enabled (not sure if it did that before but `WIFI-HW` as missing`.

agners, 2023-05-09:
Can you check the host logs if the firmware has been loaded, e.g. 

```
ha host logs -t kernel -n 1000000 -b 0 | grep firmware
```

Also check if the upgrade to the dev version worked using

```
ha os info
```

agners, 2023-05-31:
This should be resolved with HAOS 10.2.

closed 2023-05-31 by agners

PhilipGarnero, 2023-06-05:
Same issue for me on my old macbook pro 2012 A1278.
The wifi card isn't detected. Apparently, it is a common issue on some linux platforms where the b43 driver is missing (in our case, because they removed the non-free folder from the alpine distribution).
It didn't work on 10.1, I just upgraded to 10.2 but it still isn't showing with `nmcli radio`

agners, 2023-06-05:
@PhilipGarnero so yours is a BCM43* then? Can you open a separate issue?

@ahmetem can you check the host logs if you see firmware errors or similar?

```
ha host logs -t kernel -n 1000000 -b 0 | grep firmware
```

PhilipGarnero, 2023-06-06:
Hi @agners !
No it's the same family of broadcom BCM94* cards. Looking online, I only found [some](https://wiki.debian.org/MacBook/Wireless) [references](https://gist.github.com/niftylettuce/5619c2be9906bcbd893e1e1a25b9d795) to a missing b43-firmware for Macbooks, so maybe it isn't related after all.
I've run the command you gave and it yielded nothing. I've tried adding bcm or broadcom or wlan or wifi but nothing either.
I skimmed around the full logs but nothing seems to be resulting in an error.

agners, 2023-06-06:
@PhilipGarnero do you have host console access? If so, can you run `login` and `lspci`?

PhilipGarnero, 2023-06-06:
Ok, so it was indeed a BCM4331 based on lspci output.
Sorry about that, specs on the web were not right.
I'll open another issue !