home-assistant/operating-system #2533: Large log file not rotated

johanneswuerbach, 2023-05-03:
### Describe the issue you are experiencing

An issue in the tradfri plugin cause a lot of logs, which all ended up in: `/mnt/data/supervisor/homeassistant/home-assistant.log`

Eventually the file grow >10 GB, the system ran out of disk space and HA crashed.

As the size accumulated over multiple days, I think this should have been catched by log rotation, but I'm not sure who would be in charge for rotating the supervisor logs. The supervisor itself or the OS?

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Watch logs grow over time.

### Anything in the Supervisor logs that might be useful for us?

```txt
I was working as intended, just kept too many logs.
```

### Anything in the Host logs that might be useful for us?

```txt
Not really.
```

agners, 2023-05-03:
This is the Home Assistant Core log. This is handled by the Core itself. ideally we should consider to disable logging entirly as we log the stdout of the Core container in `systemd-journald` as well.

But in any case, this should be reported in the Core issue tracker.

closed 2023-05-03 by github-actions[bot]

johanneswuerbach, 2023-05-03:
Thanks for the quick response @agners , looks like core actually has an option to enable log rotation https://github.com/home-assistant/core/blob/aa78962a9a3cf0f4d8d787c44a5aec7084975977/homeassistant/__main__.py#L123, which I guess just needs to be enabled somehow 🤔