home-assistant/operating-system #2535: Bluetooth failing after upgrade to HAOS 10 on Intel Bluetooth hardware

agners, 2023-05-04:
> Can you check the kernel log, to see if the problem starts similarly, e.g. using:
> 
> ```
> ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100
> ```

@agners This is the result of the command
[image]

I don't know if this is related but In Home Assistant System>Logs> Host is full of messages like this:

May 03 20:09:04 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-799946efbb48db91e285d1e93f36945c7175e095d2a5a9fe838fde71f8e6ffdf-runc.jJmMvJ.mount: Deactivated successfully.
May 03 20:09:20 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7aee883888a9e7d1b71b97ddbc2b231c2763cc47b14b2c728d6d0de7ab1c2e00-runc.RpO30d.mount: Deactivated successfully.
May 03 20:09:34 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-799946efbb48db91e285d1e93f36945c7175e095d2a5a9fe838fde71f8e6ffdf-runc.sMXYc7.mount: Deactivated successfully.
May 03 20:09:50 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7aee883888a9e7d1b71b97ddbc2b231c2763cc47b14b2c728d6d0de7ab1c2e00-runc.QivMYX.mount: Deactivated successfully.
May 03 20:10:04 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-799946efbb48db91e285d1e93f36945c7175e095d2a5a9fe838fde71f8e6ffdf-runc.ypP5rP.mount: Deactivated successfully.
May 03 20:10:20 homeassistant systemd[1]: run-docker-runtime\x2drunc-moby-7aee883888a9e7d1b71b97ddbc2b231c2763cc47b14b2c728d6d0de7ab1c2e00-runc.r9dWM5.mount: Deactivated successfully.

_Originally posted by @SrBurner in https://github.com/home-assistant/operating-system/issues/2485#issuecomment-1533660497_

samcrang, 2023-05-04:
Before:

```
[core-ssh ~]$ ha os info
board: ova
boot: B
data_disk: QEMU-HARDDISK-QM00001
update_available: true
version: "9.5"
version_latest: "10.1"
[core-ssh ~]$ ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth
May 02 17:31:08 homeassistant kernel: Bluetooth: Core ver 2.22
May 02 17:31:08 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Device revision is 2
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: API lock is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP socket layer initialized
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Waiting for firmware download to complete
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Firmware loaded in 6840573 usecs
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Waiting for device to boot
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Device booted in 18008 usecs
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Applying Intel DDC parameters completed
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Firmware revision 0.4 build 20 week 28 2022
May 02 17:31:23 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
May 02 17:31:23 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
May 02 17:31:23 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
```

Immediately after updating (same as before I guess because the host hasn't restarted?):

```
[core-ssh ~]$ ha os info
board: ova
boot: A
data_disk: QEMU-HARDDISK-QM00001
update_available: false
version: "10.1"
version_latest: "10.1"
[core-ssh ~]$ ha host logs -t kernel -n 1000000 -b 0 | grep -i luetooth
May 02 17:31:08 homeassistant kernel: Bluetooth: Core ver 2.22
May 02 17:31:08 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
May 02 17:31:08 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 02 17:31:08 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Device revision is 2
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: API lock is enabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014
May 02 17:31:09 homeassistant kernel: Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 02 17:31:10 homeassistant kernel: Bluetooth: BNEP socket layer initialized
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Waiting for firmware download to complete
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Firmware loaded in 6840573 usecs
May 02 17:31:16 homeassistant kernel: Bluetooth: hci0: Waiting for device to boot
[... 9 lines cut ...]
After poweroff, unplugging, leaving for a bit and powering on again:

```
[core-ssh ~]$ ha os info
board: ova
boot: A
data_disk: QEMU-HARDDISK-QM00001
update_available: false
version: "10.1"
version_latest: "10.1"
[core-ssh ~]$ ha host logs -t kernel -n 1000000 -b 0 | grep -i luetooth
May 04 13:49:13 homeassistant kernel: Bluetooth: Core ver 2.22
May 04 13:49:13 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
May 04 13:49:13 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 04 13:49:13 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 04 13:49:13 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 04 13:49:13 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 04 13:49:13 homeassistant kernel: Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
May 04 13:49:13 homeassistant kernel: Bluetooth: hci0: Boot Address: 0x24800
May 04 13:49:13 homeassistant kernel: Bluetooth: hci0: Firmware Version: 106-39.22
May 04 13:49:13 homeassistant kernel: Bluetooth: hci0: FW download error recovery failed (-19)
May 04 13:49:14 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 04 13:49:14 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 04 13:49:14 homeassistant kernel: Bluetooth: BNEP socket layer initialized
```

piechade, 2023-05-04:
After boot is everything working for me and i have a clean log like on 9.5. But after some time, it will stop working.

```
➜  ~ ha os info
board: generic-x86-64
boot: A
data_disk: Samsung-SSD-970-EVO-Plus-250GB-S4EUNM0T220383E
update_available: false
version: "10.1"
version_latest: "10.1"
➜  ~ ha host logs -t kernel -n 1000000 -b 0 | grep -i luetooth
May 04 13:46:57 homeassistant kernel: Bluetooth: Core ver 2.22
May 04 13:46:57 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
May 04 13:46:57 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 04 13:46:57 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 04 13:46:57 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 04 13:46:57 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 04 13:46:58 homeassistant kernel: Bluetooth: hci0: CSR: Setting up dongle with HCI ver=6 rev=22bb; LMP ver=6 subver=22bb; manufacturer=10
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Bootloader revision 0.0 build 0 week 30 2018
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Device revision is 2
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Secure boot is enabled
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: OTP lock is enabled
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: API lock is enabled
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Debug lock is disabled
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Minimum firmware build 1 week 10 2014
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Found device firmware: intel/ibt-19-0-0.sfi
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Boot Address: 0x24800
May 04 13:46:58 homeassistant kernel: Bluetooth: hci1: Firmware Version: 106-39.22
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Waiting for firmware download to complete
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Firmware loaded in 1593254 usecs
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Waiting for device to boot
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Device booted in 14619 usecs
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Found Intel DDC parameters: intel/ibt-19-0-0.ddc
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Applying Intel DDC parameters completed
May 04 13:47:00 homeassistant kernel: Bluetooth: hci1: Firmware revision 0.0 build 106 week 39 2022
May 04 13:47:00 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 04 13:47:00 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 04 13:47:00 homeassistant kernel: Bluetooth: BNEP socket layer initialized
May 04 13:47:00 homeassistant kernel: Bluetooth: MGMT ver 1.22
May 04 13:47:00 homeassistant kernel: Bluetooth: MGMT ver 1.22
May 04 13:47:15 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
May 04 13:47:15 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
May 04 13:47:15 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
```

samcrang, 2023-05-04:
FWIW, it doesn't appear to consistently break on boot. Sometimes it's fine, Bluetooth works for a bit, then stops working entirely:

```
[core-ssh ~]$ ha os info
board: ova
boot: A
data_disk: QEMU-HARDDISK-QM00001
update_available: false
version: "10.1"
version_latest: "10.1"
[core-ssh ~]$ ha host logs -t kernel -n 1000000 -b 0 | grep -i luetooth
May 04 14:06:09 homeassistant kernel: Bluetooth: Core ver 2.22
May 04 14:06:09 homeassistant kernel: NET: Registered PF_BLUETOOTH protocol family
May 04 14:06:09 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 04 14:06:09 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 04 14:06:09 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 04 14:06:09 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Bootloader revision 0.4 build 0 week 30 2018
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Device revision is 2
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: API lock is enabled
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014
May 04 14:06:09 homeassistant kernel: Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
May 04 14:06:11 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 04 14:06:11 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 04 14:06:11 homeassistant kernel: Bluetooth: BNEP socket layer initialized
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Waiting for firmware download to complete
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Firmware loaded in 6366552 usecs
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Waiting for device to boot
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Device booted in 15820 usecs
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Applying Intel DDC parameters completed
May 04 14:06:15 homeassistant kernel: Bluetooth: hci0: Firmware revision 0.4 build 20 week 28 2022
May 04 14:06:20 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
May 04 14:06:20 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
May 04 14:06:20 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
```

SrBurner, 2023-05-04:
As suggested in the other thread I’m downgrading to OS 9.5 via command Terminal (ha os update --version 9.5)
I Will report If all is working as intended again!
Thanks for the work @agners

samcrang, 2023-05-04:
Just seen this appear in my logs, I've not seen it before, don't know if it's a red-herring:

```
...
May 04 14:14:17 homeassistant kernel: Bluetooth: hci0: Opcode 0x200e failed: -16
May 04 14:14:21 homeassistant kernel: Bluetooth: hci0: ACL packet for unknown connection handle 3587
```

and

```
...
May 04 14:23:53 homeassistant kernel: Bluetooth: hci0: Opcode 0x200e failed: -16
May 04 14:24:32 homeassistant kernel: Bluetooth: hci0: Opcode 0x200e failed: -16
```

SrBurner, 2023-05-04:
> As suggested in the other thread I’m downgrading to OS 9.5 via command Terminal (ha os update --version 9.5)
> I Will report If all is working as intended again!
> Thanks for the work @agners

Downgrade done with success and all devices are detected and working as intended. At the moment no issues detected on OS 9.5

piechade, 2023-05-04:
> Downgrade done with success and all devices are detected and working as intended. At the moment no issues detected on OS 9.5

Same on my side.

d-d, 2023-05-07:
Fails for me on HAOS 10 as well.  H/W Intel NUC.

`➜  ~ ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100
May 07 19:58:14 homeassistant kernel: Bluetooth: Core ver 2.22
May 07 19:58:14 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
May 07 19:58:14 homeassistant kernel: Bluetooth: HCI socket layer initialized
May 07 19:58:14 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
May 07 19:58:14 homeassistant kernel: Bluetooth: SCO socket layer initialized
May 07 19:58:16 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
May 07 19:58:16 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
May 07 19:58:16 homeassistant kernel: Bluetooth: BNEP socket layer initialized
May 07 19:58:16 homeassistant kernel: Bluetooth: hci0: command 0xfc05 tx timeout
May 07 19:58:16 homeassistant kernel: Bluetooth: hci0: Reading Intel version command failed (-110)`

The tx timeout is also mentioned in kernel bug https://bugzilla.kernel.org/show_bug.cgi?id=215167

mariusnauj, 2023-05-10:
I have INTEL NUC, and I have problem with bluetooth too with HAOS 10.

agners, 2023-05-10:
> Fails for me on HAOS 10 as well. H/W Intel NUC.
> [..]
> The tx timeout is also mentioned in kernel bug https://bugzilla.kernel.org/show_bug.cgi?id=215167

@d-d your case seems to be a bit different as there are errors. The one I see (and others too I think) doesn't show anything in the logs.

The error you linked mentions the problem appears on Linux 5.15 as well.

Did your hardware work reliably with HAOS 9.5?

yildiraymeric, 2023-05-10:
I also have same problem with HAOS v10.1 on Intel(R) Atom(TM) x5-Z8500 system. I've downgraded to 9.5 and everything is working fine in BT side.

agners, 2023-05-10:
@yildiraymeric can you check the logs using

```
ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100
```

loedeman, 2023-05-11:
I also experience Bluetooth issues on two Intel NUC installations with internal Bluetooth. Both NUC 10.

As a workaround I created an automation which reloads the Bluetooth integration when a couple of sensors (to prevent empty batteries reloading Bluetooth constantly) report unavailable. I also have made an automation for my SwitchBot bot which reloads the Bluetooth integration before toggling the actual bot.

basjes1977, 2023-05-11:
same issue here.
hardware intel nuc with onboard bluetooth: Intel Corp. AX200 Bluetooth.
for now revered to 9.6

garytube, 2023-05-13:
Intel NUC Kit NUC6i7KYK Intel i7-6770HQ
Home Assistant 2023.5.2
Supervisor 2023.04.1
Operating System 10.1
Frontend 20230503.3 - latest

No config changes
Bluetooth onboard which is working

joyfie, 2023-05-16:
Just instal HAOS bare metal on a mini pc using Intel Wireless-AC 3165 - Can't even get the OS to recognise there is a bluetooth device connected

```
May 16 22:41:40 homeassistant kernel: Bluetooth: Core ver 2.22                                                                                                                                                                                                                                                                       
May 16 22:41:40 homeassistant kernel: Bluetooth: HCI device and connection manager initialized                                                                                                                                                                                                                                       
May 16 22:41:40 homeassistant kernel: Bluetooth: HCI socket layer initialized                                                                                                                                                                                                                                                        
May 16 22:41:40 homeassistant kernel: Bluetooth: L2CAP socket layer initialized                                                                                                                                                                                                                                                      
May 16 22:41:40 homeassistant kernel: Bluetooth: SCO socket layer initialized                                                                                                                                                                                                                                                        
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Bootloader revision 0.3 build 0 week 30 2018                                                                                                                                                                                                                                  
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Device revision is 2                                                                                                                                                                                                                                                          
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Secure boot is enabled                                                                                                                                                                                                                                                        
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: OTP lock is enabled                                                                                                                                                                                                                                                           
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: API lock is enabled                                                                                                                                                                                                                                                           
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Debug lock is disabled                                                                                                                                                                                                                                                        
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Minimum firmware build 1 week 10 2014                                                                                                                                                                                                                                         
May 16 22:41:40 homeassistant kernel: Bluetooth: hci0: Failed to load Intel firmware file intel/ibt-19-0-3.sfi (-2)                                                                                                                                                                                                                  
May 16 22:41:54 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3                                                                                                                                                                                                                                                   
May 16 22:41:54 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast                                                                                                                                                                                                                                                    
May 16 22:41:54 homeassistant kernel: Bluetooth: BNEP socket layer initialized
```

mariusnauj, 2023-05-25:
The problem still not solved, not bluetooth device restoring connection after couple of hours. But it working not good with HAOS 10.

[image]
[image]

osaether, 2023-05-30:
I tried 10.2 today and it still does not work. Specifically my Inkbird IBS-TH2 sensors are unavailable. Rolling back to 9.5 and everything is working again.

jaegerschnitzel, 2023-05-31:
> I tried 10.2 today and it still does not work. Specifically my Inkbird IBS-TH2 sensors are unavailable. Rolling back to 9.5 and everything is working again.

Same for me on my NUC11 (RNUC11PAHi50000).

rhdny, 2023-06-03:
Rolling back to 9.5 seems to have resolved my issues as well (Yale locks and RadonEye detectors using the BT integration)

philpownall, 2023-06-10:
I am also on 10.2 and this message is spamming my home assistant kernel logs (see below).  My hardware setup is a little different - I am running Home Assistant OS on a Beelink GK35 (Intel), and I DO NOT use the host Bluetooth (it is old bluetooth 4.0) - I use esphome esp32 devices with esp32_ble_tracker and bt_proxy and multiple Xiaomi temp/hum Bluetooth devices, and they all work fine.

My only issue is that this (bug? error message?) is spamming my host logs.  Is there a way to disable this initial HCI request check, or turn off the host system Bluetooth altogether?

```
# ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100
Jun 09 23:13:03 homeassistant kernel: Bluetooth: Core ver 2.22
Jun 09 23:13:03 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
Jun 09 23:13:03 homeassistant kernel: Bluetooth: HCI socket layer initialized
Jun 09 23:13:03 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
Jun 09 23:13:03 homeassistant kernel: Bluetooth: SCO socket layer initialized
Jun 09 23:13:08 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Jun 09 23:13:08 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
Jun 09 23:13:08 homeassistant kernel: Bluetooth: BNEP socket layer initialized
Jun 09 23:13:10 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
Jun 09 23:12:59 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
Jun 09 23:13:01 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
Jun 09 23:13:03 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
Jun 09 23:13:05 homeassistant kernel: Bluetooth: hci0: sending initial HCI reset failed (-19)
```

sairon, 2023-06-13:
I encountered this issue immediately when I upgraded to HAOS 10.0 on NUC11ATKPE and eventually downgraded to 9.5 for the time being. Today I finally had some time to spare so I looked into the issue. The problem is that the only way how the issue manifests is that HA stops receiving advertisement packets after few seconds of scanning - or according to my observations - it seems it receives only a few from a **single** device, then it only receives "new" advertisements, even if the device lies virtually next to the NUC. I don't even see the advertisements in the  `bluetoothctl` scan either (with duplicate entries enabled, ofc), so `core` is probably not to blame. It helps to mitigate the issue a bit though - if no advertisement arrives in watchdog interval (in my case it's quite busy neighborhood so it takes couple of minutes until the watchdog does its job), the scan is restarted and I can again see the packets from the "ignored" devices. The `dmesg` entries mentioned above appeared only when I had two VMs with the same USB device attached from the host, but don't appear even after prolonged time with a single VM, yet the bug with lost advertisements persists.

To better identify what component causes it, I checked out the HAOS repo and tried these combinations:

- manually built 10.2, enabled Bluetooth integration and paired some BLE sensors - fails
- the previous, but with latest `bluez5_tools` commits reverted to 5.53 - fails
- the previous, with Linux downgraded to 5.15.90 (with Kernel config from 9.5) - OK

I can do some more tests and eventually bisect Linux versions manually to find where it was introduced, but my current testing method is a bit tedious, because I always have to create a qcow2 image, create a new Proxmox VM using a script and restore a backup once it starts. @agners, don't you have a tip how to do it better, please? Isn't there a way to simulate OTA to a manually built unsigned version, maybe?

FTR, my NUC is using this Bluetooth adapter:
```Bus 001 Device 020: ID 8087:0aaa Intel Corp. Bluetooth 9460/9560 Jefferson Peak (JfP)```

Sensors used in testing are LYWSD03MMC with pvvx firmware (mostly version v3.8 or v3.9, IIRC) and one RAPT Pill.

agners, 2023-06-14:
> The problem is that the only way how the issue manifests is that HA stops receiving advertisement packets after few seconds of scanning

The same here, but for me it takes quite a while to stop receiving packets, I'd say hours.

> the previous, with Linux downgraded to 5.15.90 (with Kernel config from 9.5) - OK

I've made the same tests and same observation. 5.15 seems fine. I also tested older/newer versions of linux-firmware, with no success either. So it really seems to be the kernel.

> I can do some more tests and eventually bisect Linux versions manually to find where it was introduced, 

That was actually my next plan, but I hesitated since it takes me quite a while to definitely proof good/bad.

> but my current testing method is a bit tedious, because I always have to create a qcow2 image, create a new Proxmox VM using a script and restore a backup once it starts. @agners, don't you have a [...]

Going via raucb is what I would do here: Build the raucb file, copy it to the target and manually install it using `rauc install <updatefile>`. I am not sure about completely unsigned, I haven't tried that. We do have a developer CA, I'll contact you directly.

Thanks a lot for your effort on this issue :pray:

agners, 2023-06-15:
PR https://github.com/home-assistant/operating-system/pull/2598 seems to resolve the issue for me and @sairon. I hope this resolves Bluetooth problems for most of you.

SrBurner, 2023-06-15:
@agners is it released in home assistant OS 10.2 stable? Or is part of some BETA? In order to try it.

SrBurner, 2023-06-15:
My Home Assistant server is “on production” but I will give 10.3 a try on release!

sairon, 2023-06-16:
@pricei How far is the button from the NUC, or what are the obstacles in the way? I see that RSSI is lower than -90, which is a very weak signal. Can you try moving it closer and check if you see the problem then?

sairon, 2023-06-16:
This indeed looks somewhat similar to what I saw with my BLE sensors, although it was not just the first one packet - it was a few more, only then it stopped seemingly ignoring them. @pricei, just to check - can you confirm this was working fine on 9.5 release? If not, then we're probably talking about yet another bug.

sairon, 2023-06-16:
@pricei There is nRF Connect app both for Android and iOS (though IMHO it's better to use it on Android because iOS has limited Bluetooth API) that can be used as a generic BLE scanner/debugger - you should be able to see the advertisements including the service and/or manufacturer data there. Also it's worth trying if 9.5 resolves the issue for you - the downgrade shouldn't cause any trouble (and many users have downgraded too once they found HAOS 10 unstable) - just run `ha os update --version 9.5` in the terminal.

agners, 2023-06-23:
Is 9.5 rock solid for the same use case?

If so, please open a new issue along with logs of Core and Host logs when the issue appears.

Sorien, 2023-06-24:
@agners I'd to ask if you are trying to make that change upstream to the kernel, I know that it's not a problem for HAOS anymore but there are many HA running on different distributions affected by this bug.

contactcr, 2023-06-24:
> 

So far, yes.  The locks are very responsive and seem to work all the time with no drop outs.  I guess I'll break it one more time for the cause!

contactcr, 2023-06-24:
I opened [a new bug in OS](https://github.com/home-assistant/operating-system/issues/2612)

I didn't see much in Host logs though.  Do I need to enable some more advanced logging?

j-pas, 2023-06-25:
I can confirm some issues remain on (at least) NUC it seems. I have a NUC 10, have 2 different bluetooth devices both with simultaneous connection issues at random moments and random duration.. See below screenshot for example.

HAOS 9.5 -> 10.1 (issues began)
HOAS 10.1 -> 9.5 (issues gone, all works fine)
HAOS 9.5 -> 10.3 (issues are back)

Has another ticket been opened somewhere for this? Happy to provide more input if needed.

[image]

agners, 2023-06-26:
> @agners I'd to ask if you are trying to make that change upstream to the kernel, I know that it's not a problem for HAOS anymore but there are many HA running on different distributions affected by this bug.

I've brought it up on the Bluetooth Linux kernel mailing list, see:
https://lore.kernel.org/linux-bluetooth/0de3f0d0d5eb6d83cfc8d90cbb2b1ba1@agner.ch/T/#mb0d04fdaf6fb2484e69b4aba93f052a3e7ecb206

However, no progress so far.

j-pas, 2023-06-26:
I run Proxmox 7.4-3 on my Intel Nuc. Proxmox is shipped on top of Debian (version 11).
Within Proxmox I run Homeassistant in a VM based on the HAOS iso, currently upgraded to version 10.3.
As I pass the Bleutooth hardware (which is the onboard hardware but essentially a USB Device) from the host NUC to the VM running HAOS, I`m not sure how much of a role the underlaying Debian distro plays... but I`m not an expert on this. 

Executed on the NUC (debian): 
root@nuc:~# uname -a
Linux nuc 5.15.107-2-pve #1 SMP PVE 5.15.107-2 (2023-05-10T09:10Z) x86_64 GNU/Linux
root@nuc:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian

SrBurner, 2023-06-26:
Several days without any issue, so fixed for me!, thanks a lot @agners

Sorien, 2023-07-03:
> > @agners I'd to ask if you are trying to make that change upstream to the kernel, I know that it's not a problem for HAOS anymore but there are many HA running on different distributions affected by this bug.
> 
> I've brought it up on the Bluetooth Linux kernel mailing list, see: https://lore.kernel.org/linux-bluetooth/0de3f0d0d5eb6d83cfc8d90cbb2b1ba1@agner.ch/T/#mb0d04fdaf6fb2484e69b4aba93f052a3e7ecb206
> 
> However, no progress so far.
> 
> On what OS do you observe the issue?

my home assistant server is also HTPC it's a small Asus pn52 box running Ubuntu 22.04 after upgrading to 23.04 a see the same problem with BLE connection dropouts

yannpub, 2023-07-03:
I upgraded my Supervised installation from Debian 11 to Debian 12, and now facing the same issue with my Intel bluetooth adapter. Definitely something to do with kernel 6.1 I would say.

Sorien, 2023-09-06:
as I see proper fix was merged to linux-next so when I get it right Linux kernel 6.6 should be working again for us ...

agners, 2023-10-13:
@yannpub @Sorien I've noticed that the Bluetooth fix `Bluetooth: hci_sync: Fix handling of HCI_QUIRK_STRICT_DUPLICATE_FILTER` is listed in the [Linux 6.1.57 changelog](https://cdn.kernel.org/pub/linux/kernel/v6.x/ChangeLog-6.1.57). So the fix should trickle into distribution Linux kernel updates soon.

agners, 2023-10-13:
Note that with HAOS 11.0 the `Bluetooth: hci_sync: Fix handling of HCI_QUIRK_STRICT_DUPLICATE_FILTER` is part of HAOS as well (instead of the work around via revert).

bjornarfjelldal, 2023-10-15:
Hi! I really appreciate your efforts in trying to resolve bluetooth in HAOS 11.0, since it has been especially buggy and variable over the past year. However, after updating from 10.5 to 11.0 (on an Intel NUC using Proxmox) bluetooth seemingly has become even worse than before. I have tried to delete the BT integration, toggle passive scanning on and off and physically reboot host. Running "bluetoothctl" in terminal addon produced a lof devices logs, but doesn't help much on stability.

The console reports "Bluetooth: hci0 opcode 0x200e failed -16".

Anything I should try to do - or logs I should provide?
[image]

j-pas, 2023-10-15:
Hi all,

I am also running HAOS 11.0 on Intel NUC (NUC10i3FNH2) using Proxmox  and so far (16 hour) so good !

I must admit that it was running ok-ish on 10.x with only 2 or 3 short bluetooth gaps per day.
It was far worse some weeks ago (see my earlier post) but I found a procedure somewhere online that I tried. Not sure anymore if that was the solution but @bjornarfjelldal you can give this a try:
Shut down the physical machine (NUC) and unplug power for 10 min. After that restore power and boot. 
Although I dont like solutions in this direction I tried it anyway and I think this is what made it run more stable for some weeks now...

Let us know how that worked!

Sorien, 2023-10-29:
I updated my HTPC to Ubuntu 23.10 and kernel 6.5.7 and I can confirm Bluetooth devices are working again

Stooovie, 2023-12-17:
This STARTED happening after upgrading HA OS to 11.2. I haven't had a single BT failure since installing the machine a year ago.

agners, 2023-12-26:
@dkebler

> note kernel 6.1 is way behind current LTS of 6.6

6.1 was the last LTS, and 6.6 was announced a couple weeks ago, not sure if I would label that as "way behind".

But, yes, upgrading to 6.6 LTS is on our todo list. A PR with patches rebased and validated on all supported platforms is of course very welcome :wink:

angainor, 2024-02-18:
FYI, I have just installed HA OS version 12.0rc1. The RPi controller has connected without the passive mode, and my device has been reporting without issues for a few hours now. :crossed_fingers:

danijel-bjelancevic, 2024-11-08:
I still have this issue. Just installed Uniraid on my Asus NUC and running HA through a VM on Uniraid. When I try to pass the Intel Chip through as an USB I get this in the HA logs:

[image]