home-assistant/operating-system #2538: Odroid XU4 not booting after update to 10

DrVic2r, 2023-05-05:
### Describe the issue you are experiencing

The other day I tried to update form HAOS 9.5 to 10.0 but after updating the Odroid never booted again, and I get no response when connecting a display. I'm running HAOS on Odroid XU4 booting from eMMC. 

### What operating system image do you use?

odroid-xu4 (Hardkernel ODROID-XU4)

### What version of Home Assistant Operating System is installed?

10.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update from 9.5 to 10 using the GUI
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
can't access
```

### Anything in the Host logs that might be useful for us?

```txt
can't access
```

agners, 2023-05-09:
My test instance uses eMMC boot as well, and seems to work well. Do you have a serial console adapter with which you could check the serial console with?

Otherwise, maybe checking with a different OS flashed on a SD card if the system generally boot still, and check the content of the eMMC (e.g. check if the partitions are present using fdisk).

patvdleer, 2023-05-12:
Im using the latest 10.1 image and I have the same issue, I noticed the boot partition is pretty empty. I expected a `boot.ini` but even that isn't there. Only;

* `boot.scr`
* `cmdline.txt`
* `exynos5422-odroidxu4.dtb`

Looking into the `boot.scr` I see the following:
[image]

agners, 2023-05-16:
@patvdleer Home Assistant OS brings its own U-Boot (which is stored at a particular location as required by the boot ROM outside of the partition table). My instance has exactly the same files an boots fine:

```
# ls -la /mnt/boot/
total 86
drwxr-xr-x    2 root     root         16384 Jan  1  1970 .
drwxr-xr-x    6 root     root            68 Apr 13 11:01 ..
-rwxr-xr-x    1 root     root          2746 Apr 25 17:42 boot.scr
-rwxr-xr-x    1 root     root            36 Apr 25 17:42 cmdline.txt
-rwxr-xr-x    1 root     root         65142 Apr 25 17:42 exynos5422-odroidxu4.dtb
```

So you flashed the 10.1 image on eMMC, plugged it in and end up in a boot loop? Do you have a serial console log available?

patvdleer, 2023-05-16:
> So you flashed the 10.1 image on eMMC, plugged it in and end up in a boot loop? 

Yes

> Do you have a serial console log available?

No, I don't have the plug/connector, I'm planning removing the current connector and soldering some wires to 2.54mm headers but I haven't had the time to do so (yet).

agners, 2023-05-16:
I remember I run into issues one point during development where the eMMC boot partition had a old/bad U-Boot installation. Unfortunately, you can't access the boot partition on your host PC using an eMMC adapter. But there ways to do it, see https://github.com/home-assistant/operating-system/blob/dev/Documentation/boards/hardkernel/odroid-xu4.md.

DrVic2r, 2023-05-17:
> 

Hi, I was in the process of migrating to a larger emmc anyways and have now gotten around to do it. I can, as before, boot with 9.5 if I follow the above instructions and boot from a SD card the first time. However, I ran into the problem last time I tried to upgrade to 10.1 that I didn't even go into the boot loop. I have not gotten around to try an upgrade with the new emmc, I hope to try it again within the next few days.

DrVic2r, 2023-08-15:
With the new larger emmc updating worked fine.

closed 2023-11-21 by github-actions[bot]