home-assistant/operating-system #2545: Upgrade from 9.5 to 10.1 loses network connection

andyp05, 2023-05-11:
### Describe the issue you are experiencing

Before upgrade from 9.5 to 10.1, the CLI screen says
IPv4 addresses of enp1s0: 192.168.1.205/24

After upgrade it says:
IPv4 addresses of enp1s0:

Reinstalled 9.5. It is back:
Could not use: ha os update --version 9.5 to downgrade since I has no network.
Put down a new OS using BelenaEtcher
IPv4 addresses of enp1s0: 192.168.1.205/24

One thing I did notice is that an error was generated during boot that said unable to synchronize time.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 10.1

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
None I can capture.
```

domo-greg, 2023-05-26:
Je rejoins l'issues car j'ai le même problème qu'@andyp05
impossibilité de connexion avec la version 10.1 actuelle. je suis obligé de rester sur la version 9.5 du système d'exploitation

agners, 2023-05-26:
@andyp05 @domo-greg what type of systems do you have (make/model especially also what type of Ethernet adapter is in that system?).

domo-greg, 2023-05-26:
C'est un mini pc de la marque nipogi.
La carte ethernet est une realteck ethernet family

fcorrao1, 2023-05-28:
The network is also broken on my vm after the upgrade.  I'm running on kvm with bridged mode networking using virtio.  I have other vm's bridged to the same network interface on the kvm host and they all work fine.  From the ha console I can see the enp1s0 interface is up and an ip address is configured (manually) but I can't reach any other device on the local network.  'ip neigh' shows arp resolution is failing.  Maybe a driver issue?  Since I can't downgrade to 9.5 in this state, is there any other way to troubleshoot?

andyp05, 2023-05-30:
I can SSH in or get to the CLI. What command do I type in to get the ethernet card info?

agners, 2023-05-31:
Typically you can check this using `lspci` or similar commands, however this might not be available when using the SSH add-on.

You'd need to use the Operating System SSH access (see https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host).

gaborbernat, 2023-06-04:
I can confirm I see the same with https://www.amazon.com/dp/B0BS369VMF, same with 10.2 and 10.1. 9.5 works as expected.

gaborbernat, 2023-06-04:
The symptoms are similar to https://github.com/home-assistant/operating-system/issues/2517, however in my case the M2.SSD is not an issue.

gaborbernat, 2023-06-05:
[image]
[image]

Here's the requested information @agners

gaborbernat, 2023-06-05:
One change I noticed with 10.2 is that now the ethernet connection reports no carrier, and such cannot downgrade back to 9.5. As `ha os update` does not work with offline files my mini PC is essentially useless, as I'll have to do a fresh install...

This wasn't the case with 10.1 where I could downgrade after disabling ASPM and resetting the network connection.

gaborbernat, 2023-06-05:
I've created a bootable USB with 9.5 and booted from that which allowed me to start it up again with network access. Is it possible to use `ha os update` to update the `os` in a different disk than the one booted into to repair my PC?

agners, 2023-06-05:
We have an A/B type system. You can force the system to boot the other system next time using:

```
rauc status mark-active other
```

Then reboot the system.

gaborbernat, 2023-06-05:
I meant as in using the 9.5 OS to downgrade the other (non-live) disk to 9.5. That is because only 9.5 has network, so I cannot downgrade 10.2 once booted into it.

nochlab1, 2023-08-03:
> @agners another things I noticed is that the Wifi card is not detected, any idea for that?

Did you fix this? 
I got a 12th gen n95 mini pc and both ethernet show up but not the wifi card which i need

andyp05, 2023-08-03:
KAMRUI GK3 Plus Mini PC died 3 months in. Horrible support/warranty. Trashed it.
Moved to an Intel NUC and everything works as expected. 
I am not in a position to test this anymore. But I'll leave this issue open for everyone else having the same issues.

PlayFaster, 2023-08-08:
Just in case anyone else is experiencing this issue and (like myself) has spent some time searching, there are three (3) issues, all very similar on this matter at present (Home Assistant OS ver 10x not able to obtain an IP address, particularly impacting NiPro NUCs with Realtek Ethernet. 

See #2517 . OP resolved this by changing SSD but general issue seems broader. 
See #2630 . Open as of now, I've posted by experiences there, hoping it may help to progress this issue. 
 
Thanks

agners, 2023-11-07:
Since the OP moved to a different platform, and #2630 has been resolved, I consider this as solved too. If one of the folks speaking up here can reproduce the problem on OS 11.1 still, please open a new issue. Thanks!

closed 2023-11-07 by agners