home-assistant/operating-system #2547: Intel AX210 wifi not working

voed, 2023-05-12:
### Describe the issue you are experiencing

Just bought new wifi adapter for my Intel NUC and it does not works. After upgrading from 10.1 to the nightly release (11.0.dev20230511), it started once:
```
May 12 13:33:32 homeassistant kernel: iwlwifi 0000:02:00.0: enabling device (0000 -> 0002)
May 12 13:33:32 homeassistant kernel: iwlwifi 0000:02:00.0: api flags index 2 larger than supported by driver
May 12 13:33:32 homeassistant kernel: iwlwifi 0000:02:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
May 12 13:33:32 homeassistant kernel: iwlwifi 0000:02:00.0: loaded firmware version 72.a764baac.0 ty-a0-gf-a0-72.ucode op_mode iwlmvm
May 12 13:33:33 homeassistant kernel: iwlwifi 0000:02:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
May 12 13:33:33 homeassistant kernel: iwlwifi 0000:02:00.0: loaded PNVM version e4a49534
May 12 13:33:33 homeassistant kernel: iwlwifi 0000:02:00.0: Detected RF GF, rfid=0x10d000
May 12 13:33:33 homeassistant kernel: iwlwifi 0000:02:00.0: base HW address: f4:7b:09:ca:1d:45
May 12 13:33:33 homeassistant kernel: iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0
```
But after reboot it does not works anymore. Tried to put pnvm file to the /mnt/data/supervisor/firmware folder, but no changes.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

11.0.dev20230511

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

-

### Anything in the Supervisor logs that might be useful for us?

```txt
-
```

### Anything in the Host logs that might be useful for us?

```txt
May 12 14:00:08 homeassistant kernel: iwlwifi 0000:02:00.0: enabling device (0000 -> 0002)
May 12 14:00:08 homeassistant kernel: iwlwifi 0000:02:00.0: api flags index 2 larger than supported by driver
May 12 14:00:08 homeassistant kernel: iwlwifi 0000:02:00.0: TLV_FW_FSEQ_VERSION: FSEQ Version: 0.0.2.36
May 12 14:00:08 homeassistant kernel: iwlwifi 0000:02:00.0: loaded firmware version 72.a764baac.0 ty-a0-gf-a0-72.ucode op_mode iwlmvm
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Detected Intel(R) Wi-Fi 6 AX210 160MHz, REV=0x420
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Detected RF GF, rfid=0x10d000
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Microcode SW error detected. Restarting 0x0.
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Start IWL Error Log Dump:
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Transport status: 0x0000004A, valid: 6
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: Loaded firmware version: 72.a764baac.0 ty-a0-gf-a0-72.ucode
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000071 | NMI_INTERRUPT_UMAC_FATAL
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x002002F0 | trm_hw_status0
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000000 | trm_hw_status1
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x004DB5D2 | branchlink2
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x004D180A | interruptlink1
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x004D180A | interruptlink2
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x004DA32C | data1
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000010 | data2
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000000 | data3
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000000 | beacon time
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x000147E3 | tsf low
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000000 | tsf hi
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000000 | time gp1
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x0002866E | time gp2
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000001 | uCode revision type
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000048 | uCode version major
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0xA764BAAC | uCode version minor
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00000420 | hw version
May 12 14:00:09 homeassistant kernel: iwlwifi 0000:02:00.0: 0x00C89002 | board version
[... 206 lines cut ...]

### System information

<html>
<body>
<!--StartFragment-->

Версія | core-2023.5.2
-- | --
Тип інсталяції | Home Assistant OS
Середовище розробки | false
Supervisor | true
Docker | true
Користувач | root
Віртуальне оточення | false
Версія Python | 3.10.11
Сімейство операційних систем | Linux
Версія операційної системи | 6.1.28
Архітектура ЦП | x86_64
Часовий пояс | Europe/Kiev
Каталог конфігурацій | /config

<!--EndFragment-->
</body>
</html>

agners, 2023-05-16:
This is probably caused by #2542.

> But after reboot it does not works anymore. Tried to put pnvm file to the /mnt/data/supervisor/firmware folder, but no changes.

Probably the probing happens to early, at a point where `/mnt/data` is not mounted yet. We'll need to find a way to fix this.

But for your case, the right fix is to deploy the missing files as part of HAOS.

agners, 2023-05-25:
This should be resolved with the latest nightly builds.

closed 2023-05-25 by agners

voed, 2023-05-26:
Works fine. Thanks.