home-assistant/operating-system #2548: Wifi: Realtek r8188eu stopped working

tessharp, 2023-05-13:
### Describe the issue you are experiencing

HA OS 10.1 completely loses ability of using the r8188eu wifi card (TP-Link TL-WN725N). If starting without the card connected it starts returning a "MAC IOL_exec_cmds Failed !!!" error.  Issue was not present in version 9.5. Already made sure that the Wifi adapter works in a separate computer.

Using the Hyper-V version of HA OS w/ PCIe pass through with a USB hub.

All the connections (ZigBee, Z-Wave...) still work, just network connectivity is completely lost.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Start the HA OS machine
2. Use the HA console and type the following commands to monitor device-based messages
`login
dmesg -w`
3. Plug in the wifi dongle
4. Configure a wifi connection through `nmcli`
5. Use HA web UI until connection is lost
6. Observe that the HA console now shows the IOL_exec_cmds error

### Anything in the Supervisor logs that might be useful for us?

```txt
None that I can identify.
```

### Anything in the Host logs that might be useful for us?

```txt
None that I can identify.
```

tessharp, 2023-05-19:
Confirmed issue in 10.1 - Done upgrades/downgrades. OS 9.5 picks up the Wifi card instantly.

agners, 2023-05-22:
Hm, this is a staging driver enabled with #1811. It seems that the driver got removed in the latest Linux 6.3 release, but I'd rather stick with the 6.1 LTS driver. However, I did not found a commit/bugfix which might address the particular problem you are describing.

@cr0wbar is your device still working with HAOS 10.1?

tessharp, 2023-05-22:
I've also tried a fresh install of HA OS 10.1 (i.e. downloaded the Hyper-V disk image and booting it up). If I take the same steps, I see the same error.

cr0wbar, 2023-05-23:
> Hm, this is a staging driver enabled with #1811. It seems that the driver got removed in the latest Linux 6.3 release, but I'd rather stick with the 6.1 LTS driver. However, I did not found a [...]

Mine is actually still working.
On Home assistant 2023.5.3
Supervisor 2023.04.01
Operating system 10.1

Raspberry pi 2

Stooovie, 2023-06-25:
It's due to Linux kernel dropping support for a lot of wifi dongles for whatever reason. Getting some of the Realtek ones to work is nearly impossible now. I have absokuno idea why they do this as the 8188 chipset is still widely used and many if not most of wifi donglws you can buy are based on these.

closed 2023-10-01 by github-actions[bot]