home-assistant/operating-system #2550: No audio HW detected on Intel(R) Atom(TM) x5-Z8500 CPU based install

yildiraymeric, 2023-05-16:
### Describe the issue you are experiencing

Homeassistant audio component does not detect audio HW on my Intel(R) Atom(TM) x5-Z8500 CPU based system.

When I query for audio hardware with  ``ha audio info``command, following response is generated;

```
audio:
  application: []
  card: []
  input: []
  output:
  - applications: []
    card: null
    default: true
    description: Dummy Output
    index: 0
    mute: false
    name: auto_null
    volume: 0.989990234375
host: 172.30.32.4
update_available: false
version: 2023.04.1
version_latest: 2023.04.1
```

In home assistant web UI all audio related sections show "Dummy Audio" as follows

[image]

I've previously installed Ubuntu 22.04.2 LTS and it detected audio hardware just fine.

Then installed Debian 11.7.0 and it did not detect the audio HW but installing non-opensource intel drivers and libraries solved the problem.

Are there any way to update audio drivers in HAOS ?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Just install HA on Intel(R) Atom(TM) x5-Z8500 CPU based system (my device is Beelink T4)

### Anything in the Supervisor logs that might be useful for us?

```txt
no logs related to audio hardware
```

### Anything in the Host logs that might be useful for us?

```txt
> ha audio logs
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/alsa-mixer.sh
[15:17:13] INFO: Adjust ALSA mixer settings for /dev/snd/control*
cat: can't open '/sys/class/sound/control*/device/id': No such file or directory
cont-init: info: /etc/cont-init.d/alsa-mixer.sh exited 0
cont-init: info: running /etc/cont-init.d/filesystem.sh
cont-init: info: /etc/cont-init.d/filesystem.sh exited 0
cont-init: info: running /etc/cont-init.d/pulse-config.sh
aplay: device_list:274: no soundcards found...
cont-init: info: /etc/cont-init.d/pulse-config.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun alsa (no readiness notification)
services-up: info: copying legacy longrun pulseaudio (no readiness notification)
s6-rc: info: service legacy-services successfully started
W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set.
W: [pulseaudio] main.c: Home directory of user 'root' is not '/var/run/pulse', ignoring.
W: [pulseaudio] caps.c: Normally all extra capabilities would be dropped now, but that's impossible because PulseAudio was built without capabilities support.
W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please make sure that you actually do want to do that.
W: [pulseaudio] main.c: Please read http://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/WhatIsWrongWithSystemWide/ for an explanation why system mode is usually a bad idea.
E: [pulseaudio] module-rescue-streams.c: module-rescue-stream is obsolete and should no longer be loaded. Please remove it from your configuration.
```

### System information

## System Information

version | core-2023.5.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.10.11
os_name | Linux
os_version | 6.1.25
arch | x86_64
timezone | Europe/Istanbul
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4994
Installed Version | 1.32.1
Stage | running
Available Repositories | 1358
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.1
-- | --
update_channel | stable
supervisor_version | supervisor-2023.04.1
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 56.0 GB
disk_used | 15.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | MariaDB (2.6.1), Duck DNS (1.15.0), File editor (5.6.0), Advanced SSH & Web Terminal (14.1.0), Mosquitto broker (6.2.1), AirCast (3.5.3), Network UPS Tools (0.11.2), Spotify Connect (0.12.3), AppDaemon (0.12.2), InfluxDB (4.6.0), Matter Server (4.3.1), Glances (0.18.2), Studio Code Server (5.5.7), Inadyn (2.10.0), Portainer (2.18.2), Home Assistant Google Drive Backup (0.110.4), speedtest (v0.5.4-alpha.0), tasmocompiler (10.4.0), Simple Scheduler (2.11), ESPHome (2023.4.4), Samba share (10.0.1), Tailscale (0.11.1), OpenVPN Client (0.2.9), CloudFlare Tunnels (0.0.1), AdGuard Home (4.8.6), NGINX Home Assistant SSL proxy (3.4.2), Scrutiny (v0.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

<details><summary>Recorder</summary>

error | failed to load: unknown
-- | --

</details>

<details><summary>Sonoff</summary>

version | 3.5.1 (122751c)
-- | --
cloud_online | 2 / 6
local_online | 1 / 1

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.8
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 2

</details>

agners, 2023-05-16:
> my Intel(R) Atom(TM) x5-Z8500 CPU based system

The CPU does not really guarantees what audio hardware in your system, that mainly depends on the motherboard. What exact system is that?

yildiraymeric, 2023-05-16:
here is a much more comprehensive HW list via hwinfo command

```
➜  ~ hwinfo --short
cpu:
                       Intel(R) Atom(TM) x5-Z8500  CPU @ 1.44GHz, 2160 MHz
                       Intel(R) Atom(TM) x5-Z8500  CPU @ 1.44GHz, 2160 MHz
                       Intel(R) Atom(TM) x5-Z8500  CPU @ 1.44GHz, 2240 MHz
                       Intel(R) Atom(TM) x5-Z8500  CPU @ 1.44GHz, 2239 MHz
keyboard:
  /dev/input/event2    Logitech Unifying Receiver
  /dev/tty1            serial console
mouse:
  /dev/input/event5    Logitech Unifying Receiver
graphics card:
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Integrated Graphics Controller
sound:
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Low Power Engine Audio
network:
  enp2s0               Realtek RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
  wlp1s0               Intel Wireless 3165
network interface:
  lo                   Loopback network interface
  enp2s0               Ethernet network interface
  hassio               Ethernet network interface
  wlp1s0               Ethernet network interface
disk:
  /dev/mmcblk1boot0    Disk
  /dev/zram1           Disk
  /dev/mmcblk1boot1    Disk
  /dev/mmcblk1         Disk
  /dev/zram2           Disk
partition:
  /dev/mmcblk1p1       Partition
  /dev/mmcblk1p2       Partition
  /dev/mmcblk1p3       Partition
  /dev/mmcblk1p4       Partition
  /dev/mmcblk1p5       Partition
  /dev/mmcblk1p6       Partition
  /dev/mmcblk1p7       Partition
  /dev/mmcblk1p8       Partition
usb controller:
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series USB xHCI Controller
bios:
                       BIOS
bridge:
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCI Express Port #1
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCU
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series SoC Transaction Register
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series PCI Express Port #2
hub:
                       VIA VL813 Hub
                       VIA VL813 Hub
                       Linux Foundation 2.0 root hub
                       Genesys Logic Hub
                       Linux Foundation 3.0 root hub
memory:
                       Main Memory
bluetooth:
                       Intel Bluetooth wireless interface
unknown:
                       FPU
                       DMA controller
                       PIC
                       Keyboard controller
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Imaging Unit
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Trusted Execution Engine
                       Intel Atom/Celeron/Pentium Processor x5-E8000/J3xxx/N3xxx Series Power Management Controller
  /dev/ttyS0           16550A
                       Logitech Unifying Receiver
```

agners, 2023-05-16:
Hm, seems Linux support is not existing for that sound device: https://linux-hardware.org/?id=pci:8086-22a8-8086-7270

Is Audio working on other Linux distributions?

agners, 2023-05-16:
Oh I see:

> I've previously installed Ubuntu 22.04.2 LTS and it detected audio hardware just fine.

> Then installed Debian 11.7.0 and it did not detect the audio HW but installing non-opensource intel drivers and libraries solved the problem.

I'd rather prefer not to ship closed source binaries with HAOS :cold_sweat:

yildiraymeric, 2023-05-17:
> Oh I see:
>
>
>
> I'd rather prefer not to ship closed source binaries with HAOS 😰

I totally agree with you, but is there a way of installing those packages by users later on?

https://packages.debian.org/bullseye/firmware-intel-sound

I think this package might solve the problem

j1nx, 2023-05-20:
Not sure, but maybe the soundcard is supported by SOF.

https://github.com/thesofproject/sof-bin/

Might need some additional kernel modules though.

closed 2023-08-25 by github-actions[bot]

andrioid, 2023-10-23:
I'm having a similar issue with a mini PC (Intel Celeron) that doesn't detect the audio card. The kernel reports

```
[    3.440062] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if info 0x040100
[    3.440327] sof-audio-pci-intel-apl 0000:00:0e.0: DSP detected with PCI class/subclass/prog-if 0x040100
[    3.440475] sof-audio-pci-intel-apl 0000:00:0e.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
[    3.446945] sof-audio-pci-intel-apl 0000:00:0e.0: use msi interrupt mode
[    3.476809] sof-audio-pci-intel-apl 0000:00:0e.0: NHLT_DEVICE_I2S detected, ssp_mask 0x21
[    3.476822] sof-audio-pci-intel-apl 0000:00:0e.0: Overriding topology with MCLK mask 0x2 from NHLT
[    3.476825] sof-audio-pci-intel-apl 0000:00:0e.0: hda codecs found, mask 4
[    3.477275] sof-audio-pci-intel-apl 0000:00:0e.0: Direct firmware load for intel/sof/sof-apl.ri failed with error -2
[    3.477281] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof firmware file is missing, you might need to
[    3.477293] sof-audio-pci-intel-apl 0000:00:0e.0:        download it from https://github.com/thesofproject/sof-bin/
[    3.477299] sof-audio-pci-intel-apl 0000:00:0e.0: error: failed to load DSP firmware -2
[    3.477868] sof-audio-pci-intel-apl 0000:00:0e.0: error: sof_probe_work failed err: -2
```
So, I thought to myself "I know Linux, I can fix it!". I've gotten access to the host, and I have the firmware files ready, but I can't copy them into `/lib/firmware` because the file-system is read-only.

I absolutely get that you don't want to ship binary blobs, but can we at least get some way of installing them ourselves when needed?