home-assistant/operating-system #2552: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1 Spamming console

pete-leese, 2023-05-17:
### Describe the issue you are experiencing

I have Coral TPU and when device is plugged in and is in use within HA the console is getting spammed with the following logs:

The TPU is functioning perfectly, but the following errors are visible on the local console of the server.

xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 13
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
xhci_hcd 0000:13:00.0: Error Transfer event TRB DMA prt not part of current TD ep_index 2 comp_code 1
 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect Coral USB TPU
2. Wait a short time for activity
3. observe local output on server.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### System information

N/A

### Additional information

Running on HP MicroServer Gen8 ESXi 6.7 Update 3 hypervisor

agners, 2023-05-22:
Since you mention ESXi, I assume you are using the image optimized for virtualization (OVA)?

Kinda hard to tell who's at fault here. The error comes from the upstream Linux USB 3.0 driver (xhci). I wonder if the same error appears when running Home Assistant OS on that server directly :thinking:

pete-leese, 2023-05-22:
thanks for the reply. 

I can confirm that this is the OVA image, the underlying hardware is a HP Gen8 microserver. 

I have upgraded to esx 6.7 and have the VM configured as other 3.x to enable usb pass through of the coral TPU. 

Not really able to test it running directly on the server unfortunately as it runs other workloads. 

Is there any other debug you would like me to perform ? 

Thanks.

ericklindberg, 2023-05-26:
I also get this error. I've been trying to figure it out for a couple months, I use Proxmox though. If I pass the entire PCI controller to the VM the error goes away. If I only pass the USB to the VM, I get spammed with this error. Everything is functioning as it should, though I have to restart every other day as the RAM gets completely eaten up by the log.

pete-leese, 2023-05-26:
Yeah I find my VM performance really tails off after a few days and need to reboot. 

Have tried to pass through the usb controller but all that did was totally frag my esx install and had to reinstall the hypervisor.

ericklindberg, 2023-05-26:
This thread re-ignited my desire to fix the issue. I ended up spinning up Frigate in a separate LXC. I was able to pass through the Coral to that LXC without issue. I apologize, I am more of a lurker here so if formatting is off that's why.

Here is my docker-compose.yml file.

`version: '3.9'
services:
  frigate:
    container_name: frigate
    privileged: true
    restart: unless-stopped
    image: blakeblackshear/frigate:stable-amd64
    devices:
      - /dev/bus/usb/002/003:/dev/bus/usb/002/003
    volumes:
      - /etc/localtime:/etc/localtime:ro
      - /mnt/usbstorage/frigate/config/frigate.yml:/config/config.yml:ro
      - /media/frigate:/media/frigate
      - type: tmpfs # Optional: 1GB of memory, reduces SSD/SD Card wear
        target: /tmp/cache
        tmpfs:
          size: 1000000000
    ports:
      - '5000:5000'
      - '1935:1935'
    environment:
      FRIGATE_RTSP_PASSWORD: 'pass1'
    build:
      context: .
      shm_size: '256m'
    shm_size: '256m'
`
Ran the command:

`$ ls -l /dev/bus/usb/002/003`

Output was 

`crw-rw-r-- 1 root plugdev 189, 130 May 26 10:00 /dev/bus/usb/002/003`

Added to my LXC container config to bind the USB Device(I know, I'll probably have to reconfigure. Didn't want to pass the full bus):

`lxc.cgroup.devices.allow = c 189:514 rwm
lxc.mount.entry = /dev/bus/usb/002/003 dev/bus/usb/002/003 none bind,optional,create=file`

Checked 

In the LXC I added commands

`usermod -aG plugdev root
`
And

`chmod 666 /dev/bus/usb/002/003
`

Configured the frigate.yml as outlined here: [https://docs.frigate.video/configuration/](url). Restarted the LXC, now Frigate works. Now I need to configure Home Assistant to connect, but that won't be too bad. BTW Inferencing speeds went from over 100 to 4 by doing this. Hope this helps!

linedpaper, 2023-08-27:
I'm still having this issue, also using the ova.  Too many issues with the ova I think, it might be time to use a regular os.

adamburgoyne, 2023-10-17:
@ericklindberg Did the LXC container completely solve this issue for you? I'm also getting the exact same error when trying to pass through a USB coral (either by port or device) to my Proxmox HA VM, or any other VM for that matter. 

I was going to look at passing through the entire USB bus as a last resort but I've read that can cause issues in itself so I'd rather not if possible. Does the LXC container handle port emulation differently to a VM on Proxmox? Thanks!

adamburgoyne, 2023-10-19:
For me the solution was to move over to an LXC instead of VM. They both handle USB interactions differently, the VM via USB emulation (unless you pass through the whole PCI bus) and the LXC directly via the host.

I can finally report I no longer see any TPU/USB restarts and as an added bonus my inference speed is down from near 30ms in the VM to 9ms on the LXC. Seems like the better choice unless for some reason you must use a VM, passing through the whole PCI bus can have it's own issues so for me would be a last resort.

Video solution I followed in my post here, was all very easy and took around 15 mins in total: https://github.com/blakeblackshear/frigate/issues/8193

closed 2024-01-25 by github-actions[bot]