home-assistant/operating-system #2554: WiFi card not detected - Intel 9461

rayxxxxx, 2023-05-20:
### Describe the issue you are experiencing

The built-in WiFi card of my BMAX B2 PRO - a Intel 9461 - is not being recognized.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

6.1.28

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install Home Assistant generic image
2. Check if WiFi works (verified through nmcli device wifi)

### Anything in the Supervisor logs that might be useful for us?

```txt
23-05-20 13:55:37 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-05-20 13:55:37 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.MULTIPLE_DATA_DISKS/ContextType.SYSTEM
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-05-20 13:55:38 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-05-20 13:57:03 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/w35l3y/hassio-addons repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/esphome/home-assistant-addon repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/thomasmauerer/hassio-addons repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/brenner-tobias/ha-addons repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/dianlight/hassio-addons repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sabeechen/hassio-google-drive-backup repository
23-05-20 14:00:53 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
23-05-20 14:00:55 INFO (MainThread) [supervisor.store] Loading add-ons from store: 80 all - 0 new - 0 remove
23-05-20 14:00:55 INFO (MainThread) [supervisor.store] Loading add-ons from store: 80 all - 0 new - 0 remove
23-05-20 14:27:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-05-20 14:27:28 INFO (MainThread) [supervisor.host.info] Updating local host information
23-05-20 14:27:29 INFO (MainThread) [supervisor.host.services] Updating service information
23-05-20 14:27:29 INFO (MainThread) [supervisor.host.network] Updating local network information
23-05-20 14:27:29 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-05-20 14:27:29 INFO (MainThread) [supervisor.host.manager] Host information reload completed
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_IPV6_ERROR/ContextType.DNS_SERVER
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.DNS_SERVER_FAILED/ContextType.DNS_SERVER
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.MULTIPLE_DATA_DISKS/ContextType.SYSTEM
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.IPV4_CONNECTION_PROBLEM/ContextType.SYSTEM
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.check] System checks complete
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-05-20 14:55:38 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
23-05-20 14:57:04 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
[... 195 lines cut ...]
version_api | ok
installed_addons | File editor (5.6.0), Samba share (10.0.1), Advanced SSH & Web Terminal (14.1.0), Grafana (8.2.2), Samba Backup (5.2.0), FTP (4.6.3), Mosquitto broker (6.2.1), InfluxDB (4.6.0), Cloudflare Tunnel (1.7.0), Plex Media Server (3.2.4), Uptime Kuma (0.7.3)

</details>

<details><summary>Dashboards</summary>

dashboards | 10
-- | --
resources | 8
views | 36
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 19 maggio 2023, 08:55
-- | --
current_recorder_run | 20 maggio 2023, 15:33
estimated_db_size | 164.42 MiB
database_engine | sqlite
database_version | 3.40.1

</details>

agners, 2023-05-20:
Can you share the full dmesg?

Also, what exact version of OS 11 are you using?

rayxxxxx, 2023-05-21:
Home Assistant OS 11.0.dev20230511

[    0.000000] Linux version 6.1.28 (builder@9c7f2bed4880) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g553e6c546) 11.3.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT_DYNAMIC Thu May 11 23:04:09 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt4)/bzImage root=PARTUUID=a3ec664e-32ce-4665-95ea-7ae90ce9aa20 rootwait zram.enabled=1 zram.num_devices=3 net.naming-scheme=v250 systemd.machine_id=e9013ebc84d943ec99032590e8a5be7d fsck.repair=yes console=tty1 rauc.slot=B
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[3]:  576, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  640, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 704 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 2032
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000003dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000003e000-0x000000000003ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000040000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000077153fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000077154000-0x0000000079a18fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079a19000-0x0000000079a39fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000079a3a000-0x0000000079aa0fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000079aa1000-0x0000000079e08fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000079e09000-0x0000000079ea6fff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000079ea7000-0x000000007a775fff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a776000-0x000000007a821fff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007a822000-0x000000007abfffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ac00000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d0000000-0x00000000d0ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d3709fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe042000-0x00000000fe044fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe900000-0x00000000fe902fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI 2.0=0x79a23000 ACPI=0x79a23000 TPMFinalLog=0x79a6f000 SMBIOS=0x79cc1000 SMBIOS 3.0=0x79cc0000 MEMATTR=0x740ee018 ESRT=0x761cf498 
[    0.000000] SMBIOS 3.2.0 present.
[    0.000000] DMI: AMI Intel/Intel, BIOS V1.0_223 03/13/2023
[    0.000000] tsc: Detected 1497.600 MHz processor
[    0.000022] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000028] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000052] last_pfn = 0x280000 max_arch_pfn = 0x400000000
[    0.000096] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000275] last_pfn = 0x7ac00 max_arch_pfn = 0x400000000
[    0.007527] esrt: Reserving ESRT space from 0x00000000761cf498 to 0x00000000761cf4d0.
[    0.007538] e820: update [mem 0x761cf000-0x761cffff] usable ==> reserved
[    0.007582] Using GB pages for direct mapping
[    0.008078] Secure boot disabled
[    0.008082] ACPI: Early table checksum verification disabled
[    0.008088] ACPI: RSDP 0x0000000079A23000 000024 (v02 ALASKA)
[    0.008096] ACPI: XSDT 0x0000000079A230C0 0000EC (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.008106] ACPI: FACP 0x0000000079A30310 000114 (v06 ALASKA A M I    01072009 AMI  00010013)
[    0.008116] ACPI: DSDT 0x0000000079A23250 00D0BD (v02 ALASKA A M I    01072009 INTL 20160930)
[    0.008123] ACPI: FACS 0x0000000079AA0080 000040
[    0.008128] ACPI: FPDT 0x0000000079A30430 000044 (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.008134] ACPI: FIDT 0x0000000079A30480 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.008139] ACPI: MSDM 0x0000000079A30520 000055 (v03 ALASKA A M I    01072009 AMI  00010013)
[    0.008145] ACPI: MCFG 0x0000000079A30580 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.008151] ACPI: DBG2 0x0000000079A305C0 000072 (v00 INTEL  GLK-SOC  00000003 BRXT 0100000D)
[    0.008156] ACPI: DBGP 0x0000000079A30640 000034 (v01 INTEL  GLK-SOC  00000003 BRXT 0100000D)
[    0.008162] ACPI: HPET 0x0000000079A30680 000038 (v01 INTEL  GLK-SOC  00000003 BRXT 0100000D)
[    0.008168] ACPI: LPIT 0x0000000079A306C0 00005C (v01 INTEL  GLK-SOC  00000003 BRXT 0100000D)
[... cut ...]
[ 8041.831126] audit: type=1334 audit(1684679399.487:389): prog-id=96 op=UNLOAD
[ 8041.831168] audit: type=1334 audit(1684679399.487:390): prog-id=95 op=UNLOAD
[ 8041.831172] audit: type=1334 audit(1684679399.487:391): prog-id=94 op=UNLOAD
[ 8042.182165] audit: type=1334 audit(1684679399.838:392): prog-id=99 op=UNLOAD
[ 8042.182176] audit: type=1334 audit(1684679399.838:393): prog-id=98 op=UNLOAD
[ 8042.182179] audit: type=1334 audit(1684679399.838:394): prog-id=97 op=UNLOAD
[13424.696348] audit: type=1334 audit(1684684782.455:395): prog-id=100 op=LOAD

agners, 2023-05-22:
> Home Assistant OS 11.0.dev20230511

11.0.dev20230517 comes with newer/update firmwares for Intel WiFi along with some additional firmwares, can you update to that release/install that release from https://os-builds.home-assistant.io/11.0.dev20230517/?

agners, 2023-05-22:
The driver for Intel Wireless-AC 9461 devices is compiled and the firmware should be present.

It looks like the card isn't detected at all from the logs. What kind of system is that? Does the card work with other operating systems?

rayxxxxx, 2023-05-22:
the wifi card worked with windows 11
lspci: command not found

rayxxxxx, 2023-05-25:
Sorry, can you tell me how to do it?

agners, 2023-05-26:
You can install it from scratch the very same way you installed the original image.

Otherwise you can force an upgrade using the following command in the SSH/Web Terminal Add-on:

```
ha os update --version=10.2
```

rayxxxxx, 2023-05-26:
I forced upgrade with ha os update --version=10.2 but I doesn't work.
Now, in supervisor log file I have an error.

[supervisor_2023-05-26T12-44-03.877Z.log](https://github.com/home-assistant/operating-system/files/11575344/supervisor_2023-05-26T12-44-03.877Z.log)

Niaphtaed, 2023-08-03:
Hi,
I got the same problem and would be interessted if you solved the issue @rayxxxxx or did you still got the issue.
Btw. I‘m running HA OS 10.3

rayxxxxx, 2023-08-05:
@Niaphtaed I still got the issue

rayxxxxx, 2023-10-13:
Operating System 11.0 solved the issue.

closed 2023-10-13 by rayxxxxx