home-assistant/operating-system #2558: HA in kvm won't shut down on host shut down

aapeliv, 2023-05-23:
### Describe the issue you are experiencing

I'm running the generic-x86-64 image in KVM on a Ubuntu host. Whenever I shut down the computer I have to wait an extra 2 minutes for HA to be killed because it won't shut down by itself like my other VMs.

This seems related to #1896, but I'm not running the OVA image (I need the WiFi drivers so I can connect ESP32s).

How can I modify the systemd config or work around this?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Run in VM
2. Turn host off
3. Have to wait 2 minutes

### Anything in the Supervisor logs that might be useful for us?

```txt
nope
```

### Anything in the Host logs that might be useful for us?

```txt
nope
```

agners, 2023-05-23:
Use the OVA image, it is meant to be used for virtualization and shuts down automatically on ACPI events.

aapeliv, 2023-05-23:
So how does one get the intel wifi drivers on the OVA image?

agners, 2023-05-23:
You are using WiFi from within your virtual machine? What is your use case?

closed 2023-05-25 by agners

aapeliv, 2023-05-25:
Awesome, thank you!!