home-assistant/operating-system #2573: Upgrade to HAOS 10.2 from 10.1 fails on Odroid M1

jldg, 2023-06-03:
### Describe the issue you are experiencing

I have been unable to upgrade to HAOS 10.2 from 10.1 on my Odroid M1. Each time I try, it gets to the point where it restarts Home Assistant but is never able to successfully boot. I have the M1 connected to a display and there is no signal to the display after it attempts to reboot. If I unplug the M1 and plug it back in, it will boot up, but it's back on 10.1. On occasion it will successfully reboot, but didn't upgrade to 10.2

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Go to settings
2. Click on Home Assistant Operating System under updates
3. Click Install
4.  Home assistant "installs" the update and reboots.
5. Odroid M1 does not successfully boot.

### Anything in the Supervisor logs that might be useful for us?

```txt
[32m23-06-02 20:46:18 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1[0m
[32m23-06-02 20:46:18 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1[0m
[32m23-06-02 20:46:18 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.host.services] Updating service information[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.host.network] Updating local network information[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor[0m
[32m23-06-02 20:46:19 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2022.11.0[0m
[32m23-06-02 20:46:19 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin[0m
[32m23-06-02 20:46:21 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/aarch64-hassio-cli with version 2022.11.0 - 172.30.32.5[0m
[32m23-06-02 20:46:21 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1[0m
[32m23-06-02 20:46:21 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin[0m
[32m23-06-02 20:46:23 INFO (SyncWorker_1) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1 - 172.30.32.3[0m
[32m23-06-02 20:46:23 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf[0m
[32m23-06-02 20:46:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.04.1[0m
[32m23-06-02 20:46:23 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin[0m
[32m23-06-02 20:46:25 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.04.1 - 172.30.32.4[0m
[32m23-06-02 20:46:25 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0[0m
[32m23-06-02 20:46:25 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0[0m
[32m23-06-02 20:46:25 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin[0m
[32m23-06-02 20:46:27 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0 - Host[0m
[32m23-06-02 20:46:27 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json[0m
[32m23-06-02 20:46:27 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets[0m
[32m23-06-02 20:46:27 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/odroid-m1-homeassistant with version 2023.5.4[0m
[32m23-06-02 20:46:27 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 10.1 / BootSlot A[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/1e14b3fb repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/03cabcc9 repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/ae6e943c repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/f4f71350 repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/cebe7a76 repository[0m
[32m23-06-02 20:46:30 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/5c53de3b repository[0m
[32m23-06-02 20:46:33 INFO (MainThread) [supervisor.store] Loading add-ons from store: 90 all - 90 new - 0 remove[0m
[32m23-06-02 20:46:33 INFO (MainThread) [supervisor.addons] Found 16 installed add-ons[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-whisper with version 0.2.0[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to sabeechen/hassio-google-drive-backup-aarch64 with version 0.110.4[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_4) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.2.1[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_6) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-zwave_js with version 0.1.83[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_5) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-silabs-multiprotocol with version 1.1.3[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-matter-server with version 4.5.1[0m
[32m23-06-02 20:46:33 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to tsightler/ring-mqtt with version 5.2.2[0m
[... 240 lines cut ...]
resources | 3
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 24, 2023 at 10:20 PM
-- | --
current_recorder_run | June 2, 2023 at 8:47 PM
estimated_db_size | 303.46 MiB
database_engine | sqlite
database_version | 3.40.1

</details>

<details><summary>Sonoff</summary>

version | 3.5.1 (122751c)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

agners, 2023-06-05:
I've just tried to reproduce that on my ODROID-M1, but wasn't able to: Updating from 10.1 to 10.2 worked without problems, the system came up with the proper OS version reported.

What type of boot device are you using? Is this reproducible?

jldg, 2023-06-08:
I'm using a sandisk microsd card, but I also had an emmc attached. I gave up on trying to update and was going to finally switch to the emmc. That didn't go well. I flashed the 10.2 image to the emmc and upon the first boot my power went out (a squirrel jumped on the transformer next door and got fried). When I booted it back up again, home assistant was stuck on "preparing home assistant". Petit boot wouldn't come up either, I popped the microsd card back in and that would no longer boot. It kept booting straight to the emmc. I ended up removing the emmc from the socket and the microsd card booted right up. I figured I would try to update again, and it worked. There must be something wrong with having both the microsd and emmc installed at the same time.  I guess this issue can be closed. On a side note, I've tried flashing the spiupdate_odroidm1_20220304.img.xz image to a different microsd to get petitboot back, but it's not doing anything. I was previously successful upgrading to spiboot-20230328. If you have any suggestions on how to get it to work, I'm open to them.

closed 2023-09-14 by github-actions[bot]