home-assistant/operating-system #2574: USB device fails initialization on boot after upgrade to HAOS 10.2, works after USB reconnect

geekifier, 2023-06-03:
### Describe the issue you are experiencing

I [originally opened this issue](https://github.com/hassio-addons/addon-nut/issues/300) in the `addon-nut` repo, but was asked to report it here.

After upgrading the Home Assistant OS from 9.5 to 10.2, and NUT can no longer connect to the UPS after the host system is rebooted. This is caused by the fact, that the USB device of the UPS is no longer recognized by HA OS after reboot.

The workaround is to unplug, then plug the USB device back, then it appears in `dmesg` and the NUT add-on starts successfully. There were no changes in my config, other than upgrading from HAOS 9.5 -> 10.2.

I am able to reproduce this failure on-demand by rebooting the system, so please let me know what other info I can provide to help with this.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Restart the Home Assistant OS, observe the following in the NUT addon logs:

```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/nut.sh
[15:56:49] INFO: Setting mode to netserver...
[15:56:50] INFO: Generating /etc/nut/upsd.users...
[15:56:50] INFO: Configuring user: #####
[15:56:51] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[15:56:52] INFO: Configuring user: #####
[15:56:52] INFO: Password is NOT in the Have I Been Pwned database! Nice!
[15:56:54] INFO: Configuring Device named 1500PFCLCD...
[15:56:54] INFO: Starting the UPS drivers...
No matching HID UPS found
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
```

Also observe the device does not show up in kernel log or in `lsusb`.

```bash
$ lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 008: ID 0658:0200
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 001: ID 1d6b:0003

$ dmesg | grep "usb 1-1.4"
[    4.627832] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[... 126 lines cut ...]
   0.047170	Logged into UPS 1500PFCLCD@localhost
   3.607610	User upsclient@192.168.1.27 logged into UPS [1500PFCLCD]
```

Also confirmed via kernel log and lsusb.

```bash

$ dmesg | grep "usb 1-1.4"
[...]
[  210.388780] usb 1-1.4: new full-speed USB device number 9 using xhci_hcd
[  210.535817] usb 1-1.4: New USB device found, idVendor=0764, idProduct=0501, bcdDevice= 0.01
[  210.535830] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[  210.535837] usb 1-1.4: Product: CP1500PFCLCD
[  210.535843] usb 1-1.4: Manufacturer: CPS
[  210.535849] usb 1-1.4: SerialNumber: 000000000000

$ lsusb

Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 008: ID 0658:0200
Bus 001 Device 002: ID 2109:3431
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 009: ID 0764:0501
```

geekifier, 2023-06-04:
I downgraded the OS to 9.5 and the issue persists, so it's not a regression. I wonder if there was some firmware update for rpi4 in one of the releases?

To help with troubleshooting, I shut down the Pi completely and power-cycled the PoE on the switch port. The next several reboots, the USB devices got detected properly. However, after that the failure came back, though it seems to work _sometimes_.

The device in question is `usb 1-1.4`.

### When it worked

```bash

$ dmesg | grep -i usb
[    0.111521] usbcore: registered new interface driver usbfs
[    0.111584] usbcore: registered new interface driver hub
[    0.111649] usbcore: registered new device driver usb
[    0.111864] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.112022] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    1.619802] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 1
[    1.624249] xhci_hcd 0000:01:00.0: new USB bus registered, assigned bus number 2
[    1.624283] xhci_hcd 0000:01:00.0: Host supports USB 3.0 SuperSpeed
[    1.624538] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
[    1.624573] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.624601] usb usb1: Product: xHCI Host Controller
[    1.624622] usb usb1: Manufacturer: Linux 6.1.21-v7l xhci-hcd
[    1.624644] usb usb1: SerialNumber: 0000:01:00.0
[    1.625201] hub 1-0:1.0: USB hub found
[    1.625912] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.01
[    1.625946] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.625974] usb usb2: Product: xHCI Host Controller
[    1.625996] usb usb2: Manufacturer: Linux 6.1.21-v7l xhci-hcd
[    1.626017] usb usb2: SerialNumber: 0000:01:00.0
[    1.626517] hub 2-0:1.0: USB hub found
[    1.628323] usbcore: registered new interface driver uas
[    1.628413] usbcore: registered new interface driver usb-storage
[    1.628511] usbcore: registered new interface driver usbserial_generic
[    1.628555] usbserial: USB Serial support registered for generic
[    1.637894] usbcore: registered new interface driver usbhid
[    1.637904] usbhid: USB HID core driver
[    1.727663] usb usb2-port1: over-current change #1
[    1.907397] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.967546] usb usb2-port2: over-current change #1
[    2.088623] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.20
[    2.088655] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.088675] usb 1-1: Product: USB2.0 Hub
[    2.089938] hub 1-1:1.0: USB hub found
[    2.198361] usb 1-1-port1: over-current change #1
[    2.207622] usb usb2-port3: over-current change #1
[    2.439440] usb 1-1-port2: over-current change #1
[    2.467485] usb usb2-port4: over-current change #1
[    2.679480] usb 1-1-port3: over-current change #1
[    2.717624] usb usb2-port1: over-current change #2
[    2.922308] usb 1-1-port4: over-current change #1
[    2.957512] usb usb2-port2: over-current change #2
[    3.157844] usb 1-1-port1: over-current change #2
[    3.197511] usb usb2-port3: over-current change #2
[    3.408151] usb 1-1-port2: over-current change #2
[    3.437865] usb usb2-port4: over-current change #2
[    3.657852] usb 1-1-port3: over-current change #2
[    4.197416] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[    4.343949] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[    4.343979] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    4.757434] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[    4.904019] usb 1-1.4: New USB device found, idVendor=0764, idProduct=0501, bcdDevice= 0.01
[    4.904039] usb 1-1.4: New USB device strings: Mfr=3, Product=1, SerialNumber=2
[    4.904050] usb 1-1.4: Product: CP1500PFCLCD
[    4.904059] usb 1-1.4: Manufacturer: CPS
[    4.904068] usb 1-1.4: SerialNumber: 000000000000
[    4.916000] hid-generic 0003:0764:0501.0001: hiddev96,hidraw0: USB HID v1.10 Device [CPS CP1500PFCLCD] on usb-0000:01:00.0-1.4/input0
[    5.073465] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[    5.073550] usbcore: registered new interface driver cdc_acm
[    5.073556] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[... 62 lines cut ...]
[    5.447897] usb 1-1.4: Device not responding to setup address.
[    5.667610] usb 1-1.4: device not accepting address 5, error -71
[    5.767645] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[    5.767879] usb 1-1.4: Device not responding to setup address.
[    5.997761] usb 1-1.4: Device not responding to setup address.
[    6.217605] usb 1-1.4: device not accepting address 6, error -71
[    6.217915] usb 1-1-port4: unable to enumerate USB device
[    6.218299] usb 1-1-port1: over-current change #2
[    6.458478] usb 1-1-port2: over-current change #2
[    6.698358] usb 1-1-port3: over-current change #2
[    7.237603] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[    7.374637] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[    7.374649] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    7.377589] usb 1-1-port4: over-current change #2
[    7.767923] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[    7.768322] usbcore: registered new interface driver cdc_acm
[... cut ...]
[   75.227370] usb 1-1.3: new full-speed USB device number 8 using xhci_hcd
[   75.373939] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[   75.373952] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   75.379406] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
```

At this point I am not really sure where to look next, I have a workaround (though an annoying one), so I could live with this behavior, but it sure is not satisfying not getting to the bottom of the issue :).

agners, 2023-06-05:
> I downgraded the OS to 9.5 and the issue persists, so it's not a regression. I wonder if there was some firmware update for rpi4 in one of the releases?

Home Assistant OS does not update the EEPROM firmware of the Raspberry Pi, so that can't be the reason.

This is probably a timing issue. Maybe using a different port and/or adding a USB hub in between helps?

We use the same kernel sources as the Raspberry Pi OS is using. What would be interesting if you can reproduce the behavior with a vanilla Raspberry Pi OS. If so, I'd recommend to report the problem upstream at https://github.com/raspberrypi/linux/.