home-assistant/operating-system #2581: Cannot add network storage at storage tab

dengke80622, 2023-06-09:
### Describe the issue you are experiencing

Win10 SMB v1 & v2 enabled
Openwrt Samba setting:
max protocol = SMB2
min protocol = SMB2
Both SMB sharings can be used on other win10 PCs, phones and tablets without any issue

On home assistant storage page:
1. win10 smb sharing can be mounted via storage page
2. openwrt smb sharing cannot be mounted with error: Mounting openwrt_smb did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-openwrt_smb.mount for details.

And then I ssh into HAOS with port 22222
1. Creat Folder: mkdir /mnt/data/supervisor/media/ssd960gb
2. Succeed mount openwrt sharing: mount -t cifs -o vers=2.0,noserverino,username=root,password=root,domain=WORKGROUP "//192.168.10.2/ssd960gb" /mnt/data/supervisor/media/ssd960gb

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.set openwrt smb sharing protocol to SMB2
2.add network location within storage tab
3. error occours

### Anything in the Supervisor logs that might be useful for us?

```txt
23-06-09 14:40:16 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: openwrtsmbtesting
23-06-09 14:40:16 INFO (MainThread) [supervisor.mounts.mount] Creating folder for mount: /data/mounts/openwrtsmbtesting
23-06-09 14:40:16 ERROR (MainThread) [supervisor.mounts.mount] Mounting openwrtsmbtesting did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-openwrtsmbtesting.mount for details.
23-06-09 14:42:09 INFO (MainThread) [supervisor.mounts.manager] Creating or updating mount: openwrtsmbtesting
23-06-09 14:42:09 ERROR (MainThread) [supervisor.mounts.mount] Mounting openwrtsmbtesting did not succeed. Check host logs for errors from mount or systemd unit mnt-data-supervisor-mounts-openwrtsmbtesting.mount for details.
```

### Anything in the Host logs that might be useful for us?

```txt
keyword: cifs
Jun 09 06:56:09 homeassistant kernel: CIFS: Attempting to mount \\192.168.10.2\cmr1tb
Jun 09 06:56:45 homeassistant kernel: CIFS: VFS: Use of the less secure dialect vers=1.0 is not recommended unless required for access to very old servers
Jun 09 06:56:45 homeassistant kernel: CIFS: Attempting to mount \\192.168.10.2\ssd960gb
Jun 09 06:56:46 homeassistant kernel: CIFS: VFS: cifs_mount failed w/return code = -95

keyword: smb
Jun 09 06:53:45 homeassistant systemd-coredump[216192]: Process 216189 (smbd) of user 0 dumped core.
Jun 09 06:53:46 homeassistant systemd-coredump[216200]: Process 216194 (smbd) of user 0 dumped core.
Jun 09 06:53:48 homeassistant systemd-coredump[216209]: Process 216207 (smbd) of user 0 dumped core.
Jun 09 06:53:49 homeassistant kernel: audit: type=1701 audit(1686293629.144:896): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=216212 comm="smbd" exe="/usr/sbin/smbd" sig=6 res=1
Jun 09 06:53:49 homeassistant systemd-coredump[216218]: Process 216212 (smbd) of user 0 dumped core.
Jun 09 06:53:50 homeassistant kernel: audit: type=1701 audit(1686293630.233:903): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=216220 comm="smbd" exe="/usr/sbin/smbd" sig=6 res=1
Jun 09 06:53:50 homeassistant systemd-coredump[216226]: Process 216220 (smbd) of user 0 dumped core.
Jun 09 06:53:51 homeassistant systemd-coredump[216234]: Process 216231 (smbd) of user 0 dumped core.
Jun 09 06:54:07 homeassistant kernel: audit: type=1701 audit(1686293647.640:917): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=216300 comm="smbd" exe="/usr/sbin/smbd" sig=6 res=1
Jun 09 06:54:08 homeassistant systemd-coredump[216303]: Process 216300 (smbd) of user 0 dumped core.
Jun 09 06:54:08 homeassistant kernel: audit: type=1701 audit(1686293648.728:924): auid=4294967295 uid=0 gid=0 ses=4294967295 subj=docker-default pid=216308 comm="smbd" exe="/usr/sbin/smbd" sig=6 res=1
Jun 09 06:54:09 homeassistant systemd-coredump[216311]: Process 216308 (smbd) of user 0 dumped core.
Jun 09 06:54:10 homeassistant systemd-coredump[216319]: Process 216316 (smbd) of user 0 dumped core.
Jun 09 06:54:11 homeassistant systemd-coredump[216328]: Process 216325 (smbd) of user 0 dumped core.
```

### System information

## System Information

version | core-2023.6.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.3
os_name | Linux
os_version | 6.1.29
arch | x86_64
timezone | Asia/Shanghai
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4995
Installed Version | 1.32.1
Stage | running
Available Repositories | 1352
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.2
-- | --
update_channel | stable
supervisor_version | supervisor-2023.06.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 234.0 GB
disk_used | 15.7 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), Hass.io FRP client (0.0.1), Terminal & SSH (9.7.1), MariaDB (2.6.1), Traccar (0.19.1), ESPHome (2023.5.5), Advanced SSH & Web Terminal (15.0.2), Mosquitto broker (6.2.1), Node-RED (14.2.2), Samba share (10.0.2), Zigbee2mqtt (1.25.1-1), chrony (2.6.0), go2rtc (1.5.0), Plex Media Server (3.2.4)

</details>

<details><summary>Dashboards</summary>

dashboards | 2
-- | --
resources | 6
views | 10
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 22, 2023 at 07:28
-- | --
current_recorder_run | June 8, 2023 at 23:14
estimated_db_size | 311.88 MiB
database_engine | mysql
database_version | 10.6.12

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.7.9
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 1
total_devices | 20

</details>

### Additional information

do an upgrade on vmware hass os host, same issue with 2023.6.1

Johnjr00, 2023-06-11:
Just wanted to chime in and say I have the exact same issue with an Asus router.

nstrelow, 2023-06-11:
Intersting, same for me, using the Netgear R7800 router, without a user or password.

agners, 2023-06-12:
The Linux kernel requests SMB2.1 or later by default. Earlier versions must be selected by mount options. However, as mount options are set by the Supervisor, this needs to be resolved in Supervisor.

In general, if you have the option on your NAS to enable SMB2.1 or 3.0, I recommend doing that.

If not, you'll have to wait for an option to enable earlier versions. There is already an open issue/feature request for this: https://github.com/home-assistant/supervisor/issues/4364.

closed 2023-06-12 by agners