home-assistant/operating-system #2585: cifs 'stale file handle'

JustinScott, 2023-06-09:


agners, 2023-06-12:
What hardware/board are you running on (or rather, which Home Assistant OS image did you use)?

If you just updated recently, it might be related to the restart needed. Did you try/can you still reproduce after a full system reboot?

JustinScott, 2023-06-13:
I'm running HAOS in a VM on my Unraid server. I think I used 'haos_ova-9.0.qcow2' for the install. I just did a full reboot 'red button' and I get the same result. I can write one file to the share, then I get errors.

agners, 2023-06-13:
What target are you mounting? What type of system is that?

I haven't seen such an issue, probably some weird behavior with the server your are using.

JustinScott, 2023-06-13:
The share is on my Unraid server. 

```
Unraid is a proprietary Linux-based operating system designed
to run on home media server setups that operates as a 
network-attached storage device, application server, and virtualization host.
```

This is the target: \\\\192.168.2.100\haos-media 
I don't have issues accessing it from my Windows PC. Are there any special settings HAOS requires the share to have? This is the share config page in Unraid.

[image]

agners, 2023-06-14:
What firmware version on Unraid are you using? Could that be the problem?

> You are running Unraid 6.9.  That version of Unraid uses NFSv3 that is know to have stale file handle issues.  Upgrade to 6.10 that uses NFSv4 that does not have the stale file handle issues. 

From: https://forums.unraid.net/topic/127609-nfs-stale-file-handle/

wanno-drijfhout, 2023-06-18:
I'm getting similar problems with FRITZ!Box SMB file share. It seems to occur [more often](https://community.home-assistant.io/t/samba-error-with-fritz-nas-cifs-stale-file-handle/567110). According to [this](https://askubuntu.com/a/1265165) source it is indeed a matter of adding the `noserverino` flag to the mount properties.

**This is the case!***

I've used SSH and portainer to dive as deep into HAOS as possible and hack the solution into my installation. My journey...

First I had to unhide hidden containers per [these instructions](https://community.home-assistant.io/t/point-media-dirs-at-nas-share/227698/5). Then I logged on the `homeassistant` container and opened a console.

What I did notice was that the SMB mounts are actually made with the `serverino` attribute, which contradicts the `noserverino`, what we (supposedly) need.
```sh
$ cat /proc/mounts | grep share
//10.69.1.1/NAS/SERVICE/www /data/share/www cifs rw,relatime,vers=3.1.1,cache=strict,username=homeassistant,uid=0,noforceuid,gid=0,noforcegid,addr=10.69.1.1,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=65536,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1,closetimeo=5 0 0
```

A bit hackish, but I tried to add the `noserverinfo` attribute to https://github.com/home-assistant/supervisor/blob/65545e721850a014f3085d0172322475cc68f948/supervisor/mounts/mount.py#L361 locally (`/usr/supervisor/...`). I then recreated the mount and restarted the Home Assistant machine. Subsequently, I no longer had my problem with "stale file handle". Also `/proc/mounts` didn't show the attribute `serverino` anymore. (It didn't show `noserverino` either, by the way, but that could be expected.) So, my hack seems to have had the desired effect!

Now the question is what the proper solution ought to be...

wanno-drijfhout, 2023-06-18:
I'm getting similar problems with FRITZ!Box SMB file share. It seems to occur [more often](https://community.home-assistant.io/t/samba-error-with-fritz-nas-cifs-stale-file-handle/567110). According to [this](https://askubuntu.com/a/1265165) source it is indeed a matter of adding the `noserverino` flag to the mount properties.

**This is the case!***

I've used SSH and portainer to dive as deep into HAOS as possible and hack the solution into my installation. My journey...

First I had to unhide hidden containers per [these instructions](https://community.home-assistant.io/t/point-media-dirs-at-nas-share/227698/5). Then I logged on the `hassio_supervisor` container and opened a console.

What I did notice was that the SMB mounts are actually made with the `serverino` attribute, which contradicts the `noserverino`, what we (supposedly) need.
```sh
$ cat /proc/mounts | grep share
//10.69.1.1/NAS/SERVICE/www /data/share/www cifs rw,relatime,vers=3.1.1,cache=strict,username=homeassistant,uid=0,noforceuid,gid=0,noforcegid,addr=10.69.1.1,file_mode=0755,dir_mode=0755,soft,nounix,serverino,mapposix,rsize=65536,wsize=65536,bsize=1048576,echo_interval=60,actimeo=1,closetimeo=5 0 0
```

A bit hackish, but I tried to add the `noserverinfo` attribute to https://github.com/home-assistant/supervisor/blob/65545e721850a014f3085d0172322475cc68f948/supervisor/mounts/mount.py#L361 locally (`/usr/src/supervisor/supervisor/mounts/mount.py`). I then recreated the mount and restarted the Home Assistant machine. Subsequently, I no longer had my problem with "stale file handle". Also `/proc/mounts` didn't show the attribute `serverino` anymore. (It didn't show `noserverino` either, by the way, but that could be expected.) So, my hack seems to have had the desired effect!

Now the question is what the proper solution ought to be...

closed 2023-06-21 by frenck