home-assistant/operating-system #2595: Possible Memory Leak in 10.2

backcountrymountains, 2023-06-14:
### Describe the issue you are experiencing

Picture worth at least 1024 words:
[image]

Image description: flat line of memory use until I upgraded to HAOS 10.2, then chaos.

I updated my HAOS to 10.x on June 7th around 4pm. Since then my memory allocation has been drawing nice graphs of sledding hills. It has caused out of memory reboots when it gets to ~90%. I was on OS version 6 (maybe?) until June 6 and didn't experience the issue.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Use HAOS 10.2. Maybe it gets worse when I have my only camera feed open using go2rtc? I haven't started disabling things to try and pin it down.

### Anything in the Supervisor logs that might be useful for us?

```txt
.log and ```homeassistant.log.1``` weren't super informative because the machine died with out of memory error and didn't manage to write anything useful to those logs; the logs just show the last thing that was done successfully: 
```Jun 13 09:14:41 homeassistant addon_core_configurator[471]: INFO:2023-06-13 03:14:41,777:hass_configurator.configurator:127.0.0.1 - "GET / HTTP/1.1" 200 -``` 
and then the new log just shows normal startup.

I did manage to do the top secret ssh on port 22222 and use journalctl to find:
```
Jun 13 09:14:59 homeassistant hassio_supervisor[471]: 23-06-13 03:14:59 WARNING (MainThread) [supervisor.homeassistant.websocket] Connection is closed
Jun 13 09:14:58 homeassistant systemd[1]: docker-7f399a191dea38b318f7e9419a85423dbde2229c5ef7cc972154f53a3b648b66.scope: A process of this unit has been killed by the OOM killer.
Jun 13 09:14:58 homeassistant kernel: Out of memory: Killed process 267303 (python3) total-vm:3600048kB, anon-rss:1293204kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:5612kB oom_score_adj:-300
Jun 13 09:14:58 homeassistant kernel: oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/system.slice/docker-7f399a191dea38b318f7e9419a85423dbde2229c5ef7cc972154f53a3b648b66.scope,task=python3,pid=267303,uid=0
Jun 13 09:14:58 homeassistant kernel: [ 377510]     0 377510      400        0    40960        1          -300 sleep
Jun 13 09:14:58 homeassistant kernel: [ 377251]     0 377251      397        1    40960        0           200 sleep
Jun 13 09:14:58 homeassistant kernel: [ 283965]     0 283965     4428        1    65536      130          -300 git
Jun 13 09:14:58 homeassistant kernel: [ 270256]     0 270256     4488        1    69632      129          -300 git
Jun 13 09:14:58 homeassistant kernel: [ 267303]     0 267303   900012   323300  5746688   109901          -300 python3
Jun 13 09:14:58 homeassistant kernel: [ 225903]     0 225903     8823      778   106496      405           200 smbd
Jun 13 09:14:58 homeassistant kernel: [ 193860]     0 193860       53        1    28672        5          -300 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [ 193829]     0 193829       47        0    24576        3          -300 s6-ipcserverd
Jun 13 09:14:58 homeassistant kernel: [ 193821]     0 193821       53        1    28672        5          -300 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [ 193820]     0 193820       53        1    24576        4          -300 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [ 193813]     0 193813       50        0    24576        3          -300 s6-linux-init-s
Jun 13 09:14:58 homeassistant kernel: [ 193811]     0 193811       53        1    28672        5          -300 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [ 193790]     0 193790       52        1    28672        7          -300 s6-svscan
Jun 13 09:14:58 homeassistant kernel: [ 193768]     0 193768   384842      484   225280      210          -998 containerd-shim
Jun 13 09:14:58 homeassistant kernel: [   6702]     0  6702    83081    14029  2764800     3910           200 node
Jun 13 09:14:58 homeassistant kernel: [   6700]     0  6700       53        1    28672        4           200 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [   6517]     0  6517       47        0    24576        3           200 s6-ipcserverd
Jun 13 09:14:58 homeassistant kernel: [   6509]     0  6509       53        1    28672        4           200 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [   6508]     0  6508       53        1    28672        4           200 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [   6502]     0  6502       50        0    24576        3           200 s6-linux-init-s
Jun 13 09:14:58 homeassistant kernel: [   6499]     0  6499       53        1    28672        4           200 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [   6441]     0  6441       53        1    28672        6           200 s6-svscan
Jun 13 09:14:58 homeassistant kernel: [   6422]     0  6422   384778      462   221184      188          -998 containerd-shim
Jun 13 09:14:58 homeassistant kernel: [   5721]     0  5721     1929        0    45056      294           200 nginx
Jun 13 09:14:58 homeassistant kernel: [   5718]     0  5718     2247        0    53248      317           200 nginx
Jun 13 09:14:58 homeassistant kernel: [   5708]     0  5708     1934       25    57344      273           200 nginx
Jun 13 09:14:58 homeassistant kernel: [   5672]     0  5672     1869        1    49152      235           200 nginx
Jun 13 09:14:58 homeassistant kernel: [   5649]     0  5649     9746       67   114688     6176           200 sqlite_web
Jun 13 09:14:58 homeassistant kernel: [   5580]     0  5580    83370     1009   159744     5777           200 esphome
Jun 13 09:14:58 homeassistant kernel: [   5577]     0  5577     2131        0    53248      211           200 nginx
Jun 13 09:14:58 homeassistant kernel: [   5572]     0  5572       51        0    32768        3           200 s6-supervise
Jun 13 09:14:58 homeassistant kernel: [   5571]     0  5571       51        0    32768        4           200 s6-supervise
[... 283 lines cut ...]
error | /config/notexist.yaml not found
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | May 22, 2023 at 10:18 PM
-- | --
current_recorder_run | June 13, 2023 at 10:31 PM
estimated_db_size | 3023.71 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

My ```deps``` folder is empty; no weird stuff going on there.

```kswapd0 Not tainted``` made me chuckle. I'm glad my swap remained pure.

Side note: is there an easier way to get the full system logs and use journalctl? It seems helpful for issues like this where the other logs didn't capture the problem. (Am I doing it wrong?)

Thanks.

agners, 2023-06-14:
This is largely an intended change: With the change from zram to zswap we changed the swapiness level. With that RAM will get used before it gets swapped out. Before, it got swapped out (and compressed) earlier.

See also https://github.com/home-assistant/operating-system/pull/2420.

Do you see any downside/changes in performance?

backcountrymountains, 2023-06-14:
My system reboots every time the memory reaches 90%. Each vertical line in the graph is where my system rebooted due to an OOM error.

I'd prefer it if my system didn't reboot whenever memory filled up.

I installed a fresh HAOS 9.5 and I'm seeing if that doesn't suffer from the same problem.

[image]

backcountrymountains, 2023-06-14:
How is system rebooting due to OOM error ```invalid```?

agners, 2023-06-15:
> My system reboots every time the memory reaches 90%. Each vertical line in the graph is where my system rebooted due to an OOM error.

The overall avaialble memory should be higher with OS 10. So if the very same system, with the very same configuration works with OS 9.5, but runs into OOM with 10, something is buggy indeed.

> I installed a fresh HAOS 9.5 and I'm seeing if that doesn't suffer from the same problem.

A fresh installation is not a good indicator as you don't have all the things running in your production system.

You can downgrade/upgrade using

```
ha os update --version 9.5
```

Ideally you should compare the very same system. I'd expect that you run into OOM situation as well on 9.5. Then it is likely that Home Assistant Core or some add-on has a memory leak indeed.

closed 2023-09-22 by github-actions[bot]