home-assistant/operating-system #2600: No WIFI or ethernet card recognized

gaborbernat, 2023-06-15:
### Describe the issue you are experiencing

When using https://www.amazon.com/dp/B0BS369VMF neither the WIFI or Ethernet is recognized with HAOS 10.2.

GK3 Plus WiFi is: Realtek 8821ce wireless lan 802.11ac PCI-E NIC; RJ45 network port is: Realtek PCIe.GBE Family Controller

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Install and try to connect to network.

### Anything in the Supervisor logs that might be useful for us?

```txt
N.A.
```

### Anything in the Host logs that might be useful for us?

```txt
N.A.
```

### System information

https://www.amazon.com/dp/B0BS369VMF

gaborbernat, 2023-06-19:
This is not a New feature request, is a bug because it was working with 9.5, please reopen.

gaborbernat, 2023-06-19:
ping @agners

Simonh98, 2023-06-22:
The integrated ethernet port didn't work for me either, but I can't say if it worked before version 9.5. For now, I'm using a USB to Ethernet adapter as a temporary workaround.

gaborbernat, 2023-06-22:
Which adapter are you using?

Simonh98, 2023-06-22:
> Which adapter are you using?

I had [this one](https://www.amazon.de/UGREEN-Netzwerk-Ethernet-kompatibel-unterst%C3%BCtzt/dp/B00MYT481C/ref=asc_df_B00MYT481C/?tag=googshopde-21&linkCode=df0&hvadid=309882193183&hvpos=&hvnetw=g&hvrand=2232084470203126862&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9044107&hvtargid=pla-406645188215&psc=1&th=1&psc=1&tag=&ref=&adgrpid=62407011740&hvpone=&hvptwo=&hvadid=309882193183&hvpos=&hvnetw=g&hvrand=2232084470203126862&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9044107&hvtargid=pla-406645188215) lying around, but I think most should work?

fajrstartr, 2023-08-03:
I try to update and lost network connecton.

Thrifty666, 2023-08-04:
I also have this issue with a similar mini pc, looks the same but is called NiPoGi. Only work around I could quickly do was to downgrade to 9.5.

fajrstartr, 2023-08-04:
How did you downgrade. I can access os if I select 'Slot A (OK=1 TRY=0)"  from GNU GRUB. But if machine restarts it boots in "Slot B" again.
Running HA OS Generic.

Thrifty666, 2023-08-04:
I put a 9.5 image on the SSD. Clean install then added bits from previous back up files.
It's a right ball ache.

TempMark, 2023-10-02:
@agners Is there any chance to add also the ethernet driver to one of next HAOS versions. With HAOS 10.5 it didn't work.

sairon, 2023-12-04:
The WiFi driver for bare-metal installs has been added in #2659 (available from HAOS 10.5), since #2951 (only development builds now) it's enabled for all platforms, including VMs in the OVA image.

About the wired ethernet driver, I can't find any public information what driver does this box use. Could anyone get me ideally both complete `dmesg` logs and `lspci` output? It doesn't matter if from HAOS or any other system.

gaborbernat, 2023-12-04:
dmesg:

```
[    0.000000] Linux version 5.15.90 (builder@bf9ef715830d) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g14dcc6f72e) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Tue Jan 24 18:21:21 UTC 2023
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor systemd.machine_id=577eb97384e6462abeeea4eaa51e2320 fsck.repair=yes console=tty1 rauc.slot=A
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x200: 'Protection Keys User registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[9]:  832, xstate_sizes[9]:    8
[    0.000000] x86/fpu: Enabled xstate features 0x207, context size is 840 bytes, using 'compacted' format.
[    0.000000] signal: max sigframe size: 3632
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009e000-0x000000000009efff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007264dfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007264e000-0x000000007574dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007574e000-0x000000007581afff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007581b000-0x00000000758dafff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000758db000-0x0000000075f65fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000075f66000-0x0000000075ffefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000075fff000-0x0000000075ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000076000000-0x0000000079ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aa00000-0x000000007abfffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x00000000803fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000047fbfffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.80 by American Megatrends
[    0.000000] efi: ACPI=0x758b7000 ACPI 2.0=0x758b7014 TPMFinalLog=0x75886000 SMBIOS=0x75ccb000 SMBIOS 3.0=0x75cca000 MEMATTR=0x6e9fe018 ESRT=0x6a45f598 
[    0.000000] SMBIOS 3.5.0 present.
[    0.000000] DMI: Default string Default string/Default string, BIOS 5.27 01/09/2023
[    0.000000] tsc: Detected 1700.000 MHz processor
[    0.000000] tsc: Detected 1689.600 MHz TSC
[    0.000218] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000221] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000231] last_pfn = 0x47fc00 max_arch_pfn = 0x400000000
[    0.000268] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000563] e820: update [mem 0x7c000000-0xffffffff] usable ==> reserved
[    0.000567] e820: update [mem 0x47fc00000-0x483bfffff] usable ==> reserved
[    0.000572] last_pfn = 0x76000 max_arch_pfn = 0x400000000
[    0.012037] esrt: Reserving ESRT space from 0x000000006a45f598 to 0x000000006a45f648.
[    0.012042] e820: update [mem 0x6a45f000-0x6a45ffff] usable ==> reserved
[    0.012062] Using GB pages for direct mapping
[    0.012321] Secure boot could not be determined
[    0.012323] ACPI: Early table checksum verification disabled
[    0.012326] ACPI: RSDP 0x00000000758B7014 000024 (v02 ALASKA)
[    0.012330] ACPI: XSDT 0x00000000758B6728 0000F4 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.012335] ACPI: FACP 0x0000000075818000 000114 (v06 ALASKA A M I    01072009 AMI  01000013)
[    0.012340] ACPI: DSDT 0x00000000757A3000 074670 (v02 ALASKA A M I    01072009 INTL 20200717)
[    0.012343] ACPI: FACS 0x00000000758DA000 000040
[    0.012345] ACPI: FIDT 0x00000000757A2000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.012348] ACPI: SSDT 0x000000007581A000 00038C (v02 PmaxDv Pmax_Dev 00000001 INTL 20200717)
[    0.012350] ACPI: SSDT 0x000000007579C000 005D0B (v02 CpuRef CpuSsdt  00003000 INTL 20200717)
[    0.012353] ACPI: SSDT 0x0000000075799000 002935 (v02 SaSsdt SaSsdt   00003000 INTL 20200717)
[    0.012355] ACPI: SSDT 0x0000000075795000 0033D3 (v02 INTEL  IgfxSsdt 00003000 INTL 20200717)
[    0.012358] ACPI: HPET 0x0000000075819000 000038 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.012360] ACPI: APIC 0x0000000075794000 0001DC (v05 ALASKA A M I    01072009 AMI  01000013)
[    0.012363] ACPI: MCFG 0x0000000075793000 00003C (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.012365] ACPI: SSDT 0x0000000075791000 001F1A (v02 ALASKA Ther_Rvp 00001000 INTL 20200717)
[... cut ...]
[45663.824586] audit: type=1334 audit(1701704096.005:512): prog-id=137 op=LOAD
[45663.824792] audit: type=1334 audit(1701704096.005:513): prog-id=138 op=LOAD
[45663.824846] audit: type=1334 audit(1701704096.005:514): prog-id=139 op=LOAD
[45663.985818] audit: type=1334 audit(1701704096.166:515): prog-id=140 op=LOAD
[45663.985893] audit: type=1334 audit(1701704096.166:516): prog-id=141 op=LOAD
[45663.985917] audit: type=1334 audit(1701704096.166:517): prog-id=142 op=LOAD
```

lscpi command not found.

nelsonjchen, 2024-01-07:
These are from one of these with a 8821CE:

https://www.aliexpress.us/item/3256805048523628.html

[image]

```
[    0.000000] Linux version 6.1.70-haos (builder@d9ccab0ec44a) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g2d89a0f9) 11.4.0, GNU ld (GNU Binutils) 2.38) #1 SMP PREEMPT_DYNAMIC Thu Jan  4 22:04:37 UTC 2024
[    0.000000] Command line: BOOT_IMAGE=(hd0,gpt2)/bzImage root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootwait zram.enabled=1 zram.num_devices=3 systemd.machine_id=dd4fe146d6ee44c0a0b5ac23635faa98 fsck.repair=yes console=tty1 rauc.slot=A
[    0.000000] x86/split lock detection: #AC: crashing the kernel on kernel split_locks and warning on user-space split_locks
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009f000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007617cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000007617d000-0x000000007867cfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007867d000-0x00000000788fcfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000788fd000-0x00000000789fcfff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000789fd000-0x0000000078efefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000078eff000-0x0000000078ffefff] type 20
[    0.000000] BIOS-e820: [mem 0x0000000078fff000-0x0000000078ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000079000000-0x000000007fbfffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000c0000000-0x00000000cfffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed00000-0x00000000fed00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed20000-0x00000000fed7ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000002803fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.70 by American Megatrends
[    0.000000] efi: ACPI=0x789a7000 ACPI 2.0=0x789a7014 TPMFinalLog=0x789aa000 SMBIOS=0x78d7b000 SMBIOS 3.0=0x78d7a000 MEMATTR=0x75089018 ESRT=0x751fd898 
[    0.000000] SMBIOS 3.3.0 present.
[    0.000000] DMI: Intel Corporation Jasper Lake Client Platform/JSL MRD, BIOS T8PRO001 10/25/2022
[    0.000000] tsc: Detected 2000.000 MHz processor
[    0.000000] tsc: Detected 1996.800 MHz TSC
[    0.000582] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000584] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000591] last_pfn = 0x280400 max_arch_pfn = 0x400000000
[    0.000642] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.000832] last_pfn = 0x79000 max_arch_pfn = 0x400000000
[    0.017363] esrt: Reserving ESRT space from 0x00000000751fd898 to 0x00000000751fd8d0.
[    0.017368] e820: update [mem 0x751fd000-0x751fdfff] usable ==> reserved
[    0.017681] Secure boot disabled
[    0.017683] ACPI: Early table checksum verification disabled
[    0.017687] ACPI: RSDP 0x00000000789A7014 000024 (v02 ALASKA)
[    0.017690] ACPI: XSDT 0x00000000789A6728 0000D4 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.017696] ACPI: FACP 0x00000000788F9000 000114 (v06 ALASKA A M I    01072009 AMI  01000013)
[    0.017700] ACPI: DSDT 0x00000000788B5000 043B3E (v02 ALASKA A M I    01072009 INTL 20191018)
[    0.017703] ACPI: FACS 0x00000000789FB000 000040
[    0.017705] ACPI: MCFG 0x00000000788FA000 00003C (v01 ALASKA A M I    01072009 MSFT 00000097)
[    0.017707] ACPI: FIDT 0x00000000788B4000 00009C (v01 ALASKA A M I    01072009 AMI  00010013)
[    0.017709] ACPI: SSDT 0x00000000788B1000 0025EE (v02 CpuRef CpuSsdt  00003000 INTL 20191018)
[    0.017711] ACPI: SSDT 0x00000000788FB000 001CCD (v02 SaSsdt SaSsdt   00003000 INTL 20191018)
[    0.017713] ACPI: SSDT 0x00000000788AD000 0032CD (v02 INTEL  IgfxSsdt 00003000 INTL 20191018)
[    0.017715] ACPI: HPET 0x00000000788AC000 000038 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.017717] ACPI: APIC 0x00000000788AB000 00012C (v04 ALASKA A M I    01072009 AMI  01000013)
[    0.017719] ACPI: PRAM 0x00000000788AA000 000030 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.017721] ACPI: SSDT 0x00000000788A9000 000E06 (v02 ALASKA Ther_Rvp 00001000 INTL 20191018)
[    0.017723] ACPI: SSDT 0x00000000788A6000 002FBC (v02 INTEL  xh_jslv2 00000000 INTL 20191018)
[    0.017725] ACPI: NHLT 0x00000000788A5000 0001FB (v00 ALASKA A M I    01072009 AMI  01000013)
[    0.017727] ACPI: LPIT 0x00000000788A4000 000094 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.017729] ACPI: SSDT 0x00000000788A0000 002720 (v02 ALASKA PtidDevc 00001000 INTL 20191018)
[    0.017731] ACPI: SSDT 0x000000007889F000 000122 (v02 ALASKA TbtTypeC 00000000 INTL 20191018)
[    0.017733] ACPI: DBGP 0x000000007889E000 000034 (v01 ALASKA A M I    01072009 AMI  01000013)
[    0.017735] ACPI: DBG2 0x000000007889D000 000054 (v00 ALASKA A M I    01072009 AMI  01000013)
[    0.017737] ACPI: DMAR 0x000000007889C000 000088 (v02 INTEL  EDK2     00000002      01000013)
[    0.017739] ACPI: SSDT 0x000000007889B000 000144 (v02 Intel  ADebTabl 00001000 INTL 20191018)
[... cut ...]
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821CE 802.11ac PCIe Wireless Network Adapter
```

```
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 008: ID 0d8c:0014
Bus 001 Device 007: ID 10c4:ea60
Bus 001 Device 006: ID 10c4:ea60
```

I think the Wi-Fi card is recognized, but I haven't tested it. What isn't working is the bluetooth part of the card. I haven't used that in the past either but I could have sworn when I first installed HAOS in October, it was working. Some recently update has made Bluetooth disappear.

closed 2024-04-09 by gaborbernat

Simonh98, 2024-04-10:
Can confirm that it works for me now. 👍