home-assistant/operating-system #2602: No host internet connection - Supervisor 2023.06.02

dgarciadonado, 2023-06-19:
### Describe the issue you are experiencing

I use HAOS as a VM inside Proxmox. I use two ethernet connections (one with internet output 192.168.1.8 and the other one 192.168.10.8 without internet output)
 
I have recently realized that I cannot update any addon nor restore a backup from any addon. The error message that shows is:
'AddonManager.install' blocked from execution, no host internet connection'. Is suspect it is since HA Supervisor update to 2023.05.xx

Investigating with [ha interface info] I have the flag host_internet = false but everything is working as expected.

Please find below the full output. Just to mention that the DNS server in enp0s18 interface is a pihole with 192.168.1.20 I have tested disabling blocking in pihole without success. I have also tested using 1.1.1.1 or 8.8.8.8 as DNS but same error.

```
docker:
  address: 172.30.32.0/23
  dns: 172.30.32.3
  gateway: 172.30.32.1
  interface: hassio
host_internet: false
interfaces:
- connected: true
  enabled: true
  interface: enp0s18
  ipv4:
    address:
    - 192.168.1.8/24
    gateway: 192.168.1.1
    method: static
    nameservers:
    - 192.168.1.20
    ready: true
  ipv6:
    address:
    - fe80::7cba:8892:df61:95fc/64
    gateway: null
    method: disabled
    nameservers: []
    ready: true
  primary: true
  type: ethernet
  vlan: null
  wifi: null
- connected: true
  enabled: true
  interface: enp1s27u3
  ipv4:
    address:
    - 192.168.10.8/24
    gateway: 192.168.10.1
    method: static
    nameservers:
    - 192.168.10.1
    ready: true
  ipv6:
    address:
    - fe80::11b8:768f:e61f:6ca/64
    gateway: null
    method: disabled
    nameservers: []
    ready: true
  primary: false
  type: ethernet
  vlan: null
  wifi: null
supervisor_internet: true
```

It seems that DNS is working, because I can ping google and also if I execute:
curl https://version.home-assistant.io/online.txt
the result is:
NetworkManager is online
[... 199 lines cut ...]
	"vethf331410"
	ethernet (veth), EE:1D:D1:9A:BB:FF, sw, mtu 1500

vethfd9d03e: unmanaged
	"vethfd9d03e"
	ethernet (veth), 5E:34:2F:82:BE:0D, sw, mtu 1500

vethff5c6d8: unmanaged
	"vethff5c6d8"
	ethernet (veth), 2A:AE:5F:92:84:7F, sw, mtu 1500

vethffcc42d: unmanaged
	"vethffcc42d"
	ethernet (veth), 46:67:15:78:FA:12, sw, mtu 1500

lo: unmanaged
	"lo"
	loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536

DNS configuration:
	servers: 192.168.1.20
	interface: enp0s18

	servers: 192.168.10.1
	interface: enp1s27u3

dgarciadonado, 2023-06-19:
Thank you for your fast answer!
Just checked and it seems that I can access:

[image]

My DNS server (192.168.1.20) is provided by a pi-hole in a raspberry pi. I have disabled Blocking on pihole but nothing change. Maybe is another option I need to change in pi-hole but for me it is weird because it has been working with this configuration various months/years.

Thank you again!

Taaitaaiger, 2023-06-22:
Have you tried manually setting the date? I've been struggling with the same issue and I think I just fixed it that way (because according to the logs a docker image is currently being downloaded).

`date --set="YYYY-MM-DD HH:MM"`

dgarciadonado, 2023-06-23:
Thank you for the tip! I applied on the VM inside proxmox but unfortunately didn't work. Any other option to see?
Any other log that should be useful and I can provide?
Thanks!

donkeykangwang10, 2023-06-24:
I have the same issue, my device is X86-64. my host_internet and supervisor_internet are all false. I have no idea.

home110120, 2023-06-25:
Same here. Both host and supervisor internet are false, on a VM of Synology NAS, and I did not apply any customized setting on HA, it use DHCP to get a constant IP as defined on my router, and nothing changed recently on network environment. The only abnormal is that supervisor failed to initiate in the past several month, sometimes everything is just fine with a reboot.

harfordhawk, 2023-06-25:
Not sure , but for me, OS10 still does not work.  I use proxmox 7 vm inside.  Multitude of connectivity problems , high processor usage, etc.. I reverted back to OS 9.5

dgarciadonado, 2023-07-01:
Please, could you point me how to revert to OS 9.5?
Or you just restore and old backup with that OS?
Thanks!

harfordhawk, 2023-07-01:
pretty sure the command is ha os update --version 9.5

in the console

dgarciadonado, 2023-07-05:
Thanks! I have reverted with the command you have proposed and right now I have connectivity but I cannot update addons nor supervisor.
[image]

Investigating, maybe the issue it could be related with the fact that I cannot fully disable IPV6.
It shows disabled in both interfaces:
[image]
But I have an IPV6 ip;
[image]

If I check both interfaces with nmcli IPV6 is in mode link-local:
[image]

and if I try to disable with nmcli command this is the result (which to be honest I don't understand why the response is related to ipv4-dns parameter instead of ipv6-method):
[image]

Any kindly hint?
Thanks!!

closed 2023-10-12 by github-actions[bot]