home-assistant/operating-system #2605: 10.3 Breaks Z-Wave JS + ZHA

ezfman, 2023-06-20:
### Describe the issue you are experiencing

Z-Wave JS fails to "initialize the driver" as an ACK from the USB stick is timing out.  ZHA also times out, so this may be some issue with USB devices on the Pi 4 v10.3.

Happy to share more detailed logs if necessary, but they're all timeout exceptions as far as I can tell.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Update OS to 10.3 from 10.2
2. Observe logs/status of USB-based hardware
...

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

stewartsmith, 2023-06-20:
Having the same issue on `x86-64` on an Intel NUC.

agners, 2023-06-20:
There isn't really a change for Raspberry Pi which would explain this (we still use the same kernel on Raspberry Pi boards as in 10.2). Do you use `/dev/serial/by-id/...` paths to reference serial ports?

Can you downgrade again to see if this indeed resolves the problem?

```
ha os update --version 10.2
```

ezfman, 2023-06-20:
For Z-Wave I use `/dev/tty/USB0`; downgrading to 10.2 does not seem to fix the issue.  I can fiddle around some tonight and see what more I can find.

stewartsmith, 2023-06-21:
So, I tried the 10.2 downgrade and that didn't fix it for me. However, unplugging the USB stick and plugging it in again (to a different port, not sure if significant, but probably isn't), it all started working. So that's.... fun and weird.

gokou340, 2023-06-21:
Same thing here. Had to restore from a previous day backup and the issue resolved itself.

brunokc, 2023-06-21:
Seeing similar issues after upgrading to 10.3. I noticed that I now have ttyS0, ttyAMA0, ttyAMA4 and ttyAMA5. Serial ttyAMA1, which used to be used by Zigbee/Thread disappeared.

ezfman, 2023-06-21:
I can confirm that removing and reinserting USB devices has fixed the issue.  Apologies for the pre-mature issue; it seems like this is an [old issue](https://www.reddit.com/r/homeassistant/comments/pfit07/zwave_js_timeout_while_waiting_for_an_ack_from/), and I should have done more research before opening it.

I'm going to close the issue, but it is interesting to note that several people have this issue on OS update.  Maybe a good reference for the Z-Wave JS folks?

closed 2023-06-21 by ezfman

bubba925, 2023-06-24:
I thought I had the same issue, but reinserting the device did not fix, it seems by HA Blue has finally been hit by the USB failure.

orrc, 2023-06-25:
I had a similar issue. After upgrading to 10.3, all of my Z-Wave devices were unavailable.

Looking at the Z-Wave JS addon configuration, my Aeotec Gen5 stick had disappeared (which used a `/dev/serial/by-id/…` path), and I was left with `/dev/ttyAMA0` as the only available device. That didn't work, and downgrading to 10.2 didn't help.

Due to [compatibility reasons](https://www.home-assistant.io/docs/z-wave/controllers/#aeotec-z-stick), the stick is connected to the RPi 4 via a USB 2 hub. Plugging the hub into different USB ports on the RPi didn't help. The device only re-appeared (as `/dev/serial/by-id/usb-0658_0200-if00`) when I moved the stick to a different port on the hub. After restarting the Z-Wave JS addon, Z-Wave worked again.

After upgrading again to 10.3 via the web UI, the device was still there and Z-Wave continued to work.

bubba925, 2023-06-25:
Interesting. I guess I didn’t fully troubleshoot my usb ports and assumed they all failed, I’ll verify this week but still seems likely since I booted up a VM and restored the same backup that was not working on my HA blue and it’s fine.

orrc, 2023-08-23:
For info: I just updated from 10.3 to 10.5 and the exact same thing happened again — my Z-Wave stick's device disappeared from the Z-Wave JS add-on settings, and stayed gone after a reboot.

The same fix worked: move the Z-Wave stick to another port on my USB hub and reboot. Then `/dev/serial/by-id/usb-0658_0200-if00` re-appeared and Z-Wave worked again.