home-assistant/operating-system #2606: Bluetooth hci0 unexpected event for opcode0x2005

MStapelfeldt, 2023-06-21:
### Describe the issue you are experiencing

After upgrade I got the error in title. I tried to reinstall HA on fresh system (SSD format, new download from virtual Linux), but get exact the same issue. Hardware is a 
Minis Forum EliteMini GK41
Prozessor	Intel® Celeron® Processor J4125
GPU Intel® UHD Graphics 600
Ram DDR4 8GB (on Bord)
HDD M.2 2280 128GB SATA SSD
Network connection IEEE 802.11ac Dual-Band Wi-Fi
BlueTooth
Ethernet	1000Mbps LAN

I think it's related to the fix from #2485 or #2535 from last release

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Format pc
2. Plugin Linux USB operation system 
3. Install with link from manual over Balena Etcher
4. Reboot system and watch failing
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Logs not available during install
```

### Anything in the Host logs that might be useful for us?

```txt
Logs not available during install
```

### Additional information

In bios I can't disable Bluetooth

agners, 2023-06-22:
> After upgrade I got the error in title. I tried to reinstall HA on fresh system (SSD format, new download from virtual Linux), but get exact the same issue.

What is happening after the error shows up?

#2485 was a different Opcode, and #2535 didn't show messages in dmesg. It could also be kernel update or something else entirely.

MStapelfeldt, 2023-06-22:
> What is happening after the error shows up?
Installation gets stucked with message
"Something went wrong loading onboarding, try refreshing"
I tried today with 10.2 and 10.1 (fresh download from [release page)](https://github.com/home-assistant/operating-system/releases/tag/10.2) and same issue. What I did:

1. Deleted all partition on SSD, create one 128gb partition
2. Download release file from rep
3. Used Balena Etcher to copy to SSD
4. Reboot system

After seeing CI prompt the first time, it takes about 30 seconds till I see `Bluetooth hci0 unexpected event for opcode0x2005`
I didn't do anything else, then starting installation.

agners, 2023-06-22:
The Bluetooth error is unrelated to the issue you are having with onboarding.

So you can reach the onboarding page at `http://homeassistant.local:8123`?

> I tried today with 10.2 and 10.1 (fresh download from [release page)](https://github.com/home-assistant/operating-system/releases/tag/10.2) and same issue. What I did:

There is already 10.3 available.

MStapelfeldt, 2023-06-22:
> So you can reach the onboarding page at http://homeassistant.local:8123?

I can reach :8123 (with spinning button and error message) and :4357 with Connected / Supported / Healthy (all green).

> The Bluetooth error is unrelated to the issue you are having with onboarding.

I had the issue first time while upgrading 10.2 to 10.3 (HA not reachable, console showed BL error). Then I tried fresh reinstall of 10.3, 10.2 and 10.1 all 3 installations failed. only thing I can see is 
```
Waiting for Home Assistant CLI to be ready...
LOGO
Waiting for Supervisor to starting...
System information
.....
OS Version: 10.1
Home Assistent Core 2023.6.2
ha > 
[25.226729] Bluetooth: hci0 unexpected event for opcode0x2005
[35.916884] Bluetooth: hci0 unexpected event for opcode0x2005
....
```

agners, 2023-06-22:
Can you type `supervisor logs` and `core logs` in the shell and see if it has information about what went wrong while loading onboarding?

MStapelfeldt, 2023-06-23:
After trying to access Home Assistant by IP instead of homeassistant:8123 I was able to access and use it again. So the issue is not related to Bluetooth and after deactivating BT integration the errors are gone in log.
I've tried to restore my HA, which worked fine, but after a reboot, the site crashed again and is not reachable. IU have ssh access and can use console, :4357 and supervisor log shows everything green, core log has some issues (orange, template variable warning).
Now trying to find the new issue (or same issue after upgrade 10.2 to 10.3) and fixing this.

MStapelfeldt, 2023-06-23:
I finally (after 7 times installing 10.3) found out what causes the system to crash:
after installing duckdns I added
```
http:
  ssl_certificate: /ssl/fullchain.pem
  ssl_key: /ssl/privkey.pem
  ip_ban_enabled: true
  login_attempts_threshold: 5
```
as descriped in every tutorial. after adding this, the system will not boot and warning
`[homeassistant.components.ssdp] could not set up UPnP/SSDP server, as a presentation 
URL could not be determinated`
Really strange this causes a failure in newest version, but not before (10.2) but it's the only possibility I can think about by try and error...

Feel free to close issue, because it's not really related to core or HA, but duckdns.

closed 2023-09-30 by github-actions[bot]