home-assistant/operating-system #2611: HAOS Support for Bluetooth using integrated AX201 on Intel NUC

tommyjlong, 2023-06-24:
### Describe the issue you are experiencing

_This is a split off from [This Issue ](https://github.com/home-assistant/core/issues/91155#issuecomment-1595917441)_
NUC: NUC11TNHi5 (i5Gen11) with AX201=>WiFi6/BT5 (Not to be confused with an AX210).

QEMU/KVM x86_64 running HA 2023.6.3, Supervisor 2023.06.2, Operating System 10.3  (uname -r => 6.1.34)

[Ubuntu 22.04 Host]
$lsusb =>Bus 003 Device 004: ID 8087:0026 Intel Corp. AX201 Bluetooth

[HAOS as Guest]:
#lsusb => Bus 001 Device 006: ID 8087:0026
#dmesg:
....
[    0.421912] Bluetooth: Core ver 2.22
[    0.422216] NET: Registered PF_BLUETOOTH protocol family
[    0.422502] Bluetooth: HCI device and connection manager initialized
[    0.422848] Bluetooth: HCI socket layer initialized
[    0.423213] Bluetooth: L2CAP socket layer initialized
[    0.423485] Bluetooth: SCO socket layer initialized
[    2.840236] usb 1-6: New USB device found, idVendor=8087, idProduct=0026, bcdDevice= 0.02
[    2.841196] usb 1-6: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    2.951226] usbcore: registered new interface driver btusb
[    5.015097] Bluetooth: hci0: command 0xfc05 tx timeout
[    5.015097] Bluetooth: hci0: Reading Intel version command failed (-110)

BTW, this is my first attempt at passing the NUC's integrated bluetooth to HA Guest, so can't tell if it worked on other HA kernel versions or not.

This does work on the NUC's Ubuntu Host (when not passing to the VM). 
Ubuntu uname -r =>5.19.0-43-generic

FYI, there is also a dependency difference on the `btusb` module between Host and HAOS as Guest:
[Ubuntu 22.04 Host ] 
```
$modinfo btusb: 
....
depends:        bluetooth,btintel,btmtk,btbcm,btrtl
```
[HAOS as Guest ]
```
# modinfo btusb
...
depends:        btintel,btrtl,btbcm
```

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. modprobe -r btusb
2. modprobe btusb
3. look at dmesg log
[362215.081098] usbcore: deregistering interface driver btusb
[362218.357222] usbcore: registered new interface driver btusb
[362220.416390] Bluetooth: hci0: command 0xfc05 tx timeout
[362220.416397] Bluetooth: hci0: Reading Intel version command failed (-110)

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

axute, 2023-06-30:
Same adapter, does not work.

```
[    1.640322] Bluetooth: hci0: Device revision is 2
[    1.640325] Bluetooth: hci0: Secure boot is enabled
[    1.640326] Bluetooth: hci0: OTP lock is enabled
[    1.640327] Bluetooth: hci0: API lock is enabled
[    1.640328] Bluetooth: hci0: Debug lock is disabled
[    1.640329] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    1.640330] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[    1.640596] Bluetooth: hci0: Failed to load Intel firmware file intel/ibt-0040-1050.sfi (-2)
```

tommyjlong, 2023-07-01:
@axute can you describe your HA installation? I'm wondering if you have full blown HA/HAOS installed on baremetal NUC.  You setup seems to have progressed further than mine.

axute, 2023-07-02:
Hello, that's right, I have a "Trigkey Green G5". This is an Intel N100 with 16GB DDR5 and 512 NVMe, 2 NICs and there are a lot of bios settings regarding wifi and bluetooth. WiFi is not recognized either...

About the installation:
Home Assistant 2023.6.3
Maintainer 2023.06.4
Operating system 10.3
Frontend 20230608.0 - current

What exactly do you need in the way of details?

tommyjlong, 2023-07-02:
@axute Here is a [link ](https://forums.gentoo.org/viewtopic-t-1163072.html?sid=90dba32692ee2125aa582fe839c53050)that may be the problem you are seeing.  In the link, the users have an AX101 device that is being detected as an AX201.  Googling around a little bit I think maybe you actually have an AX101 chip.  Apparently the upstream linux kernel is just now (ver 6.4) supporting the AX101 properly (HAOS 10.3 uses version 6.1.34), but still hasn't caught up to have the missing file your log says it needs.  I think the kernel will eventually have a version that will work for you, just may have to wait a while.

Since your system is running bare metal HAOS and your bluetooth initialization setup makes more progress than mine, I have a feeling that my QEMU/KVM system simply can not deal with the AX201 properly as bluetooth USB device.

axute, 2023-07-02:
I have ths same device id, why should it be another chip?

tommyjlong, 2023-07-04:
Very good question.  I am mostly going by the link I referenced above where the originating poster says he has an AX101 but he says the BT driver is claiming its AX201. I could not actually find any spec on your Trigkey Green that says what it has.

I can use my imagination to come up with a way that this could happen but I may be entirely incorrect about the following:  As I understand it, any AXx01 device is not a full blown WiFI/BT-USB combo device.  These x01 devices are just RF radio modulator/demodulator that converts the analog RF to digital baseband and sends it over some Intel proprietary interface, and some other Intel internal processor core does the actual WiFi and Bluetooth baseband and MAC layer processing.  This processor core requires firmware usually with filename iwlwifi-blah-blah.  This firmware is essentially performing as a software WiFI/BT chip and would be presenting the "system" side as a PCIe device, and USB Device according to the firmware.  So I could see where the firmware for an AX101 presents the USB device with whatever device id it wanted to (since its soft) and it could present it with the same device id used with the AX201.

axute, 2023-07-05:
But it doesn't play a role for me at the moment, I agree with you, there are so many settings in the bios that your idea would fit. But that still brings me to the conclusion that there is a kernel module that works.
https://linux-hardware.org/?id=usb:8087-0026&page=1#status
Just not for home assistants.

0x8000-0000, 2023-08-03:
https://askubuntu.com/questions/1459856/intel-alder-lake-n100-wifi-and-bluetooth-issues# suggests using a different firmware file and it report success.

I have copied the file to /lib/firmware/intel/ibt-0040-1050.sfi but that directory seems to be wiped out on reboot.

zephirus, 2023-09-16:
I have the same issue. So, at the moment there is no solution?

paulvylie, 2023-09-20:
same issue for me
ax101
supported in newer linux kernel

zephirus, 2023-09-20:
If I understand correctly, this is solved in a newer linux kernel that HAOS is not using? Am I correct?

paulvylie, 2023-09-20:
> If I understand correctly, this is solved in a newer linux kernel that HAOS is not using? Am I correct?
yes, correct

zephirus, 2023-09-20:
Is there any clue of when HAOS will update its kernel? At the moment I'm thinking about buying a usb bluetooth adapter.

paulvylie, 2023-09-20:
no idea
wasn't able to find any roadmap

paulvylie, 2023-09-24:
update
debian with 6.5 kernel and ha supervised still not working

thewan056, 2023-10-04:
suggesting an alternative, just change to a supported wifi card. it is not that much more expensive than a decent usb bluetooth dongle.

lantech69420, 2024-01-21:
This isn't resolved yt

doctorgaul, 2024-02-14:
No resolution for me either, having the same problem still.  Beelink SER5 PRO 5800H with a Intel Corp. AX200 Bluetooth module, and Home Assistant won't recognize it.

sairon, 2024-02-14:
@doctorgaul @lantech69420 There are multiple different issues mixed up in this thread. Can you at least share `dmesg` output from your system? On current stable release (HAOS 11.5) Bluetooth for all cards mentioned in this thread (AX101, AX200 and AX201) should work, the upcoming 12.0 release will include also drivers for AX101 WiFi, which had support only for Bluetooth in 6.1 kernel (BT firmware for AX101 was added in 11.5, so some comments are not relevant anymore).

sairon, 2024-02-14:
@tommyjlong I'm not sure if your problem has been resolved or not, but judging from these lines:
```
[    5.015097] Bluetooth: hci0: command 0xfc05 tx timeout
[    5.015097] Bluetooth: hci0: Reading Intel version command failed (-110)
```
there are likely two options. Either there is some hardware fault, or the host system (or another VM) tries to access the USB interface of the Bluetooth card at the same time. To make sure it is not the case, you must not use it for other VM and disable the Bluetooth stack (e.g. by blacklisting bluetooth modules, or using udev rules) on the host OS.

doctorgaul, 2024-02-14:
@sairon I apologize in advance for not being the most technical user, but I believe these sections of the `dmesg` output are what you need?  Let me know if there's anything other information I can provide.  Really appreciate your help!!

[image]

[image]

lantech69420, 2024-02-14:
In the latest HAOS, my bluetooth started working on my Beelink Mini S12 Pro.

sairon, 2024-02-14:
@lantech69420 I see, then in your case I guess it was either a different card (AX101 likely?) or the firmware upgrade helped.

@doctorgaul No worries, you got it right :+1: From the screenshots I can see there is indeed something wrong with the initialization of the card, more details could be in the part that is not on the screenshots (starting with the `cut here` line, which should end with `end trace` delimiter). This points either to a problem in the Linux kernel driver, or there is some hardware problem in the card itself. Probably the easiest way to check that would be to try running a different OS (like a bootable stick with Ubuntu for example) and check kernel logs there. Another option is to try the current dev build of HAOS, or wait for 12.0 release, which updates Linux kernel to 6.6.

doctorgaul, 2024-02-14:
Thanks @sairon !  Here's the next portion of the `dmesg` output.  Let me know if there's anything else I can do here!

[image]

tommyjlong, 2024-02-14:
> @tommyjlong I'm not sure if your problem has been resolved or not, but judging from these lines:
>
> ```
> [    5.015097] Bluetooth: hci0: command 0xfc05 tx timeout
> [    5.015097] Bluetooth: hci0: [...]

Yeah I'm still seeing the same problem.  I'll play with it some more.  My suspicion is that the AX201 can't really be split into WiFi and USB cleanly, thus doing a USB passthrough to the VM while leaving WiFi on the host doesn't really work.

What kind of udev rules and actions do you have in mind?

Pilchie, 2024-03-13:
I have the same issue as @tommyjlong.  I'm also running a Beelink SER5 MAX running ubuntu 22, with HAOS running in a guest VM through qemu.  I get the same messages:

```
Bluetooth: hci0: command 0xfc05 tx timeout
Bluetooth: hci0: Reading Intel version command failed (-110)
```

This did work a few months ago, but stopped at some point, and than randomly did on one reboot, but has been failing consistently for a bit now.

tommyjlong, 2024-04-28:
I now have the AX201 working w. HAOS (as a pass-through to a HA VM using Libvirt/QEMU/KVM), so I'm going to close this.  
If anyone else is still having problems, then feel free to open up a new issue.  

`HA Operating System 12.2`

--------------------------
I'm not sure what HAOS release actually starting supporting the AX201, but afterwards, it turned out that my issue was more about passing through the AX201 to the HA VM, here is a bit of a write-up on what I did. _I'm doing this somewhat from memory, so I may have missed a thing or two or even added one or two things too many, but should be reasonably close._

This is for an Intel NUC running Ubuntu 22.04 as the Host. 

The AX201 on Intel NUC presents itself to Linux as 
- PCI Network Controller for WiFi device 
- PCI USB Controller and with it as a USB device

I passed through the AX201 PCI controller for WiFi, but I only passed-through the AX201 USB device instead of the whole PCI USB controller.

I. Pass through the AX201 PCI WiFi device
 1a) Assign vfio-pci driver to AX201 PCI device (using Grub) 
- Find the PCI BDF address of the AX201
  On the Host: `$ lspci`
  ....
  00:14.3 Network controller: Intel Corporation Wi-Fi 6 AX201 (rev 20)
  ...
- Get vendor-id:device-id info on the AX201  WiFI PCI device  
    $ lspci -vnn -s 0000:14.3
    00:14.3 Network controller [0280]: Intel Corporation Wi-Fi 6 AX201 [8086:a0f0] (rev 20)
            DeviceName: Onboard - Ethernet
            Subsystem: Intel Corporation Wi-Fi 6 AX201 [8086:0074]
- Configure Grub to assign vfio-pci driver to this PCI device
   Add a line in /etc/default/grub
   `GRUB_CMDLINE_LINUX="iommu=pt intel_iommu=on vfio-pci.ids=8086:a0f0"`
  and update grub using: `$sudo update-grub`
- Reboot the host to pick up the changes.

 1b) Alternative way to assign vfio-pci driver to AX201 PCI device (using modprob)
   _As this should be an alternative to 1a), I actually did both 1a) and 1b).  At least the iommu config in 1a) should still be done._
- Get PCI device alias
  `$ cat /sys/bus/pci/devices/0000\:00\:14.3/modalias`
  `pci:v00008086d0000A0F0sv00008086sd00000074bc02sc80i00 `
- create a new file in /etc/modprobe.d/vfio.conf and add these 2 lines:
  `alias pci:v00008086d0000A0F0sv00008086sd00000074bc02sc80i00 vfio-pci`
  `options vfio-pci ids=8086:a0f0`

  Note: both these lines are probably not needed, but I used both anyway.
  Note: you could blacklist items in this file, but I didn't.
  Although probably not necessary, you can update initramfs to be safe:
  `$ sudo update-initramfs -u -k all`
- Reboot the host to pick up the changes.

 2) Add the AX201 PCI WiFi device to the VM.
   You can use your favorite method to configure your QEMU/KVM HAOS VM to pick up the AX201 PCI WiFi device.
   I use Virt-Manager: Select the HA VM then "Add Hardware" ->"PCI Host Device" then Select the AX201

II. Pass through the AX201 USB Bluetooth device
  1) Black list the host's bluetooth kernel modules
   _What I found out was that when libvirt adds the Bluetooth USB device to the VM, Libvirt first attempts to remove the host's control of the Bluetooth controller using `rfkill`, but this didn't work well enough, so blacklisting is about the only way I could get the host to completely relinquish control._
  `$lsmod | grep bluetooth`
   `bluetooth            1081344  15 btrtl,btmtk,btintel,btbcm,bnep,btusb,rfcomm`

  - create a new file `/etc/modprobe.d/blacklist-bluetooth.conf`
   and add a line for each module listed in the previous step:
   `blacklist bluetooth`
   `blacklist btrtl`
   `blacklist btmtk`
   etc. etc,
   Although probably not necessary, you can update initramfs to be safe:
   `$ sudo update-initramfs -u -k all`
- Reboot the Host to pick up the changes.

 2) Add the AX201 Bluetooth device to the VM
[... 19 lines cut ...]
[    4.034941] Bluetooth: hci0: Secure boot is enabled
[    4.034942] Bluetooth: hci0: OTP lock is enabled
[    4.034942] Bluetooth: hci0: API lock is enabled
[    4.034943] Bluetooth: hci0: Debug lock is disabled
[    4.034943] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    4.038862] Bluetooth: hci0: Found device firmware: intel/ibt-19-0-4.sfi
[    4.038882] Bluetooth: hci0: Boot Address: 0x24800
[    4.038883] Bluetooth: hci0: Firmware Version: 249-27.23
[    4.130892] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    4.130895] Bluetooth: BNEP filters: protocol multicast
[    4.130898] Bluetooth: BNEP socket layer initialized
[    6.077227] Bluetooth: hci0: Waiting for firmware download to complete
[    6.077983] Bluetooth: hci0: Firmware loaded in 1991326 usecs
[    6.078045] Bluetooth: hci0: Waiting for device to boot
[    6.093341] Bluetooth: hci0: Device booted in 14981 usecs
[    6.096907] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-19-0-4.ddc
[    6.099093] Bluetooth: hci0: Applying Intel DDC parameters completed
[    6.100205] Bluetooth: hci0: Firmware revision 0.4 build 249 week 27 2023
[    6.102122] Bluetooth: hci0: HCI LE Coded PHY feature bit is set, but its usage is not supported.
[    6.171430] Bluetooth: MGMT ver 1.22
[   13.366386] Bluetooth: RFCOMM TTY layer initialized
[   13.366392] Bluetooth: RFCOMM socket layer initialized
[   13.366395] Bluetooth: RFCOMM ver 1.11

```

closed 2024-04-28 by tommyjlong

rizlas, 2025-02-02:
Unfortunately for me, the bluetooth card, after your steps, was recognized by the VM but it's still not working due to firmware issues. 
The host machine is a debian 12.

[image]

tovy14, 2025-04-02:
> I now have the AX201 working w. HAOS (as a pass-through to a HA VM using Libvirt/QEMU/KVM), so I'm going to close this. If anyone else is still having problems, then feel free to open up a new [...]

Just came to confirm that the intel AX211 works after following these instructions

mercertom, 2025-11-20:
@tovy14 You're a fucking treasure. Hope other people come to this, and I'll add more SEO terms to help them find this:

Proxmox bluetooth USB passthrough failing beelink 

No unconfigured bluetooth adapters found.

Read @tovy14 's guide. You need to blacklist all the bluetooth 'services' because the host isn't releasing the Intel wifi to the Home Assistant VM. I have a Beelink, but this is for anyone running home assistant on proxmox.