home-assistant/operating-system #2612: Bluetooth still unreliable with Yale BLE after 10.3 upgrade

contactcr, 2023-06-24:
### Describe the issue you are experiencing

On 9.5 my BLE Yale locks are speedy and seem to be reliable for long periods of time.  10.x had its own problems and I didnt stick with it long but 10.3 was supposed to fix bluetooth on intel NUC.  Well, the locks are mostly staying connected by seem to still drop out or take 10-20 seconds to do things randomly.  They will sometimes become unavailable still also.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

core logs have the errors [core logs.txt](https://github.com/home-assistant/operating-system/files/11858259/core.logs.txt)

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing of relevance in supervisor
```

### Anything in the Host logs that might be useful for us?

```txt
dont see much in host logs [host_2023-06-24T20-44-32.104Z.log](https://github.com/home-assistant/operating-system/files/11858260/host_2023-06-24T20-44-32.104Z.log)
```

### System information

core-2023.6.2 on NUC12 i5

j-pas, 2023-06-27:
I also have a NUC (I have Nuc 10), have 2 different bluetooth devices (air sensor and plant sensor) both with simultaneous connection issues at random moments and random duration.. See below screenshot for example.

I changed back and forth the HAOS version with below results:
HAOS 9.5 -> 10.1 (issues began)
HOAS 10.1 -> 9.5 (issues gone, all works fine)
HAOS 9.5 -> 10.3 (issues are back)

I run Proxmox 7.4-3 on my Intel Nuc. Proxmox is shipped on top of Debian (version 11).
Within Proxmox I run Homeassistant in a VM based on the HAOS iso.
uname -a on the host machine (debian):
Linux nuc 5.15.107-2-pve #1 SMP PVE 5.15.107-2 (2023-05-10T09:10Z) x86_64 GNU/Linux

Happy to provide any input if needed!
[image]

jan-kuechler, 2023-07-12:
I have similar problems, using a quite old RaspberryPI 2 and a FSC-BP119 bluetooth adapter. I'm connecting to bluetooth thermostats, and starting with 10.1 I experienced connection issues. After a while, the bluetooth adapter becomes fully unresponsive (connection doesn't back, even after days) and the only thing helping is either rebooting the Pi, or re-plugging the bluetooth stick. Upgrading to 10.2 and 10.3 didn't help, but since a week I've downgraded the system to 9.5 and do not have any bluetooth issues since.

megaz555, 2023-07-14:
Same issue for me. 10.3 was a lot better than 10.2 and 10.1, but it is still unreliable. I have Airthings BLE devices and they keep on getting disconnected. Forcing a reboot on the Intel NUC brings devices back online, then they go back to offline again in less than 24 hours.

Can't someone get this fixed? 3 major releases in, and we are still dealing with Bluetooth issues.

yannpub, 2023-07-14:
I am not on HaOS, but suffering the same symptoms since I upgraded to the latest kernel in the new Debian. I used to restart HA several times a day, but I'm now running the following automation every 5 minutes as a workaround, as it only restarts the Bluetooth integration :
https://github.com/home-assistant/core/issues/93036#issuecomment-1585774135

contactcr, 2023-08-03:
Anyone want to try 10.4?  I've reverted 3 times now so i'm going to let someone else do it.

DorjeDorf, 2023-08-03:
I've had all my USB devices fail on 10.4. rolled back to 10.3 brought them back, so it's a hell no for me ;)

Lafunamor, 2023-08-10:
For me on my Raspberry PI 3 bluetooth is still not reliable on 10.4. I have a bunch of Xiaomi temperature and humidity sensors and in passive scanning mode it will sometimes take many hours until it receives a BLE message from a sensor. With OS 9.5 it was rather reliably every 5-10 mins.
I'm wondering if I should upgrade to a PI4 now that they are available again, as it seems that the devs are testing with those.

gradius-v, 2023-08-10:
> I'm wondering if I should upgrade to a PI4 now that they are available again
On my Pi4 Bluetooth is unusable since 10.x
After a full reboot all seems fine. But after a few hours it becomes unresponsive. Sometimes it will successfully reset after a few minutes and work some more hours before it's completely dead.

megaz555, 2023-08-13:
Same issues as everyone else is having. Intel NUC, Bluetooth constantly fails since 10.0 release. Tried 10.4, still random disconnects. Going back to 9.5 fixes all the issues!

agners, 2023-08-16:
When the problem is happening on HAOS 10.4, can you run the following command to get the kernel logs and share the kernel log file generated kerne log file?

```
dmesg > /config/kernel.log
```

agners, 2023-08-17:
Also, can you share the exact Intel NUC models you are using?

ZTHawk, 2023-08-22:
Maybe related to this #2142 
I had to revert from 10.4 and 10.5 back to 10.3.
I have an ASUS PN41 (chip: Intel Wireless AC 9560)

Lafunamor, 2023-09-02:
@agners I'm running now on 10.5 and still see a signifikant drop in updates of my BLE devices. Here are my dmsessages: [kernel.log](https://github.com/home-assistant/operating-system/files/12503849/kernel.log)
I see a lot of `[ 3699.380055] Bluetooth: hci0: Frame reassembly failed (-84)` but those I see with 9.5 as well.
One thing that might be new since 10.x are those messages (unfortunately, I son't have old logs to compare):
```
[ 7112.169823] Bluetooth: hci0: unexpected event 0x03 length: 3 < 11
[ 7868.105749] Bluetooth: hci0: Received unexpected HCI Event 0x00
```
Please let me know if there is anything else I can provide to help narrowing this down.

Lafunamor, 2023-09-02:
To provide a bit more insights, here are two screenshots of the same sensor. First with OS version 9.5: 
[image]

And this one with OS version 10.5:
[image]

While also OS version 9.5 has intermittent failures, it's clearly visible that the version 10.5 data is less granular. Basically the graphs should have a similar pattern with a spike that goes up immediately followed by a low spike.

bjornarfjelldal, 2023-09-04:
Can confirm same bluetooth issues on Intel NUC 12th Gen with Proxmox 8 running latest Home Assistant (10.5 + 2022.8)

mcc05, 2023-09-28:
Similar issue here NUC running HASIO in VM with AC9560 Bluetooth chipset, works for a while, then starts throwing the errors below and eventually bluetooth becomes inoperable and VM needs restarting.
  
[image]

sairon, 2023-10-09:
I think there are multiple issues getting mixed here, but anyway - if the Bluetooth adapter keeps being disconnected, please make sure the host OS isn't interfering anyhow with the USB device. In every case please share `dmesg` and `lsusb` output from HAOS, and if the host system is Linux as well, please do share the same from the host too.

If you're using devices that broadcast data over BLE advertisement beacons (no active connection is used for communication), it's also worth checking if toggling "Passive scans" option in the [Bluetooth integration config](https://my.home-assistant.io/redirect/integration?domain=bluetooth) makes some difference. I explicitly say toggling, because we already have some anecdotal evidence that disabling this option improved the situation for passive BT device too.

ZTHawk, 2023-11-08:
anyone tested with OS 11 or 11.1?

gcarrarom, 2023-11-09:
I'd like to note that HAOS 11 made my dongle start failing as well. No onboard bluetooth on RPI4 or dongle working anymore.

contactcr, 2023-11-14:
A user reported [kernel 6.5.7 fixes this](https://github.com/home-assistant/operating-system/issues/2535#issuecomment-1784219788) but i've seen that claim before I guess there is a lot of variation in people's configuration.  How far behind in OS versions can I get before I start to see other problems I wonder.

gcarrarom, 2023-11-14:
I guess my question is: is there a dongle that actually solves this issue and I don't have to think about it? Has anyone come across such a thing? I've tried 2 from the recommended list, but they fail as much as the onboard card from the PI4

bjornarfjelldal, 2023-11-14:
> I guess my question is: is there a dongle that actually solves this issue and I don't have to think about it? Has anyone come across such a thing? I've tried 2 from the recommended list, but they fail as much as the onboard card from the PI4

My  solution was to disable all internal bluetooth as well as removing any dongles, and solely depend on ESPhome proxies

agners, 2023-11-15:
> A user reported https://github.com/home-assistant/operating-system/issues/2535#issuecomment-1784219788 but i've seen that claim before I guess there is a lot of variation in people's configuration.

We typically move to the next LTS Linux kernel in Q1.

> How far behind in OS versions can I get before I start to see other problems I wonder.

From a [ADR0015](https://github.com/home-assistant/architecture/blob/master/adr/0015-home-assistant-os.md) perspective we guarantee support for 3 months, but in practice we hadn't have the case where we break a old OS version. Some features of newer Supervisor/Core might not work as expected (such as network storage) but generally very old versions work still fine. Even if you would get into unsupported state, you should be able to upgrade still. In worst case, you can make a full backup and restore that on a new installation.

closed 2024-02-21 by github-actions[bot]