home-assistant/operating-system #2613: Home Assistant OS not show cpu temperature

Sergmasharov, 2023-06-25:
### Describe the issue you are experiencing

Hello! I have intel cpu fanless industrial pc with Intel Atom(TM) CPU D2000 family CPU. Finally I tied of updating linux and decide to move to home assistant OS. But I cant get cpu temp. Build-in integration not work.
I made manually sensor by command line:
  - sensor:
     scan_interval: 300
     name: cpu_temperature
     command: 'cat /sys/class/thermal/thermal_zone0/temp'
     unit_of_measurement: '°C'
     value_template: '{{ value | multiply(0.001) }}'

It always show 26.8 withot changes.
I tryed to edit grub with adding line: GRUB_CMDLINE_LINUX="acpi_enforce_resources=lax"
And no results.
I initiated ssh conection to port 22222 and tryed to edit pcoc/config.gz 
CONFIG_SENSORS_CORETEMP
But no luck - no permissions to edit file...

Please help me, CPU temp is most important value for system.
Before installation HA OS I run my system on container in Debian 11 and have no problems with systemmonitor and cpu temperature.

Thanks!

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Decide to move to HA OS
2. Install latest HA OS
3. Restore backup from my old Debian 11 installation
4. Have problems with reading cpu temperature...

### Anything in the Supervisor logs that might be useful for us?

```txt
no problems in log
```

### Anything in the Host logs that might be useful for us?

```txt
Logger: homeassistant.components.systemmonitor.sensor
Source: components/systemmonitor/sensor.py:358
Integration: systemmonitor (documentation, issues)
First occurred: 24 июня 2023 г. в 12:00:54 (1 occurrences)
Last logged: 24 июня 2023 г. в 12:00:54

Cannot read CPU / processor temperature information
```

### System information

## System Information

version | core-2023.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.34
arch | x86_64
timezone | Europe/Samara
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1277
Downloaded Repositories | 18

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.3
-- | --
update_channel | stable
supervisor_version | supervisor-2023.06.2
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 233.1 GB
disk_used | 8.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), RPC Shutdown (2.4), Nginx Proxy Manager (0.12.3), MariaDB (2.6.1), Home Assistant Google Drive Backup (0.111.1), Duck DNS (1.15.0), Samba share (10.0.2), Advanced SSH & Web Terminal (15.0.2), FTP (4.7.1), Mosquitto broker (6.2.1), Zigbee2MQTT (1.31.2-1), ESPHome (2023.6.2), HassOS SSH port 22222 Configurator (0.9.1a), Terminal & SSH (9.7.1)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 9
views | 9
mode | yaml

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 16 июня 2023 г. в 12:02
-- | --
current_recorder_run | 24 июня 2023 г. в 12:00
estimated_db_size | 243.17 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Sonoff</summary>

version | 3.5.1 (122751c)
-- | --
cloud_online | 6 / 6
local_online | 3 / 3

</details>

<details><summary>Xiaomi Gateway 3</summary>

version | 3.3.1 (81042bb)
-- | --

</details>

### Additional information

[core-ssh ~]$ cat /proc/cpuinfo
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 54
model name      : Genuine Intel(R) CPU        @ 1.86GHz
stepping        : 1
microcode       : 0x10d
cpu MHz         : 1861.949
cache size      : 512 KB
physical id     : 0
siblings        : 1
core id         : 0
cpu cores       : 1
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 10
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts nopl nonstop_tsc cpuid aperfmperf pni dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm movbe lahf_lm dtherm arat
bugs            :
bogomips        : 3723.77
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:

kdelios, 2023-06-25:
Try zone1

Sergmasharov, 2023-06-25:
> Try zone1

I have only one zone0 and two cooling devices with no temperature info...

Sergmasharov, 2023-08-03:
Fixed in 10.4

closed 2023-08-03 by Sergmasharov

Opontios, 2023-08-08:
> > Try zone1
> 
> I have only one zone0 and two cooling devices with no temperature info...

It was working with "zone2" for me before the fix. It was a known issue, and I think if you had put "zone2" it would work for you also. Now I haven't updated yet. Don't know if  it will need to put "zone0" after the update.