home-assistant/operating-system #2616: Addons page of HaOS does not load unless you visit it then force refresh the browser

bharvey88, 2023-06-27:
### Describe the issue you are experiencing

using:  Home Assistant 2023.6.3 Supervisor 2023.06.2 Operating System 10.3 Frontend 20230608.0 - latest 

Addons page of HaOS does not load unless you visit it then force refresh the browser

https://github.com/home-assistant/frontend/issues/16885 I am reposting this here because I believe it might be an OS issue not a frontend issue. I asked in the discord server but was unable to get a confirmation so i went ahead and created this issue.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. open haos
2. go to addons page
3. addons wont load you just see a blank page
4. press f5 or click refresh in browser, now addons show up
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I do not see anything in supervisor logs relating to failed loading of addons page or anything out of the ordinary.
```

### Anything in the Host logs that might be useful for us?

```txt
I do not see anything in the host logs relating to the page faling to load either.
```

super-Alfi, 2023-07-07:
Problem disappeared since some updates

bharvey88, 2023-07-08:
https://github.com/home-assistant/frontend/pull/17146 looks to have been fixed here - closing

closed 2023-07-08 by bharvey88