home-assistant/operating-system #2619: HA OS 10.3

th3uiguy, 2023-06-29:
### Describe the issue you are experiencing

ext4 filesystem read/write errors when using an SSD on Raspberry Pi 4

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Write HA OS 10.3 image to SSD though Raspberry Pi Imager
2. Start up the Raspberry Pi with SSD (USB) setup as the boot image location
3. Notice file system errors

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

agners, 2023-06-30:
This sounds like a HW issue. Have you tried a different SD card and/or Raspberry Pi?

Baxxy13, 2023-06-30:
For me it sounds like another problematic/incompatible USB->SATA Adapter because no SD-Card is involved.

There is already a Blacklist of Adapters which having problems with UAS-Mode. 
Maybe your Adapter should be blacklistet too.

Read here: #1743

th3uiguy, 2023-06-30:
@agners It is a Kingston SSD (only a few months old). You think the drive when out? I do not see these errors on 10.2 so I don't know why it would be a hardware issue. They happened after updating to the 10.3 OS and then any time I try to reinstall using 10.3 is see them fill the screen, sorry I do not have screenshots.

th3uiguy, 2023-06-30:
I am using the StarTech adapter: USB312SAT3CB

th3uiguy, 2023-06-30:
I'll work on getting more details

zillion42, 2023-07-04:
I think it's the same as this. I have not heard anyone booting from usb ssd on pi4 above HA OS 9.5 without errors:
https://github.com/home-assistant/operating-system/issues/2536

markusmauch, 2023-07-09:
I have the same issue using the ORICO Extreme PRO Portable SSD External Hard Drive 128 GB USB 3.1 Gen-2 Type-C https://amzn.eu/d/e4Yx9yw

It worked perfectly prior to Home Assistant OS 10. Now I'm either stuck with 9.5 or have a system crash at least every second day.

closed 2023-10-16 by github-actions[bot]