home-assistant/operating-system #2620: [wget: bad address 'version.home-assistant.io'] occurred while compiling rpi3  image

chengjing30, 2023-06-30:
### Describe the issue you are experiencing

Use "scripts/enter.sh make rpi3_64" for rpi3 images, Failed in Step 7/7 :
------------------------------------------------------------------------------------
root@U16:/home/chengj/HAOS/haos# scripts/enter.sh make rpi3_64
Sending build context to Docker daemon  14.34kB
Step 1/7 : FROM debian:bullseye
 ---> 7df9c1a77ca2
Step 2/7 : SHELL ["/bin/bash", "-o", "pipefail", "-c"]
 ---> Using cache
 ---> 977860c9599e
Step 3/7 : RUN apt-get update && apt-get install -y --no-install-recommends         apt-transport-https         ca-certificates         curl         gpg-agent         gpg         dirmngr         software-properties-common     && curl -fsSL https://download.docker.com/linux/debian/gpg | apt-key add -     && add-apt-repository "deb https://download.docker.com/linux/debian $(lsb_release -cs) stable"     && apt-get update && apt-get install -y --no-install-recommends         docker-ce     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 3cc08215bbb5
Step 4/7 : RUN apt-get update && apt-get install -y --no-install-recommends         bash         bc         binutils         build-essential         bzip2         cpio         file         git         graphviz         jq         make         ncurses-dev         openssh-client         patch         perl         python3         python3-matplotlib         python-is-python3         qemu-utils         rsync         skopeo         sudo         unzip         vim         wget         zip     && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> 30d4b3a8557c
Step 5/7 : COPY scripts/entry.sh /usr/sbin/
 ---> Using cache
 ---> f6a4ded3b347
Step 6/7 : ENTRYPOINT ["/usr/sbin/entry.sh"]
 ---> Using cache
 ---> 8236e0b11fc0
Step 7/7 : WORKDIR /build
 ---> Using cache
 ---> 35c6bfe7a802
Successfully built 35c6bfe7a802
Successfully tagged hassos:local
config rpi3_64
/usr/bin/make -C /build/buildroot O=/build/output BR2_EXTERNAL=/build/buildroot-external "rpi3_64_defconfig"
make[1]: Entering directory '/build/buildroot'
  GEN     /build/output/Makefile
#
# configuration written to /build/output/.config
#
make[1]: Leaving directory '/build/buildroot'
build rpi3_64
/usr/bin/make -C /build/buildroot O=/build/output BR2_EXTERNAL=/build/buildroot-external VERSION_DEV="dev20230630"
make[1]: Entering directory '/build/buildroot'

/usr/bin/make -j1 O=/build/output HOSTCC="/usr/bin/gcc" HOSTCXX="/usr/bin/g++" syncconfig
make[2]: Entering directory '/build/buildroot'
  GEN     /build/output/Makefile
make[2]: Leaving directory '/build/buildroot'
>>> hassio 1.0.0 Installing to images directory
/build/buildroot-external/package/hassio/create-data-partition.sh "/build/output/build/hassio-1.0.0" "/build/output/images"
mke2fs 1.46.2 (28-Feb-2021)
Discarding device blocks: done
Creating filesystem with 327680 4k blocks and 81920 inodes
Filesystem UUID: 541b86fb-311d-42bb-b4d1-905fb75bccd8
Superblock backups stored on blocks:
        32768, 98304, 163840, 229376, 294912

Allocating group tables: done
Writing inode tables: done
Creating journal (8192 blocks): done
Writing superblocks and filesystem accounting information: done

Unable to find image 'docker:20.10-dind' locally
20.10-dind: Pulling from library/docker
8a49fdb3b6a5: Pull complete
db1d8fde5ab0: Pull complete
f3759a44eb9f: Pull complete
1465fd8ca4ab: Pull complete
95c1bc752a8b: Pull complete
f5ce33fb9a21: Pull complete
4d37eec86745: Pull complete
49d0e23604c0: Pull complete
ac56a29f30eb: Pull complete
0184c8e99504: Pull complete
23430c064101: Pull complete
e2a9b00eaf87: Pull complete
004faac3bc86: Pull complete
765a9f2e0717: Pull complete
Digest: sha256:af96c680a7e1f853ebdd50c1e0577e5df4089b033102546dd6417419564df3b5
Status: Downloaded newer image for docker:20.10-dind
Waiting for Docker daemon...
Loading container images...
Loaded image: ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.06.4
Loaded image: ghcr.io/home-assistant/aarch64-hassio-audio:2023.04.2
Loaded image: ghcr.io/home-assistant/aarch64-hassio-cli:2023.05.0
Loaded image: ghcr.io/home-assistant/aarch64-hassio-dns:2022.04.1
Loaded image: ghcr.io/home-assistant/aarch64-hassio-multicast:2022.02.0
Loaded image: ghcr.io/home-assistant/aarch64-hassio-observer:2022.10.0
Loaded image: ghcr.io/home-assistant/raspberrypi3-64-homeassistant:landingpage
wget: bad address 'version.home-assistant.io'
make[1]: *** [package/pkg-generic.mk:374: /build/output/build/hassio-1.0.0/.stamp_images_installed] Error 1
make[1]: Leaving directory '/build/buildroot'
make: *** [Makefile:39: rpi3_64] Error 2

--------------------------------------------------
And I have tested the domain 'version.home-assistant.io' in host, it is OK.
So what is the problem?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.git submodule update --init
2.scripts/enter.sh make rpi3_64
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

### System information

no

### Additional information

no

LauraLangdon, 2023-07-01:
+1. I'm trying to install HA on a Raspberry Pi 4 using Balena Etcher on MacOS14.

agners, 2023-07-03:
Hm, this sounds like a DNS issue of some kind. The build system tries to access the following location during build: https://version.home-assistant.io/stable.json. Can you  access the location from you system?

chengjing30, 2023-07-03:
> Hm, this sounds like a DNS issue of some kind. The build system tries to access the following location during build: https://version.home-assistant.io/stable.json. Can you access the location from you system?

yes，wget is ok

[image]

closed 2023-10-10 by github-actions[bot]