home-assistant/operating-system #2625: Sonnet 10gbe adapter not detected

sdholden28, 2023-07-02:
### Describe the issue you are experiencing

Sonnet Thunderbolt to 10gbe sfo+ is not detected by HAOS.

https://www.sonnettech.com/product/solo10g-sfp-tb3/overview.html

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect sonnet 10gbe adapter via Intel NUC thunderbolt port.
2. Boot HAOS
3. Adapter not detected. No Ethernet connection.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Issue is reproduced on clean HAOS installation. I do not see any errors in the logs that would be related.
```

### Anything in the Host logs that might be useful for us?

```txt
Issue is reproduced on clean HAOS installation. I do not see any errors in the logs that would be related.
```

agners, 2023-07-03:
Is the device working on other Linux distributions (e.g. Debian 11)? If so, can you please post the output of `dmesg` `lsmod` and/or `lspci` to identify which driver is being used.

sdholden28, 2023-07-03:
Supposedly drivers are included in Linux kernel 5.0 and above.

```
dmesg:

dmesg: 
```sh
[187673.190325] thunderbolt 0-1: new device found, vendor=0x8 device=0x36
[187673.190326] thunderbolt 0-1: Sonnet Technologies, Inc Solo 10G SFP+ Thunderbolt 3 Edition
[187673.440677] pcieport 0000:03:01.0: pciehp: Slot(1): Card present
[187673.440678] pcieport 0000:03:01.0: pciehp: Slot(1): Link Up
[187673.576497] pci 0000:05:00.0: [8086:15da] type 01 class 0x060400
[187673.576679] pci 0000:05:00.0: enabling Extended Tags
[187673.576966] pci 0000:05:00.0: supports D1 D2
[187673.576967] pci 0000:05:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.577192] pci 0000:05:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x4 link at 0000:03:01.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[187673.577268] pci 0000:05:00.0: Adding to iommu group 17
[187673.578925] pcieport 0000:03:01.0: ASPM: current common clock configuration is inconsistent, reconfiguring
[187673.579029] pci 0000:05:00.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[187673.579278] pci 0000:06:01.0: [8086:15da] type 01 class 0x060400
[187673.579489] pci 0000:06:01.0: enabling Extended Tags
[187673.579760] pci 0000:06:01.0: supports D1 D2
[187673.579760] pci 0000:06:01.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.580048] pci 0000:06:01.0: Adding to iommu group 17
[187673.581814] pci 0000:05:00.0: PCI bridge to [bus 06-39]
[187673.581832] pci 0000:05:00.0:   bridge window [io  0x0000-0x0fff]
[187673.581841] pci 0000:05:00.0:   bridge window [mem 0x00000000-0x000fffff]
[187673.581859] pci 0000:05:00.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[187673.581864] pci 0000:06:01.0: bridge configuration invalid ([bus 00-00]), reconfiguring
[187673.582087] pci 0000:07:00.0: [1d6a:80b1] type 00 class 0x020000
[187673.582210] pci 0000:07:00.0: reg 0x10: [mem 0x00000000-0x0000ffff 64bit]
[187673.582252] pci 0000:07:00.0: reg 0x18: [mem 0x00000000-0x00000fff 64bit]
[187673.582294] pci 0000:07:00.0: reg 0x20: [mem 0x00000000-0x003fffff 64bit]
[187673.582317] pci 0000:07:00.0: reg 0x30: [mem 0x00000000-0x0003ffff pref]
[187673.582345] pci 0000:07:00.0: enabling Extended Tags
[187673.582598] pci 0000:07:00.0: supports D1 D2
[187673.582598] pci 0000:07:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[187673.582752] pci 0000:07:00.0: 8.000 Gb/s available PCIe bandwidth, limited by 2.5 GT/s PCIe x4 link at 0000:03:01.0 (capable of 31.504 Gb/s with 8.0 GT/s PCIe x4 link)
[187673.582803] pci 0000:07:00.0: Adding to iommu group 17
[187673.583814] pci 0000:06:01.0: PCI bridge to [bus 07-39]
[187673.583832] pci 0000:06:01.0:   bridge window [io  0x0000-0x0fff]
[187673.583840] pci 0000:06:01.0:   bridge window [mem 0x00000000-0x000fffff]
[187673.583858] pci 0000:06:01.0:   bridge window [mem 0x00000000-0x000fffff 64bit pref]
[187673.583859] pci_bus 0000:07: busn_res: [bus 07-39] end is updated to 07
[187673.583868] pci_bus 0000:06: busn_res: [bus 06-39] end is updated to 07
[187673.583882] pci 0000:05:00.0: BAR 14: assigned [mem 0xd4000000-0xe9efffff]
[187673.583883] pci 0000:05:00.0: BAR 15: assigned [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.583884] pci 0000:05:00.0: BAR 13: assigned [io  0x5000-0x5fff]
[187673.583885] pci 0000:06:01.0: BAR 14: assigned [mem 0xd4000000-0xe9efffff]
[187673.583886] pci 0000:06:01.0: BAR 15: assigned [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.583886] pci 0000:06:01.0: BAR 13: assigned [io  0x5000-0x5fff]
[187673.583888] pci 0000:07:00.0: BAR 4: assigned [mem 0xd4000000-0xd43fffff 64bit]
[187673.583924] pci 0000:07:00.0: BAR 6: assigned [mem 0xd4400000-0xd443ffff pref]
[187673.583925] pci 0000:07:00.0: BAR 0: assigned [mem 0xd4440000-0xd444ffff 64bit]
[187673.583958] pci 0000:07:00.0: BAR 2: assigned [mem 0xd4450000-0xd4450fff 64bit]
[187673.583991] pci 0000:06:01.0: PCI bridge to [bus 07]
[187673.583996] pci 0000:06:01.0:   bridge window [io  0x5000-0x5fff]
[187673.584009] pci 0000:06:01.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584017] pci 0000:06:01.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584034] pci 0000:05:00.0: PCI bridge to [bus 06-07]
[187673.584038] pci 0000:05:00.0:   bridge window [io  0x5000-0x5fff]
[187673.584051] pci 0000:05:00.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584060] pci 0000:05:00.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584076] pcieport 0000:03:01.0: PCI bridge to [bus 05-39]
[187673.584079] pcieport 0000:03:01.0:   bridge window [io  0x5000-0x5fff]
[187673.584086] pcieport 0000:03:01.0:   bridge window [mem 0xd4000000-0xe9efffff]
[187673.584091] pcieport 0000:03:01.0:   bridge window [mem 0x90000000-0xb1ffffff 64bit pref]
[187673.584128] pcieport 0000:05:00.0: enabling device (0000 -> 0003)
[187673.584541] pcieport 0000:06:01.0: enabling device (0000 -> 0003)
[187673.585008] atlantic 0000:07:00.0: enabling device (0000 -> 0002)
```

sdholden28, 2023-07-03:
http://sonnettech.com/support/kb/kb.php?cat=548&expand=&action=a3#a3

closed 2023-10-10 by github-actions[bot]