home-assistant/operating-system #2630: HA can't find internet connection after upgrade 10.3

domo-greg, 2023-07-04:
### Describe the issue you are experiencing

Hello. 

After installing the 10.3 operator system, I no longer have an internet connection. I had to downgrade to version 9.5, with which I have no connection problems.

I'm running on a nipogi nuc with haos flashed directly to ssd. The network card is from realtek PCIe GBE Family.

I apologize in advance if any information is missing or if my English is bad. This is my first report on github
Translated from French by deepl

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Upgrade from system operator version 9.5 to system operator version 10.3

### Anything in the Supervisor logs that might be useful for us?

```txt
Not having a connection with version 10.3, I didn't have access to my instance and couldn't retrieve the
```

### Anything in the Host logs that might be useful for us?

```txt
Not having a connection with version 10.3, I didn't have access to my instance and couldn't retrieve the
```

### System information

## System Information

version | core-2023.6.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 5.15.90
arch | x86_64
timezone | Europe/Paris
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4996
Installed Version | 1.32.1
Stage | running
Available Repositories | 1268
Downloaded Repositories | 39

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 5 juillet 2023 à 02:00
relayer_connected | true
relayer_region | eu-central-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
remote_server | eu-central-1-4.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 9.5
-- | --
update_channel | stable
supervisor_version | supervisor-2023.06.4
agent_version | 1.4.1
docker_version | 20.10.22
disk_total | 234.0 GB
disk_used | 8.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.2), File editor (5.6.0), Mosquitto broker (6.2.1), Terminal & SSH (9.7.1), Zigbee2MQTT (1.32.0-2), Advanced SSH & Web Terminal (15.0.2), Home Assistant Google Drive Backup (0.111.1), CEC Scanner (3.0), Spotify Connect (0.12.4), go2rtc (1.5.0), AdGuard Home (4.8.10), Studio Code Server (5.8.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 5
-- | --
resources | 33
views | 23
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 24 juin 2023 à 20:15
-- | --
current_recorder_run | 3 juillet 2023 à 23:15
estimated_db_size | 392.53 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

Here's an image of the "ha info" and "network info" rendering on the CLI interface at the time of installation of version 10.3

[image]
[image]

And here, images of the hardware used on the mini pc

[image]
[image]

pvizeli, 2023-07-05:
Your network is disabled (see ha network info)

domo-greg, 2023-07-05:
> Votre réseau est désactivé (voir info réseau ha)

My network is active with version 9.5 (see my hardware configuration)

With later versions, the network is no longer recognized by HA.

agners, 2023-07-05:
Does this problem resolve when you downgrade? (E.g. using `ha os update --version 9.5`)?

Can you share the output of `ha> network info` from HAOS 9.5?

domo-greg, 2023-07-05:
Yes, the problem disappears with a downgrade to 9.5

Here's the network info

[image]

agners, 2023-07-05:
@domo-greg hm, so it really seems that something is wrong on HAOS 10.3.

Can you upgrade again and run the following command?

```
ha> login
# ip link show dev enp1s0
# ip addr show dev enp1s0
# dmesg | grep r8169
```

domo-greg, 2023-07-05:
I'll get back to you as soon as I can. 

Do you think you'll need more information via ha commande? If so, I'll leave version 10.3 installed until the problem is solved. If not, while waiting for you to find a solution, I'll put back version 9.5, as I don't have another machine reliable enough to run my home automation system.

domo-greg, 2023-07-05:
Here is the requested information 

[image]

Baxxy13, 2023-07-05:
you had a typo in the last command...
(dsmeg instead of correct dmesg)

domo-greg, 2023-07-05:
My apologies. Here is the correction 

[image]

agners, 2023-07-06:
Hm, it seems that your system simply considers the Ethernet interface to be down :thinking: 

Can you maybe share the output of `dmesg | grep r8169` on HAOS 9.5 for comparison?

To me it seems that this is a Linux kernel 6.1 bug. Is Ethernet working on other Linux 6.1 based distributions (e.g. Debian Bookworm)?

domo-greg, 2023-07-06:
Here is the command under 9.5

[image]

I don't know if this works on other distributions because I ran ubuntu on a bootable usb stick to flash my ssd. So I don't have any other distribution on hand to test.

AkazaRenn, 2023-07-08:
I'm having issue clean installing 10.3, could be related. Full log here from the initialization:

```log
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/udev.sh
[04:22:43] INFO: Using udev information from host
cont-init: info: /etc/cont-init.d/udev.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun supervisor (no readiness notification)
services-up: info: copying legacy longrun watchdog (no readiness notification)
s6-rc: info: service legacy-services successfully started
[04:22:43] INFO: Starting local supervisor watchdog...
23-07-08 04:22:46 INFO (MainThread) [__main__] Initializing Supervisor setup
23-07-08 00:22:46 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4-64
23-07-08 00:22:46 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor ghcr.io/home-assistant/aarch64-hassio-supervisor with version 2023.07.1
23-07-08 00:22:46 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
23-07-08 00:22:47 INFO (SyncWorker_1) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/aarch64-hassio-supervisor:2023.06.2']
23-07-08 00:22:49 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
23-07-08 00:22:49 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-07-08 00:22:49 INFO (MainThread) [__main__] Setting up Supervisor
23-07-08 00:22:49 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
23-07-08 00:22:49 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.manager] Connected to system D-Bus.
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.agent] Load dbus interface io.hass.os
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.hostname] Load dbus interface org.freedesktop.hostname1
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.logind] Load dbus interface org.freedesktop.login1
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.network] Load dbus interface org.freedesktop.NetworkManager
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.rauc] Load dbus interface de.pengutronix.rauc
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.resolved] Load dbus interface org.freedesktop.resolve1
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.systemd] Load dbus interface org.freedesktop.systemd1
23-07-08 00:22:49 INFO (MainThread) [supervisor.dbus.timedate] Load dbus interface org.freedesktop.timedate1
23-07-08 00:22:50 INFO (MainThread) [supervisor.host.services] Updating service information
23-07-08 00:22:50 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
23-07-08 00:22:51 INFO (MainThread) [supervisor.host.network] Updating local network information
23-07-08 00:22:51 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
23-07-08 00:22:51 INFO (MainThread) [supervisor.docker.monitor] Started docker events monitor
23-07-08 00:22:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2023.05.0
23-07-08 00:22:51 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1
23-07-08 00:22:52 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
23-07-08 00:22:52 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2023.04.2
23-07-08 00:22:52 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
23-07-08 00:22:52 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
23-07-08 00:22:52 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
23-07-08 00:22:52 INFO (MainThread) [supervisor.plugins.manager] cli does not have the latest version 2023.06.0, updating
23-07-08 00:22:52 INFO (MainThread) [supervisor.resolution.module] Create new issue IssueType.FREE_SPACE - ContextType.SYSTEM / None
23-07-08 00:22:52 WARNING (MainThread) [supervisor.jobs] 'PluginCli.update' blocked from execution, not enough free space (0.9GB) left on the device
23-07-08 00:22:52 ERROR (MainThread) [supervisor.plugins.manager] Can't update cli to 2023.06.0, the Supervisor healthy could be compromised!
23-07-08 00:22:52 INFO (MainThread) [supervisor.resolution.module] Create new suggestion SuggestionType.EXECUTE_UPDATE - ContextType.PLUGIN / cli
23-07-08 00:22:52 INFO (MainThread) [supervisor.resolution.module] Create new issue IssueType.UPDATE_FAILED - ContextType.PLUGIN / cli
23-07-08 00:22:52 INFO (MainThread) [supervisor.plugins.manager] dns does not have the latest version 2023.06.2, updating
23-07-08 00:22:52 WARNING (MainThread) [supervisor.jobs] 'PluginDns.update' blocked from execution, not enough free space (0.9GB) left on the device
[... cut ...]
23-07-08 00:22:57 INFO (MainThread) [supervisor.host.network] Updating local network information
23-07-08 00:22:58 INFO (MainThread) [supervisor.host.manager] Host information reload completed
23-07-08 00:22:58 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
23-07-08 00:22:58 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
23-07-08 00:22:58 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
```

agners, 2023-07-10:
There have been reports on the mailing list that buggy power saving modes could be the culprit. Can you check you BIOS settings if you can disable power saving options (in particular ASPM)?

See also: https://lore.kernel.org/netdev/37b1001d-688c-fa35-0d8a-cbbbae5e6fa8@gmail.com/T/

domo-greg, 2023-07-10:
I've searched the bios after aspm but can't find anything. 

Looking on the net, I came across this. 

[image]

This may be a lead for you. My linux skills don't allow me to know how to set up this command line directly in the HA CLI.
Perhaps you could tell me how to do it.

domo-greg, 2023-07-18:
Hi, I'm coming back to you to see if you've found a solution so that I can update my system?

agners, 2023-07-18:
Hm, `pcie_aspm=off` looks promising. You'd append it to the cmdline.txt file. e.g. like this from the console:

```
ha > login
# echo " pcie_aspm=off" >> /mnt/boot/cmdline.txt
```

You can check the current cmdline using:
```
cat /mnt/boot/cmdline.txt
```

domo-greg, 2023-07-18:
`cat /mnt/boot/cmdline.txt` return no such file or directory

[image]

I'm on holiday away from the terminal, so I can try to modify the file manually when I get back.

ColtigerFPV, 2023-07-31:
any news? got the same problem with my nipogi :-(

agners, 2023-07-31:
@ColtigerFPV the command in https://github.com/home-assistant/operating-system/issues/2630#issuecomment-1641055840 should work. You just can't run it from the SSH & Terminal add-on remotely, you need to be connected locally using monitor and keyboard,

domo-greg, 2023-07-31:
That's what I thought.
I'll do it when I get back from vacation 😉.

ColtigerFPV, 2023-07-31:
@agners i will try it later. thx. i cant run it from terminal cause i cant get a connection from my ha-server with other devices ;-)

PlayFaster, 2023-08-08:
Hi There. I also have a NiPro NUC, new, on which I am installing Home Assistant OS for the first time (had intended to migrate my Pi4 install to the NiPro if it worked). I am observing the same issue. Home Assistant OS will not obtain an IP address. 

@agners - I have attempted the `pcie_aspm=off` suggested fix, it does NOT work unfortunately. 

Note that: 
 - Home Assistant OS ver 9.5 is NOT affected. This works correctly, as IP address is obtained, HA boots and works, as per OP. 
 - v10.1, 10.2, 10.3 and 10.4 are all affected, none will obtain an IP address 
 - A fresh install of v10x has the issue, so does a working install of v9.5 that updates to 10x.
 - On the same NiPro NUC, Windows 11, Linux (Mint) and a Linux Mint USB boot disk all work as expected, obtain an IP address and can get online. 

I am happy to share any information requested that might help to resolve this issue. 

Thanks

PlayFaster, 2023-08-17:
Hi There. 

I just tried the (beta) Home Assistant OS ver 10.5 and it shows the same behaviour - Cannot get IP Address - sticks at Wait Until Kernal Time Synchronized during boot because of this. 

Hoping that something can be done to resolve this issue as, other then not working with HA OS v10x, the NiPoGi 'NUC' seems like a very good machine for the price. 

Thanks

PlayFaster, 2023-09-01:
Hi. Just wondering if there are any updates on this? I would like to be able to get HA OS working on my NiPoGi 'NUC' and capable of getting OS updates as well. Any thoughts appreciated. 

Thanks

ccorderor, 2023-09-17:
Any updates about the issue?

PlayFaster, 2023-10-01:
Hi. Like @ccorderor , I'm also wondering if there are any updates here, any likely estimate for  a fix? Thanks!

PlayFaster, 2023-10-09:
Hi There. I see a HA OS v11 Release Candidate just became available. Any info on whether this release addresses this issue? I'll give it a try myself in the coming days when time allows, just wondering if there was something in the release for this, can't see anything specific in teh release notes. Thanks

agners, 2023-10-09:
@PlayFaster can you run the commands from https://github.com/home-assistant/operating-system/issues/2630#issuecomment-1621698061? In particular, is your system also showing the error with `rtl_chipcmd_cond` in dmesg?

agners, 2023-10-09:
In general, Linux 6.1.53 comes with a fix named `r8169: fix ASPM-related issues on a number of systems with NIC version from RTL8168h`. It seems likely that this addresses the problem you are seeing.

HAOS 10.5 came with Linux  6.1.45, and HAOS 11.0.rc1 with 6.1.55. So quite likely that HAOS 11.0.rc1 addresses this problem.

ccorderor, 2023-10-09:
Hi,
I tried to other day a nightime dev build of the next 11.0 version and worked perfectly, so I think I can assure the problem has been solved

PlayFaster, 2023-10-10:
Hi Folks. I can also confirm that I have today been able to install the v11 RC1 onto my NiPoGi NUC and it is working successfully - Home Assistant installs and boots, similar to v9.5 but which none of the 10x releases could do. Thanks for the info and support here!

closed 2023-11-07 by agners