home-assistant/operating-system #2636: Cannot manage vlan connections with nmcli due to ipv4.dns-data member

Particpant, 2023-07-09:
### Describe the issue you are experiencing

If I try to modify any of my vlan connections, say to rename them, or to mark as 'never-default', I run into an error from nmcli:
```
nmcli con mod <uuid> ipv4.never-default 'yes'
Error: Failed to modify connection 'Supervisor enp3s0.8': ipv4.dns-data: unknown property
```

```
nmcli con mod <uuid> ipv4.dns-data ''
Error: invalid property 'dns-data': 'dns-data' not among [method, dns, dns-search, dns-options, dns-priority, addresses, gateway, routes, route-metric, route-table, routing-rules, replace-local-rule, ignore-auto-routes, ignore-auto-dns, dhcp-client-id, dhcp-iaid, dhcp-timeout, dhcp-send-hostname, dhcp-hostname, dhcp-fqdn, dhcp-hostname-flags, never-default, may-fail, required-timeout, dad-timeout, dhcp-vendor-class-identifier, link-local, dhcp-reject-servers, auto-route-ext-gw].
```

nmcli similarly refuses to apply modifications via the interactive editor.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Have vlan connections set up in nmcli (followed tutorial here: https://community.home-assistant.io/t/setup-vlan-and-ha-tutorial/87705/111)
2. Be on current HAOS (unknown if this is newly introduced or longstanding.
3. Attempt to modify any existing connection via nmcli
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I didn't see anything relevant in these logs
```

### Anything in the Host logs that might be useful for us?

```txt
I didn't see anything relevant in these logs
```

### System information

## System Information

version | core-2023.7.1
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.34
arch | x86_64
timezone | America/Los_Angeles
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.32.1
Stage | running
Available Repositories | 1272
Downloaded Repositories | 3

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | May 18, 2024 at 12:00 AM
relayer_connected | true
relayer_region | us-east-1
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | false
remote_server | us-east-1-8.ui.nabu.casa
certificate_status | ready
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.3
-- | --
update_channel | stable
supervisor_version | supervisor-2023.07.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 109.3 GB
disk_used | 13.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (10.0.2), File editor (5.6.0), Terminal & SSH (9.7.1), Advanced SSH & Web Terminal (15.0.3), Z-Wave JS UI (1.14.0), ESPHome (2023.6.4), Logitech Media Server (0.0.53)

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 0
views | 19
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | June 29, 2023 at 2:06 PM
-- | --
current_recorder_run | July 9, 2023 at 7:23 PM
estimated_db_size | 194.51 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

### Additional information

I've rebooted multiple times (both HA and the hardware itself) to no avail.
Additionally, I've seen a similar issue referenced elsewhere, which indicate a possible version mismatch between nmcli and NetworkManager:
https://gitlab.freedesktop.org/NetworkManager/NetworkManager/-/issues/1307

I wasn't able to run any of the suggested troubleshooting steps, as systemctl and journalctl aren't installed in HAOS:
```
~ # nmcli --version
nmcli tool, version 1.42.8
~ # systemctl restart NetworkManager.service
-bash: systemctl: command not found
~ # journalctl -b0 -u NetworkManager | grep 'is starting'
-bash: journalctl: command not found
```

agners, 2023-07-10:
Generally, you should not need to use nmcli directly. Instead use the `ha network` command for your networking needs.

What are you trying to achieve?

It sounds like there is a unsupported property in the connection settings, can you check if that is indeed the case in `cat /etc/NetworkManager/system-connections/Supervisor\ enp3s0.8.nmconnection`?

Particpant, 2023-07-10:
There's no way to assign a 'default' connection when setting up multiple vlans, or set the priority of any one network path over another, except via nmcli, as far as I can tell.

That command doesn't work, I've got an `/etc/NetworkManager/system-connections/Supervisor` folder, but there's no Supervisor subfolder.

the output of ` nmcli -p con show` for the connection doesn't have an ipv4.dns-data field.

```
-------------------------------------------------------------------------------
ipv4.method:                            manual
ipv4.dns:                               <ip address>
ipv4.dns-search:                        --
ipv4.dns-options:                       --
ipv4.dns-priority:                      0
ipv4.addresses:                        <ip address>/24
ipv4.gateway:                           <ip address>
ipv4.routes:                            --
ipv4.route-metric:                      -1
ipv4.route-table:                       0 (unspec)
ipv4.routing-rules:                     --
ipv4.replace-local-rule:                -1 (default)
ipv4.ignore-auto-routes:                no
ipv4.ignore-auto-dns:                   no
ipv4.dhcp-client-id:                    --
ipv4.dhcp-iaid:                         --
ipv4.dhcp-timeout:                      0 (default)
ipv4.dhcp-send-hostname:                yes
ipv4.dhcp-hostname:                     --
ipv4.dhcp-fqdn:                         --
ipv4.dhcp-hostname-flags:               0x0 (none)
ipv4.never-default:                     no
ipv4.may-fail:                          yes
ipv4.required-timeout:                  -1 (default)
ipv4.dad-timeout:                       -1 (default)
ipv4.dhcp-vendor-class-identifier:      --
ipv4.link-local:                        0 (default)
ipv4.dhcp-reject-servers:               --
ipv4.auto-route-ext-gw:                 -1 (default)
```

ys-li, 2023-07-13:
I am having this problem as well. Was trying to set static IPs for multiple interfaces.
I think the configuration steps are actually documented in the network.md in this repo. Following the instruction results in the error message mentioned.
[https://github.com/home-assistant/operating-system/blob/dev/Documentation/network.md#using-nmcli-to-set-a-static-ipv4-address](url)

dgarciadonado, 2023-07-14:
I think I have a similar issue. Trying to fully disable IPv6 get me an ipv4.dns-data error:
[image]

dgarciadonado, 2023-07-21:
@Particpant did you get a solution for this? Thanks!

Particpant, 2023-07-21:
No, I have not.

ryan-haver, 2023-08-03:
This is the same behavior seen when using the interactive editor....it will let me manually remove DNS and add DNS servers with the add and remove commands. With that said when saving, it validates the properties and rattles off the "ipv4.dns-data: unknown property" error if the DNS property has a value set.

I can, however, remove the DNS entry and save without issues, but now I don't have a DNS server configured....

[image]

MaxVonEvil, 2023-08-13:
Seeing exactly the same here. Thought I might be able to work around this by deleting the dns object and re-adding it. Nope. Same problem. dns-data gets added when you add a static route.. why?

To compound the problem; when re-adding the DNS server with ```set ipv4.dns 1.1.1.1```, it refuses to save as the dns entry is invalid? print ipv4.dns revealed a comma is added in front as ,1.1.1.1 - thought it might be expecting two DNS entries, so I tried set ```ipv4.dns 1.1.1.1, 1.0.0.1``` 

No dice either. I had to restore a snapshot to clean out the mess I made

fsiek, 2023-09-06:
Same here. Always get **ipv4.dns-data: unknown property** on saving a connection.
Is there a config file to edit the IP manually?

tismofied, 2023-09-20:
😿 came here was hoping for a fix. I have HA running on baremetal and I am getting the same error
`hassos: "10.5"
homeassistant: 2023.9.2
hostname: homeassistant
machine: generic-x86-64
operating_system: Home Assistant OS 10.5
supervisor: 2023.09.2
supported: true
supported_arch:
- amd64
- i386`

nberardi, 2023-10-02:
I am having the same issue on 2023.9.3

dgarciadonado, 2023-10-08:
Hi all,

It seems that on this link similar issue has been solved: [https://almalinux.discourse.group/t/unable-to-set-dns-with-nmcli-ipv4-dns/2442/4](url)

1. sudo dnf install NetworkManager-initscripts-updown
2. sudo ifdown enp1s0
3. sudo mv /etc/sysconfig/network-scripts/ifcfg-enp1s0 /root
4. sudo systemctl restart NetworkManager
5. edit connection with nmcli

My linux/network knowledge is so poor and barely know what these commands are doing so I am afraid of break something.
Maybe someone what more experience knows if these commands are harmless to try.

supermopf, 2023-10-11:
I have the same issue, but you can save the config if you remove the ipv4.dns setting:

```
nmcli connection edit "Your connection"
remove ipv4.dns
save
quit
```

Particpant, 2024-01-10:
don't close.

PeterBorfai, 2024-01-13:
Thanks for the workaround!
My problem was ethernet negotiation was turned off around 6th Jan, probably following the OS update. So the ethernet interface defaulted to 100Mbps. This caused my speedtest.net integration to measure only 90+ Mbps whereas my ISP is way faster than 100Mbps. 
nmcli con edit "Supervisor enp0s31f6"  # to select the interface 
nmcli> set 802-3-ethernet.auto-negotiate on     # fixes the problem
nmcli> save  # failed as ipv4.dns-data unknown property
nmcli> remove ipv4.dns  # as suggested by @supermopf above
nmcli> save  # now works
nmcli> quit
Then I had to host shutdown  as the interface had a "no carrier"  problem
Finally, in HA UI I restored the DNS server address.
Now the interface runs at 1Gbps and speedtest measures downloads around 850Mbps

This workaround should not be necessary. Can it not be fixed?

mpieru, 2024-01-31:
Hello, same issue here, please don't close the issue.

Verified/reproduced on HaOS 11.1 
The other thing I can observe is the following warning message when enter in the NMCLI:

Warning: nmcli (1.44.2) and NetworkManager (1.40.18) versions don't match. Restarting NetworkManager is advised.

I used nmcli on HaOS in the past, but I don't remember this message in previous HaOS versions.

Particpant, 2024-05-06:
still present

closed 2024-08-06 by agners