home-assistant/operating-system #2646: Backups not shown on Network Drive

requiemmg, 2023-07-20:
### Describe the issue you are experiencing

Backups are not shown on the Supervisor Frontend (Config->system->backups) when choosing a network drive.

[image]

Please also see this thread on the forums: https://community.home-assistant.io/t/backups-not-shown/592136

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

core-2023.7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Choose a Samba Share as the default backup location
2. Perform backups
3. Backups are physically being created but are not shown on the GUI which says "You don't have any backups yet"
4. To restore a backup, you then need to use the "Upload Backup" functionality

### Anything in the Supervisor logs that might be useful for us?

```txt
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon a0d7b954_nodered
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on 5c53de3b_esphome
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon 5c53de3b_esphome
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_letsencrypt
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_letsencrypt
23-07-20 08:18:51 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on db21ed7f_teamspeak
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon db21ed7f_teamspeak
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_silabs_multiprotocol
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_silabs_multiprotocol
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on core_samba
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon core_samba
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Building backup for add-on cebe7a76_hassio_google_drive_backup
23-07-20 08:20:28 INFO (MainThread) [supervisor.addons.addon] Finish backup for addon cebe7a76_hassio_google_drive_backup
23-07-20 08:20:29 INFO (MainThread) [supervisor.homeassistant.module] Backing up Home Assistant Core config folder
23-07-20 08:24:53 INFO (MainThread) [supervisor.homeassistant.module] Backup Home Assistant Core config folder done
23-07-20 08:24:53 INFO (MainThread) [supervisor.backups.manager] Backing up 38ac5a0a store folders
23-07-20 08:24:53 INFO (SyncWorker_1) [supervisor.backups.backup] Backing up folder share
23-07-20 08:24:53 INFO (SyncWorker_1) [supervisor.backups.backup] Backup folder share done
23-07-20 08:24:53 INFO (SyncWorker_0) [supervisor.backups.backup] Backing up folder addons/local
23-07-20 08:24:53 INFO (SyncWorker_0) [supervisor.backups.backup] Backup folder addons/local done
23-07-20 08:24:53 INFO (SyncWorker_5) [supervisor.backups.backup] Backing up folder ssl
23-07-20 08:24:53 INFO (SyncWorker_5) [supervisor.backups.backup] Backup folder ssl done
23-07-20 08:24:53 INFO (SyncWorker_4) [supervisor.backups.backup] Backing up folder media
23-07-20 08:24:53 INFO (SyncWorker_4) [supervisor.backups.backup] Backup folder media done
23-07-20 08:26:31 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
23-07-20 08:26:41 INFO (MainThread) [supervisor.backups.manager] Creating full backup with slug 38ac5a0a completed
23-07-20 08:26:42 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
23-07-20 08:26:42 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
23-07-20 08:26:42 INFO (MainThread) [supervisor.api.middleware.security] /backups/38ac5a0a/info access from cebe7a76_hassio_google_drive_backup
23-07-20 08:26:43 INFO (MainThread) [supervisor.api.middleware.security] /backups/8d89b45e access from cebe7a76_hassio_google_drive_backup
23-07-20 08:26:43 INFO (MainThread) [supervisor.backups.manager] Removed backup file 8d89b45e
23-07-20 08:26:43 INFO (MainThread) [supervisor.api.middleware.security] /backups/38ac5a0a/download access from cebe7a76_hassio_google_drive_backup
23-07-20 08:26:43 INFO (MainThread) [supervisor.api.backups] Downloading backup 38ac5a0a
23-07-20 08:29:46 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-130893' coro=<WSClient.start_listener() done, defined at /usr/src/supervisor/supervisor/homeassistant/websocket.py:88> exception=ConnectionResetError('Cannot write to closing transport')>
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/homeassistant/websocket.py", line 95, in start_listener
    await self._receive_json()
  File "/usr/src/supervisor/supervisor/homeassistant/websocket.py", line 104, in _receive_json
    msg = await self._client.receive()
[... 232 lines cut ...]
-- | --
resources | 7
views | 5
mode | storage

</details>

<details><summary>Miele</summary>

component_version | 0.1.14
-- | --
reach_miele_cloud | ok

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 18. Juli 2023 um 13:21
-- | --
current_recorder_run | 20. Juli 2023 um 07:47
estimated_db_size | 2647.94 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

agners, 2023-07-20:
> kernel: CIFS: VFS: bogus file nlink value 0

It seems something is wrong with that SMB server/share. Maybe the server manages the files on that share, which confuses the client (there is  technical descirption here: https://unix.stackexchange.com/questions/185161/what-are-bogus-file-nlink-values).

Does restarting the system brings them back?

requiemmg, 2023-07-20:
The current samba share is just a usb drive connected to my Router (AVM Fritz Box 6591 Cable) which offers a samba share. Behaviour doesn't change after either reboot of the router or the HA system. Behaviour was the same when the network share was located on a windows server system.
Sometimes the backups will start to show up after leaving the window open a while, most times not.

agners, 2023-07-20:
> Sometimes the backups will start to show up after leaving the window open a while, most times not.

Ah yes, loading them does take a while. But that is a Supervisor issue, it is not really related to the OS.

requiemmg, 2023-07-20:
Taking a while is ok.
But as I said:
1. Most times it wont load at all
2. If it takes a while to load,it should say "loading", not "you have no backups" 😉