home-assistant/operating-system #2648: Stop HA OS upgrading itself

26tajeen, 2023-07-20:
### Describe the issue you are experiencing

Does anybody know how I can stop HA updating its OS beyond v9? If it updates to 10 it breaks my installation due to a bug with 10. I’ve asked this [before](https://community.home-assistant.io/t/bit-of-a-situation-around-mosquitto-add-on-hass-os-v10/581154) and got nowhere but it’s just happened again.

It sometimes seems to update itself when the hardware is power cycled, which due to an unstable power supply, happens from time to time.

thank you

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

9.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

### Additional information

.

agners, 2023-07-20:
Home Assistant never updates the OS automatically.

We use a A/B type of system, this means at any given time there are two HAOS versions installed. What could happen here is that you've updated to OS 10 once, and it fall back to OS 9.x. 

The OS will usually try booting 3 times, before falling back to the "other" installation.

Now if you interrupt the boot process, it will fall back to OS 10 again.

agners, 2023-07-20:
> It sometimes seems to update itself when the hardware is power cycled, which due to an unstable power supply, happens from time to time.

If the power supply is unstable during boot up, that is really bad for HAOS indeed, but this is a hardware issue, the software runs as intended...

26tajeen, 2023-07-20:
Thanks - two questions if I may?

1) Is it possible to delete all traces of the OS 10 so I only have OS 9 on my Pi?
2) Is it possible to create an automation that automatically downgrades if OS 10 is 'detected' ?

thank you for your help - good to know it isn't automatically updating.

agners, 2023-07-20:
> Is it possible to delete all traces of the OS 10 so I only have OS 9 on my Pi?

Probably the easiest way to do is to create a backup and install a fresh HAOS 9.x...

> Is it possible to create an automation that automatically downgrades if OS 10 is 'detected' ?

Not really, there is no API to control which HAOS boot slot gets started. We probably could extend the Supervisor API to expose this, but it would need implementation work.

closed 2023-10-27 by github-actions[bot]