home-assistant/operating-system #2655: Swap file auto created and wasting disk space on systems with sufficient RAM

remygithub, 2023-07-26:
### Describe the issue you are experiencing

I am using HAOS (x86-64) on a re-purposed thin client with 16G of RAM and 15G SSD. 

On every reboot the system recreates a 5GB `/mn/data/swapfile` , which, obviously, is not needed and wastes disk space.

I can log in as root, unmount and delete the swap file to regain space but it is not ideal as the file is re-created on reboot.
It would be far more preferable to be able to disable swap completely. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Log in as root
2. `swapoff -a`
3. `rm /mnt/data/swapfile`
4. Reboot
5. `/mnt/data/swapfile` is re-created (5GB)

### Anything in the Supervisor logs that might be useful for us?

```txt
No problems. System runs as expected except for the re-creation of the useless swap file.
```

### Anything in the Host logs that might be useful for us?

```txt
No problems. System runs as expected except for the re-creation of the useless swap file.
```

### System information

## System Information

version | core-2023.7.3
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.34
arch | x86_64
timezone | America/New_York
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 1271
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | December 31, 2017 at 19:00
relayer_connected | false
relayer_region | null
remote_enabled | true
remote_connected | false
alexa_enabled | false
google_enabled | false
remote_server | null
certificate_status | null
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.3
-- | --
update_channel | stable
supervisor_version | supervisor-2023.07.1
agent_version | 1.5.1
docker_version | 23.0.6
disk_total | 14.0 GB
disk_used | 8.4 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.6.0), ESPHome (2022.3.1), Signal Messenger (0.67.1), Terminal & SSH (9.7.1), Samba share (10.0.2), InfluxDB (4.7.0), Grafana (9.0.2)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 0
views | 5
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 19, 2023 at 18:37
-- | --
current_recorder_run | July 26, 2023 at 12:35
estimated_db_size | 114.58 MiB
database_engine | sqlite
database_version | 3.41.2

</details>

closed 2023-07-31 by github-actions[bot]

agners, 2023-07-31:
Keep in mind that swap can be helpful even on systems with a lot of RAM (see [In defence of swap: common misconceptions](https://chrisdown.name/2018/01/02/in-defence-of-swap.html)). In the end its a trade-off, the current default is good for most cases. Ideally we'd add an advanced option to configure swap settings.

remygithub, 2023-08-11:
No argument against swap per se from me.  But in certain cases when RAM is clearly plenty and disk is not it would be nice to be able to disable it.  
Hopefully HAOS will get that advanced option soon.  
Connecting as root and deleting the swap file is not ideal and definitely not good for the SSD itself.

remygithub, 2023-08-11:
Thank you for your work on HAOS.  I know very well how hard it is to find the "extra" time!

I'm guessing that the main change in this case would the modification of the `/etc/fstab` file in the overlay with additional logic connecting the config option.  Maybe changing the swap creation script to read the option first and unmount and delete the swap file if already created.  Does this sound about right?  Sorry if I'm totally off...it's been a while since I read the HAOS docs.

I should just work on this instead of adding more to your plate... I apologize for not being able to at this time!

iakat, 2024-08-10:
workaround:

rm /mnt/data/swapfile; touch /mnt/data/swapfile; chattr +i /mnt/data/swapfile

felipecrs, 2024-11-13:
A more complete answer:

1. To disable swap permanently, run the following command in the Advanced Terminal add-on with Protection Mode disabled:

```console
docker run --privileged --pid=host -it alpine nsenter --target 1 --mount --uts --net --ipc sh -xec \ 
  'swapoff -a; rm -f /mnt/data/swapfile; touch /mnt/data/swapfile; chattr +i /mnt/data/swapfile'
```

2. To re-enable it:

```console
docker run --privileged --pid=host -it alpine nsenter --target 1 --mount --uts --net --ipc sh -xec \ 
  'chattr -i /mnt/data/swapfile; rm -f /mnt/data/swapfile; swapon -a'
```