home-assistant/operating-system #2661: HA OS 10 Bluetooth issues on Intel X86

gaborbernat, 2023-08-01:
### Describe the issue you are experiencing

Noticed the issue for the last month, SwitchBot integration was constantly breaking.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Once startup the bluetooth starts failing.

### Anything in the Supervisor logs that might be useful for us?

```txt
Nope.
```

### Anything in the Host logs that might be useful for us?

```txt
Nope.
```

### System information

Running on https://www.amazon.com/dp/B0BS369VMF

### Additional information

Not much else.

OH010, 2023-08-11:
Same problem here, however my Bluetooth device is working regardless of the constant kernel error logs. I have an Intel system with onboard bluetooth device (Asus motherboard: https://rog.asus.com/uk/motherboards/rog-strix/rog-strix-z270-i-gaming-model/ ). The errors stop when I disable the Bluetooth device in HAOS but then the InkBird IBS-TH2/P01B thermometer stops working. I also tried Passive Mode, the errors stop but so does the Inkbird. 

[core-ssh ~]$ ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100
Aug 10 12:51:06 homeassistant kernel: Bluetooth: Core ver 2.22
Aug 10 12:51:06 homeassistant kernel: Bluetooth: HCI device and connection manager initialized
Aug 10 12:51:06 homeassistant kernel: Bluetooth: HCI socket layer initialized
Aug 10 12:51:06 homeassistant kernel: Bluetooth: L2CAP socket layer initialized
Aug 10 12:51:06 homeassistant kernel: Bluetooth: SCO socket layer initialized
Aug 10 12:51:10 homeassistant kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Aug 10 12:51:10 homeassistant kernel: Bluetooth: BNEP filters: protocol multicast
Aug 10 12:51:10 homeassistant kernel: Bluetooth: BNEP socket layer initialized
Aug 10 12:51:10 homeassistant kernel: Bluetooth: MGMT ver 1.22
Aug 10 12:51:14 homeassistant kernel: Bluetooth: RFCOMM TTY layer initialized
Aug 10 12:51:14 homeassistant kernel: Bluetooth: RFCOMM socket layer initialized
Aug 10 12:51:14 homeassistant kernel: Bluetooth: RFCOMM ver 1.11
Aug 10 12:55:54 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:05 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:15 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:26 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:37 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:48 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:56:58 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:57:09 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:57:20 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:57:31 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:57:41 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:57:52 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:03 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:14 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:24 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:35 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:46 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:58:57 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:59:07 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:59:18 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:59:29 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:59:40 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 12:59:50 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 13:00:01 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 13:00:12 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005
Aug 10 13:00:23 homeassistant kernel: Bluetooth: hci0: unexpected event for opcode 0x2005

gaborbernat, 2023-08-17:
https://www.amazon.com/dp/B0BS369VMF, not sure about the adapter/card.

roccojay12, 2023-08-21:
Same issue here, I've had it since HAOS 10 was introduced. Using raspberry pi 4 8GB

agners, 2023-08-21:
@roccojay12 "same" as which report? Which opcode do you get? Can you share the output of `ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100` as well? What hardware are you using?

roccojay12, 2023-08-21:
> @roccojay12 "same" as which report? Which opcode do you get? Can you share the output of `ha host logs -t kernel -n 1000000 -b 0 | grep Bluetooth | head -n 100` as well? What hardware are you using?

The Issues I get is that once upgraded to 10.4 the switchbot integration will work for about a minute and then after that whenever I try and activate the switchbot bot it will not work until I have reloaded the Bluetooth integration but this then needs to be done on every switchbot bot activation. 

I've tried to get the logs but all I get is this: [image]

makgiber1984, 2023-08-30:
Same problem, I have a bmax b1 plus exclusively for Home Assistant and I had to downgrade to 10.4, because with the 10.5 Bluetooth stops working.

OH010, 2023-09-27:
Any update on this issue? I am still getting ha logs flooded with these errror every 10 seconds. Bluetooth is working fortunately but would be great to stop the log spam. Many thanks!

roccojay12, 2023-10-08:
@OH010, nothing yet I don't think. Bluetooth is still not working for me. Tried different adapters but nothing. Still on OS 9.5

rhpijnacker, 2023-11-21:
See also this discussion here ... https://github.com/home-assistant/core/issues/92839

closed 2024-02-28 by github-actions[bot]