home-assistant/operating-system #2663: New instalation of HAOS

FabianoFlorentin, 2023-08-01:
### Describe the issue you are experiencing

After a sucessfull instalation of HAOS, on the first atempt to acess HA and after typing user credentials, I'm stuck on the second screen because the country and currency selectors are not present. 
I tried installing on a TrueNas Scale VM and on a Windows VirtualBox VM wit the same result. I even tried older versions of the HAOS.

[image]

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.4 / 2023.7.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

After a new clean instalation

### Anything in the Supervisor logs that might be useful for us?

```txt
I'm not able to access the system
```

### Anything in the Host logs that might be useful for us?

```txt
I'm not able to access the system
```

### System information

I'm not able to access the system

closed 2023-08-03 by github-actions[bot]

agners, 2023-08-03:
First of, onboarding is not part of the operating system, so technically this is the wrong bug tracker.

That said: Onboarding got very recently reworked (with the 2023.8.0 release released yesterday). I've quickly checked onboarding using Portuguese as language, and it seems to work with the latest onboarding flow. Can you try onboarding again with a new installation?

FabianoFlorentin, 2023-08-03:
I followed @agners instructions and made a new instalation. This time I was able to finish the onboarding. 

Thanks.