home-assistant/operating-system #2669: hassio_dns docker image crashing, system unusable.

whejl, 2023-08-03:
### Describe the issue you are experiencing

With hassio_dns version 2023.6.2 the following happens:

s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
cont-init: info: running /etc/cont-init.d/corefile.sh
cont-init: info: /etc/cont-init.d/corefile.sh exited 0
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
services-up: info: copying legacy longrun coredns (no readiness notification)
**/command/s6-sleep: line 1: x-@: not found
/command/s6-sleep: line 2: syntax error: unterminated quoted string**
s6-rc: warning: unable to start service legacy-services: command exited 2
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
[19:50:12] INFO: Service restart after closing

The image crashes/exits, and leaves HA with no working DNS. Recorder is not found and 
ha dns downgrade --version 2022.04.1 works, for a while until supervisor upgrades automatically.
2022.04.1 is the latest working version. 2023.6.0, 2023.6.1 also bad.

I have downgraded OS 10.4, 10.2, 10.0 makes no difference.
HA downgraded to 2023.5, upgraded to 2023.8.0 no difference.

Possibly the issue arise when upgrading OS 10.3 and HA 7.1

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

10.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. It reproduces itself when supervisor upgrades the hassio_dns instance.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
23-08-04 00:05:08 INFO (SyncWorker_7) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/aarch64-hassio-dns:2023.06.2 to ghcr.io/home-assistant/aarch64-hassio-dns:2022.04.1
23-08-04 00:05:08 INFO (SyncWorker_7) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/aarch64-hassio-dns with tag 2022.04.1.
23-08-04 00:05:39 INFO (SyncWorker_7) [supervisor.docker.interface] Cleaning hassio_dns application
23-08-04 00:05:39 INFO (SyncWorker_3) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/aarch64-hassio-dns:2023.06.2']
23-08-04 00:05:40 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
23-08-04 00:05:41 INFO (SyncWorker_2) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2022.04.1 - 172.30.32.3
```

### Anything in the Host logs that might be useful for us?

```txt
Aug 03 22:04:00 homeassistant dockerd[572]: time="2023-08-03T22:04:00.520146033Z" level=warning msg="[resolver] failed to read from DNS server: 172.30.32.3:53, query: ;acme-v02.api.letsencrypt.org.\tIN\t AAAA" error="read udp 172.30.33.2:34595->172.30.32.3:53: i/o
```

### System information

## System Information

version | core-2023.8.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.11.4
os_name | Linux
os_version | 6.1.21-v8
arch | aarch64
timezone | Europe/Stockholm
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4903
Installed Version | 1.32.1
Stage | running
Available Repositories | 1268
Downloaded Repositories | 10

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Easee EV Charger</summary>

component_version | 0.9.54
-- | --
reach_easee_cloud | ok
connected2stream | false

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 10.0
-- | --
update_channel | stable
supervisor_version | supervisor-2023.07.1
agent_version | 1.5.1
docker_version | 23.0.3
disk_total | 28.0 GB
disk_used | 9.7 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.2.1), Samba share (10.0.2), File editor (5.6.0), Duck DNS (1.15.0), MariaDB (2.6.1), Advanced SSH & Web Terminal (15.0.5)

</details>

<details><summary>Dashboards</summary>

dashboards | 1
-- | --
resources | 2
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 juli 2023 kl. 23:28
-- | --
current_recorder_run | 4 augusti 2023 kl. 00:00
estimated_db_size | 527.92 MiB
database_engine | mysql
database_version | 10.6.12

</details>

agners, 2023-08-04:
```
/command/s6-sleep: line 1: x-@: not found
/command/s6-sleep: line 2: syntax error: unterminated quoted string
```
I don't see such errors in my installation. My best guess would be that your file system got corrupted somehow.

If you have shell access, you can try to remote the image using `ha > login` then `docker image rm ghcr.io/home-assistant/amd64-hassio-dns:2023.06.2`.

There might be other files affected too, maybe the best course of action is to take a backup and install HAOS 10.4 on a new SD card, and restore the backup.

agners, 2023-08-07:
> removing the image, "not found" because downgraded, so upgrading/downgrading seems to remove the image every time, the issue might not have been in the image itself.

Docker uses layers, and they might have been retained even though you removed/re-downloaded a particular image/tag.

> Reinstalling to fresh SD card and restoring backup worked. The DNS image of 2023.06.2 is running and working, and most of my HA instance is also working correctly.

Ok, that confirms it was somehow related with your particular installation. Glad that helped to resolve it.

Thanks for the update!

closed 2023-08-07 by agners