home-assistant/operating-system #2670: Update to HA OS 10.4 failing: Destination device '/dev/disk/by-partlabel/hassos-boot' not found

user45876, 2023-08-05:
### Describe the issue you are experiencing

 When trying to 7pdate to HA OS 10.4, I am getting this:
`
23-08-05 21:08:51 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/10.4/haos_generic-x86-64-10.4.raucb
23-08-05 21:08:57 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-10.4.raucb
23-08-05 21:08:58 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found`

Filesystem seems to be cleanly dismounted and mounted when rebooting - are the error messages something?

`
-- Boot ebeefabbaaf3449fb24c1ef370e1862e --
Jul 21 19:22:34 homeassistant systemd[1]: Starting File System Check on /dev/disk/by-label/hassos-boot...
Jul 21 19:22:34 homeassistant systemd-fsck[197]: Cannot initialize conversion from codepage 850 to ANSI_X3.4-1968: Invalid argument
Jul 21 19:22:34 homeassistant systemd-fsck[197]: Cannot initialize conversion from ANSI_X3.4-1968 to codepage 850: Invalid argument
Jul 21 19:22:34 homeassistant systemd-fsck[197]: Using internal CP850 conversion table
Jul 21 19:22:34 homeassistant systemd-fsck[197]: fsck.fat 4.2 (2021-01-31)
Jul 21 19:22:34 homeassistant systemd-fsck[197]: /dev/nvme0n1p1: 10 files, 604/16343 clusters
Jul 21 19:22:34 homeassistant systemd[1]: Finished File System Check on /dev/disk/by-label/hassos-boot.
-- Boot 66857dba201a46afacd1d875db7fe8d5 --
Aug 05 18:12:52 homeassistant systemd[1]: Starting File System Check on /dev/disk/by-label/hassos-boot...
Aug 05 18:12:52 homeassistant systemd-fsck[201]: Cannot initialize conversion from codepage 850 to ANSI_X3.4-1968: Invalid argument
Aug 05 18:12:52 homeassistant systemd-fsck[201]: Cannot initialize conversion from ANSI_X3.4-1968 to codepage 850: Invalid argument
Aug 05 18:12:52 homeassistant systemd-fsck[201]: Using internal CP850 conversion table
Aug 05 18:12:52 homeassistant systemd-fsck[201]: fsck.fat 4.2 (2021-01-31)
Aug 05 18:12:52 homeassistant systemd-fsck[201]: /dev/nvme0n1p1: 10 files, 604/16343 clusters
Aug 05 18:12:52 homeassistant systemd[1]: Finished File System Check on /dev/disk/by-label/hassos-boot.`

And this:
`
-- Boot 66857dba201a46afacd1d875db7fe8d5 --
Aug 05 18:12:54 homeassistant systemd[1]: Starting RAUC Update Service...
Aug 05 18:12:54 homeassistant rauc[398]: Using central status file /mnt/data/rauc.db
Aug 05 18:12:54 homeassistant rauc[398]: Using central status file /mnt/data/rauc.db
Aug 05 18:12:54 homeassistant systemd[1]: Started RAUC Update Service.
Aug 05 18:13:23 homeassistant rauc[398]: Failed to resolve realpath for '/dev/disk/by-partlabel/hassos-system1'                                                                      Aug 05 18:13:23 homeassistant rauc[398]: rauc mark: marked slot kernel.1 as good
Aug 05 18:46:20 homeassistant rauc[398]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Aug 05 18:46:20 homeassistant rauc[398]: Active slot bootname: B                                                                                                                     Aug 05 18:46:20 homeassistant rauc[398]: input bundle: /mnt/data/supervisor/tmp/hassos-10.4.raucb
Aug 05 18:46:20 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: started
Aug 05 18:46:20 homeassistant rauc[398]: Failed to resolve realpath for '/dev/disk/by-partlabel/hassos-system1'
Aug 05 18:46:20 homeassistant rauc[398]: Reading bundle: /mnt/data/supervisor/tmp/hassos-10.4.raucb
Aug 05 18:46:20 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: Checking and mounting bundle...
Aug 05 18:46:20 homeassistant rauc[398]: Detected CRL but CRL checking is disabled!
Aug 05 18:46:20 homeassistant rauc[398]: Verifying bundle signature...
Aug 05 18:46:20 homeassistant rauc[398]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Aug 05 18:46:20 homeassistant rauc[398]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-10.4.raucb' to '/run/rauc/bundle'
Aug 05 18:46:20 homeassistant rauc[398]: Configured loop device '/dev/loop0' for 212475904 bytes
Aug 05 18:46:20 homeassistant rauc[398]: Running bundle hook install-check
Aug 05 18:46:20 homeassistant rauc[398]: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found
Aug 05 18:46:20 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not foundAug 05 18:46:20 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: finished
Aug 05 18:46:20 homeassistant rauc[398]: installing `/mnt/data/supervisor/tmp/hassos-10.4.raucb` failed: 1                                                                           Aug 05 19:08:57 homeassistant rauc[398]: Using deprecated 'Install' D-Bus Method (replaced by 'InstallBundle')
Aug 05 19:08:57 homeassistant rauc[398]: input bundle: /mnt/data/supervisor/tmp/hassos-10.4.raucb                                                                                    Aug 05 19:08:57 homeassistant rauc[398]: Active slot bootname: B                                                                                                                     Aug 05 19:08:57 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: started                                                                              Aug 05 19:08:57 homeassistant rauc[398]: Failed to resolve realpath for '/dev/disk/by-partlabel/hassos-system1'                                                                      Aug 05 19:08:57 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: Checking and mounting bundle...                                                      Aug 05 19:08:57 homeassistant rauc[398]: Reading bundle: /mnt/data/supervisor/tmp/hassos-10.4.raucb                                                                                  Aug 05 19:08:57 homeassistant rauc[398]: Detected CRL but CRL checking is disabled!
Aug 05 19:08:57 homeassistant rauc[398]: Verifying bundle signature...
Aug 05 19:08:57 homeassistant rauc[398]: Verified detached signature by 'O = HassOS, CN = HassOS GitHub-1'
Aug 05 19:08:57 homeassistant rauc[398]: Mounting bundle '/mnt/data/supervisor/tmp/hassos-10.4.raucb' to '/run/rauc/bundle'
Aug 05 19:08:58 homeassistant rauc[398]: Configured loop device '/dev/loop0' for 212475904 bytes
Aug 05 19:08:58 homeassistant rauc[398]: Running bundle hook install-check
Aug 05 19:08:58 homeassistant rauc[398]: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not found
Aug 05 19:08:58 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: Installation error: Destination device '/dev/disk/by-partlabel/hassos-boot' not foundAug 05 19:08:58 homeassistant rauc[398]: installing /mnt/data/supervisor/tmp/hassos-10.4.raucb: finished                                                                             Aug 05 19:08:58 homeassistant rauc[398]: installing `/mnt/data/supervisor/tmp/hassos-10.4.raucb` failed: 1`

What may I be doing wrong? Past issues like this have pointed to filesystem corruption. It is on an m.2 SSD, there should not be any corruption. How would I do an tacho?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

10.4

[... 101 lines cut ...]
mode | storage

</details>

<details><summary>NextDNS</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 6. Juli 2023 um 18:55
-- | --
current_recorder_run | 5. August 2023 um 20:13
estimated_db_size | 1463.03 MiB
database_engine | mysql
database_version | 10.6.12

</details>

### Additional information

Thanks, guys 😁

agners, 2023-08-07:
> Actually, there is no /dev/disk/by-partlabel/ on my installation:

Hm, there should be. From my generic-x86-64 installation:

```
# ls -la /dev/disk/by-partlabel
total 0
drwxr-xr-x    2 root     root           200 Jul 12 11:17 .
drwxr-xr-x    9 root     root           180 Jul 12 11:17 ..
lrwxrwxrwx    1 root     root            10 Jul 12 11:17 hassos-boot -> ../../sda1
...
```

What is `fdisk -l /dev/nvme0n1` saying?

If the GPT partition label is lost, maybe other things are corrupted too. Probably the best fix is to take a full backup and start with a new HAOS 10.4 installation.

user45876, 2023-08-07:
Thanks, Stefan. This is the output:

```
fdisk -l /dev/nvme0n1
Disk /dev/nvme0n1: 465.76 GiB, 500107862016 bytes, 976773168 sectors
Disk model: CT500P3SSD8
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: DB832910-E6F3-94C2-C315-7931DD297683

Device           Start       End   Sectors   Size Type
/dev/nvme0n1p1    2048     67583     65536    32M EFI System
/dev/nvme0n1p2   69632    118783     49152    24M Linux filesystem
/dev/nvme0n1p3  120832    645119    524288   256M Linux filesystem
/dev/nvme0n1p4  647168    696319     49152    24M Linux filesystem
/dev/nvme0n1p5  698368   1222655    524288   256M Linux filesystem
/dev/nvme0n1p6 1224704   1241087     16384     8M Linux filesystem
/dev/nvme0n1p7 1243136   1439743    196608    96M Linux filesystem
/dev/nvme0n1p8 1441792 976773134 975331343 465.1G Linux filesystem
```
I am unable to grok it myself - is that good? I can not see any obvious signs of corruption.

Restore is an option, I would however try to avoid it, as it is not as hassle free, as it should be. The installation is not very old, I just put in the 'new' SSD several weeks ago.

user45876, 2023-11-06:
I restored from backup and got it fixed that way.

closed 2023-11-06 by user45876