home-assistant/operating-system #2675: Watchdog can freeze ODROID-M1 and ODROID-XU4 (maybe others)

agners, 2023-08-08:
### Describe the issue you are experiencing

Since #2628 watchdog is enabled in systemd.

It seems that when stressing the system with `stress-ng`, it can trigger a watchdog reset by systemd. Typically, this should lead to a reboot. However, it seems that certain embedded boards have bugs in their watchdog implementation or elsewhere, which causes the system to not reboot but just freeze.

This has been observed with ODROID-M1 and ODROID-XU4

### What operating system image do you use?

odroid-m1 (Hardkernel ODROID-M1)

### What version of Home Assistant Operating System is installed?

11.0.dev20230803

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Boot the board with the dev version
2. Use `stress-ng --all 4` to stress the system a lot
3. Wait for 5-10 minutes until a the board stops responding

A easier way is to use a stable release (e.g. 10.4) which does not have systemd watchdog enabled:
1. Use `cat /dev/watchdog`  to start the watchdog (this will lead to `cat: read error: Invalid argument`, but at this point the watchdog has been started because the device file has been opened.
2. Wait for a freeze to happen

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
Some board warn that the watchdog continues to run after catting the file:

Aug 08 21:03:35 ha-shelf2-om1 kernel: watchdog: watchdog0: watchdog did not stop!
```
```

mingzhangqun, 2023-08-30:
Hi, everyone. I've tried enable BR2_INIT_BUSYBOX and disable BR2_INIT_SYSTEMD, the watchdog freeze also.
So I assume it has nothing to do with systemd.

[watchdog.txt](https://github.com/home-assistant/operating-system/files/12474480/watchdog.txt)

mingzhangqun, 2023-08-30:
Using the offical source code (linux 5.10) the watchdog works well. I'm tring the newest offical sdk(linux 6.1).

mingzhangqun, 2023-08-31:
I've tried official sdk(linux 6.1), the watchdog works well.
I replaced the kernel partition(`dd if=kernel.squash of=/dev/sdx2`) with haos, the watchdog works too.
[kernel_img.zip](https://github.com/home-assistant/operating-system/files/12484672/kernel_img.zip)