home-assistant/operating-system #2676: HAOS 10.4 will not boot on fresh install on NUC7i5

TheBobC, 2023-08-09:
### Describe the issue you are experiencing

After flashing image from Balena, rebooting only leads to the following on the screen:

Booting `Slot A (OK=0 TRY=0) ’

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

None yet

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Here is what i have done
Equipment
Intel NUC7i5DNHE1
Bios DNi50079 (updated during this process)
SSD TEAMGROUP MP33 1TB SLC Cache 3D NAND TLC NVMe 1.3
8 GB of Teamgroup 2133 (2@4GB ea)

Process

Downloaded Ubuntu Desktop Image and used Balena Etcher to create bootable USB

Went in to BIOS and set UEFI and disabled secure boot

Booted into Ubuntu on the NUC via USB

Installed Balena Etcher via this process:
sudo add-apt-repository universe -y
sudo apt update
sudo apt install libfuse2

sudo apt update
wget ‘https://github.com/balena-io/etcher/releases/download/v1.14.3/balena-etcher_1.14.3_amd64.deb’
sudo dpkg -i balena-etcher_1.14.3_amd64.deb
sudo rm balena-etcher_1.14.3_amd64.deb
sudo apt update -y
sudo apt --fix-broken install -y

Opened Balena Etcher

Choose Install from URL

Pasted the following in to Balena:
https://github.com/home-assistant/operating-system/releases/download/10.4/haos_generic-x86-64-10.4.img.xz

Choose the hard drive as the target

Hit Flash

When I reboot and remove the USB after flash, the attempted bootup gives me:
Booting `Slot A (OK=0 TRY=0) ’

I have tried the following:

Download the HAOS file and flash from file - same result
Update BIOS - same result
Download a previous version (9.5) and flash from file - same result
Download an even earlier version (8.0) and flash from file - same result

### Anything in the Supervisor logs that might be useful for us?

```txt
No logs as it will not boot to get that far
```

### Anything in the Host logs that might be useful for us?

```txt
No logs as it will not boot to get that far
```

### System information

Intel NUC7i5DNHE1
Bios DNi50079 (updated during this process)
SSD TEAMGROUP MP33 1TB SLC Cache 3D NAND TLC NVMe 1.3
8 GB of Teamgroup 2133 (2@4GB ea)

agners, 2023-08-09:
Hm, sounds like the boot loader crashes very late, or the Linux kernel very early :thinking: 

You can try to enable `earlycon` in the kernel parameters and see if the kernel print something (press e in the Grub menu, then add `earlycon` or/and `earlycon=efifb` to the end of the `linux ...` line, then press F10 to start boot).

TheBobC, 2023-08-09:
Thank you for that information. I have attached a shot of the output from that. Not sure what to do with that…

Bob Campbell

agners, 2023-08-09:
> Thank you for that information. I have attached a shot of the output from that. Not sure what to do with that…

It doesn't seem like email attachments are picked up by GitHub. You'll have to post it on the GitHub issue directly.

TheBobC, 2023-08-09:
Yes sir...did that as well

TheBobC, 2023-08-09:
Also...tried using gnome-disks to wipe and create a new partition. I got an error which led me to rewriting the partition table in gdisk, then set a new partition in gnome-disks. Same result.

Process:
open gnome-disks in ubuntu
delete all partitions by selection each and clicking on the red minus button
click on the plus to create a new partition
set it to use the entire disk
set it to other
set it to no file system
got an error - udisks-error-quark, 0
went to gdisk 
verified all partitions deleted
added new partition
confirmed all
finished gdisk with w
went back to gnome-disks and successfully created partition via the above process
went back to Balena to flash HAOS
Same result as previous

TheBobC, 2023-08-09:
Hmmm...really confused now. I put a Samsung 980 SATA drive in it instead of the NVME drive...same result.

TheBobC, 2023-08-09:
Ok…so figuring it HAD to be the machine, I went in and disabled UEFI, rebooted. Then went back in to BIOS and re-enabled UEFI…works fine on either the SATA or the NVME drive now. So there was something hinkey with the BIOS not setting…dunno…odd. But fixed :)  Please close issue

agners, 2023-08-10:
:rofl: interesting. Thanks for the update!

closed 2023-08-10 by agners